KiCad/kicad-library #1137: SMD Footprints: Should not be different for Res and Cap of the same size

kaitrek, 2017-04-03:
From the looks of it, the SMD footprints do not follow IPC specifications, nor are they consistent between libraries.
For example, Capacitors_SMD:C_0603 and Resistors_SMD:R_0603 do not have the same footprint, but they should based on IPC 7351.

I think there should be just 1 library called "SMD_Chip". (There's room for debate over whether SMD_Chip_LevelA, SMD_Chip_LevelB, SMD_Chip_LevelC libraries would be more appropriate, but I think having just 1 library is easier for the user). Each footprint name would indicate density level it's following, for example "0603A", "0603B", "0603C" and hand soldering versions should also be available "0603B_HandSoldering", given that a script would generate the land patterns, I don't see the need to restrict a specific density level whether or not it is allowed to have hand soldering footprints or not.

Whether you refer to the 2005 IPC document or the proposed new C version, both do not distinguish between resistor nor capacitor footprints:
http://pcbget.ru/Files/Standarts/IPC_7351.pdf or http://www.cskl.de/fileadmin/downloads/PCBLIBRARIES/Documentation/What-is-New-in-IPC-7351C_.pdf
http://i.imgur.com/9gDuQQ5.png
http://i.imgur.com/JmZIkZ4.png

metacollin, 2017-04-03:
@kaitrek, I agree with most of what you've said, except for having a single "SMD_Chip" library.  We cannot use a unified set of 0603/0402/0805 etc. footprints because the resistor SMD footprints have different 3D models than the capacitor footprints.  If we were to use a single 0603 footprint, then boards would not have the correct appearance in the 3D view, and incorrect models would get outputted when exporting a 3D model of the board.  

So we need separate R and C SMD footprints, but the actual footprint in pcbnew should indeed be identical.

SchrodingersGat, 2017-04-03:
The R and C footprints should be consistent, and we should also provide A/B/C densities. 

However as @metacollin says they should still be kept in separate libraries.

kaitrek, 2017-04-03:
I forgot that 3D models are associated with footprints, then let's keep the 2 libraries, but the densities, consistent footprints and rounded pads should be implemented.

pointhi, 2017-04-03:
@kaitrek 

There is already some work going on. I wrote a script which can create such footprints, but it's missing actual values: https://github.com/KiCad/Resistors_SMD.pretty/issues/16#issuecomment-282728822

If you are interested, you can improve/complete the table, and we are finally able to create ipc compliant footprints: https://github.com/pointhi/kicad-footprint-generator/blob/master/scripts/Resistors_SMD/smd_chip_resistors_smd.csv

timsu100, 2017-04-09:
I would like to help to create the footprints, but I'm not sure how to interpret the tables in:
http://www.cskl.de/fileadmin/downloads/PCBLIBRARIES/Documentation/What-is-New-in-IPC-7351C_.pdf

On page 41 are the three densitiy levels (A,B,C) described.
The next page lists the different sizes with dimensions for toe, heel and side related to package height. Do I have to add them to the densitiy level table to get the final toe heel and side?

Page 44 shows how to calculate pad length and pad distance. To do this "Nominal L" is required. Where do I get this from?

Maybe SMD fuses, inductors and diodes could also be based on this.

kaitrek, 2017-04-09:
There is no defined Nominal L, it is given by the manufacturer.
However I've been looking at all the manufacturer datasheets, and most of them have a median number that can be used without the min/max tolerance.

I suggest KLC builds a table of Nominal L for internal consumption to generate these footprints.

For example, after looking at the below datasheets, I decided that the 0603 footprint should have 0.35mm Nominal L, since that is the average/median number that kept popping up.
In no particular order:
http://www.yageo.com/documents/recent/UPY-GPHC_X7R_6.3V-to-50V_17.pdf
http://datasheets.avx.com/X7RDielectric.pdf
http://www.kemet.com/Lists/ProductCatalog/Attachments/53/KEM_C1002_X7R_SMD.pdf
http://www.yuden.co.jp/productdata/catalog/en/mlcc_all_e.pdf
https://www.johansondielectrics.com/downloads/jdi-product-catalog.pdf
http://www.vishay.com/docs/45199/vjcommercialseries.pdf
http://www.samsungsem.com/kr/support/library/product-catalog/__icsFiles/afieldfile/2016/09/27/MLCC.pdf

kaitrek, 2017-04-09:
A rule for the size of hand soldering should also be defined for KLC. I think 0.7mm seemed like overkill and 0.4mm was better.

Also FYI, pads should have rounded corners 25% width.

poeschlr, 2017-04-09:
> Also FYI, pads should have rounded corners 25% width.

The library should always be compatible with the current stable version of kicad.
The problem is that the current stable version does not support rounded rectangles. This means if you want to achieve this you need to combine multiple pads of different shapes. This will change when version 5 is released. (The development version already supports rounded rectangles.)

timsu100, 2017-04-10:
So is this Calculation for 0603 Level B correct?

Length = 1.60
Width = 0.80

Toe = 0.275 + 0.35 = 0.625 ~ 0.63
Heel = 0.055 + 0.00 = 0.055 ~ 0.06
Side = 0.083 + 0.00 = 0.083 ~ 0.08
Courtyard = 0.20 + 0.25 = 0.45 ~ 0.5

Pad Length = Nominal L + Heel + Toe = 0.35 + 0.06 + 0.63 = 1.04
Pad Centers = Length + 2\*Toe - Pad Length = 1.60 + 2\*0.63 - 1.04 = 1.82
Pad Width = Width + 2\*Side = 0.80 + 2*0.08 = 0.96

kaitrek, 2017-04-10:
Where did 0.275 come from for the toe? If it's from the "New Solder Joint Goal Recommendations" slide page, I'm not quite sure what that page means. Based on the older IPC-7351 document the table on the page before that one is the right one.

Toe is defined in the table as 0.35mm for 0603.
Heel and Side are 0.0mm.
Courtyard is 0.25mm.
FYI, the round off is to 2nd decimal place, so 0.45mm is 0.45mm, not 0.5mm.

Pad Length = Nominal L + Heel + Toe = 0.35 + 0.0 + 0.35 = 0.70mm
Pad Centers = Length + 2*Toe - Pad Length = 1.60 + 2*0.35 - 0.70 = 1.60mm
Pad Width = Width + 2*Side = 0.80 + 2*0.00 = 0.80mm

timsu100, 2017-04-10:
Yes it is from that page.
I thought you have to take the "New Solder Joint Goal Recommendations" and add the Level A/B/C to it.
But this seems to make the footprints a little bit big.

timsu100, 2017-04-13:
Small Update:
[image]
These are generated from the above formula.
Should the reference designator be in the footprint in the fab layer? The font would be really small on the smaller footprints.

It seems to be hard to find the nominal l sizes, they seem to differ quite a bit.
Some are here:
http://www.kemet.com/Lists/TechnicalArticles/Attachments/29/f2100e.pdf (Table 1,2,8,9 - Dimension T)
https://media.digikey.com/pdf/Data%20Sheets/Samsung%20PDFs/RC_Series_ds.pdf (Page 2 - Dimension B)

SchrodingersGat, 2017-11-21:
@poeschlr has regenerated all these new chip devices at the new https://github.com/kicad-footprints repository. Closing this issue as "solved" :)

closed 2017-11-21 by SchrodingersGat