KiCad/kicad-library #1138: Question about courtyard

hackscribble, 2017-04-03:
Rule 7.5.iv says of the courtyard:

> Clearance is measured from exterior of pads and package outline

For an SMD package such as the one in the image, should the courtyard be the smallest bounding rectangle that satisfies the clearance distance, or should it be a polygon that is narrower at the pin end?

gauravjuvekar, 2017-04-03:
Courtyard is intended to allow space for machine-placing of components or for manual access while soldering by hand.
In your posted image, courtyard should be a rectangle.
However, exceptions can be made in case of eg a sufficiently large weirdly shaped component, where keeping it as a rectangle would waste significant board space.

hackscribble, 2017-04-03:
Thanks, @gauravjuvekar

closed 2017-04-03 by hackscribble