KiCad/kicad-library #1143: KLC 2.1 discussion

evanshultz, 2017-04-04:
@SchrodingersGat @jkriege2 @poeschlr Why wait?

Stemming from KLC 2.0: https://github.com/KiCad/kicad-library/wiki/Kicad-Library-Convention
And the FAQ: https://github.com/KiCad/kicad-library/wiki/Frequently-asked-questions

evanshultz, 2017-04-04:
**KLC**

3.12 Should the FPFilters be prepended with an asterisk too? Imagine the FPFilter is DPAK or TO-252 and that footprint name is TO-252_DPAK. In this case, we'd need an asterisk to catch the DPAK. There are several footprints which have multiple footprint names concatenated together with underscores and we need a leading asterisk to catch them. I don't see a disadvantage to using a leading asterisk on all FPFilters.

4 The use of alias isn't mentioned at all. Perhaps a quick description and example. And there is an ALIAS rule I'd like to clarify also. Should minor differences  (like temp ratings of LM319/LM219/LM119) be an ALIAS or only add one part and let the user modify the part number by hand after symbol placement?

4 Ref des and value placement. Such as center justified if placed on the y-axis, left justified if placed right of the origin, both above or ref des above while value is below, etc. See #848. I'm not sure the logic in KiCad 5.x, but if the libraries should be set up in any specific way to guide 5.x then we should do it now.

4.2 Markings for pin style or input type that aren't part of setting the Pin Type. Like adding a Schmitt trigger symbol on an appropriate logic gate, adding a bubble to show logic low or inverting input, etc. I vote yes on Schmitt trigger but no otherwise (leave it up to KiCad by Pin Type selection).

4.2 Some symbols, especially ADCs and DACs, have intricate inner circuit "summarized" in the symbol. Do we encourage or discourage this? Don't care?

6.6.i/6.6.ii Remove initial "e.g." since this isn't connected to a preceding statement.

10.4.ii Is this rule only a recommendation or a requirement?

The notes in section 5, 6, and the KLC Helper Script have different font/style but all seem to be notes of equal importance. Why aren't all notes using a consistent font/style?

**FAQ**

I think the FAQ should be enumerated so it's easier to refer to a specific item.

https://github.com/KiCad/kicad-library/wiki/Frequently-asked-questions#run-klc-scripts-locally: Broken link to library utils.

https://github.com/KiCad/kicad-library/wiki/Frequently-asked-questions#footprint-association: The footprint naming shows asterisks before each footprint name, as I suggested above. Is this going to be standard practice to use a leading and trailing asterisk? FAQ and KLC need to conform.

https://github.com/KiCad/kicad-library/wiki/Frequently-asked-questions#symbol-variants: Missing underscores in symbol names. Also, why aren't these using the most generic package name instead of the LTC-specific names. Combing these two, the examples could be `LTC4357_DFN` and `LTC4357_MSOP`.

https://github.com/KiCad/kicad-library/wiki/Frequently-asked-questions#multi-unit-symbols: It shows MCP6004 power pin names as 'Vdd' and 'Vss', but the datasheet clearly is using small capital letters. How to handle this? All capital?

hackscribble, 2017-04-04:
KLC section 4 is silent on pin numbering, while `checklib.py` calls `EC01.py` to check for duplicate pin numbers and no pin numbers without a numeral.  Should the EC01 checks be formally included in KLC, either as mandatory or recommended?

evanshultz, 2017-04-10:
**KLC**

3.6 We're not covering when a part has different package types in the same package style. See [L272](http://www.st.com/content/ccc/resource/technical/document/datasheet/02/9c/bc/75/59/9c/47/bd/CD00000054.pdf/files/CD00000054.pdf/jcr:content/translations/en.CD00000054.pdf). It comes in 8 pin DIP and 16 pin DIP. Perhaps give this as an example with `L272_DIP8` and `L272_DIP16`?

3.6 Also, what about if the same part with a different manufacturer suffix is available in the same package? See [MCP6566](http://ww1.microchip.com/downloads/en/DeviceDoc/DS20002143E.pdf
). Most of the time using the simplified package name seems to work well but here it's unpleasant.

4 Does not specify text must not overlap with other elements of the schematic symbol. I assume text definitely should not overlap, but perhaps other elements should be specified as non-overlapping too?

poeschlr, 2017-04-15:
From issue #1159 

Rule 4.3.iv (pin stacking: stacked pins have the same electrical type) creates erc errors if these pins are power output pins. Not sure if we should solve this problems by creating an exception for power output pins or if the erc checker should be changed. (I think the later would be the better long term solution. But well we would need somebody who does it. In the meantime i think we should "fix" it from the library side.)

If we want to "solve" the problem via avoiding stacked power output pins maybe either add a new rule:
> If a component has multiple power output pins that are interconnected, these pins are stacked and only one pin is set as visible with electrical type power output. All other pins are invisible and electrical type passive.

Or make an exception to 4.3.iv:
> Pins must have the same electrical type (Exception: If the pins are of type power output only the visible pin has the type power output all other stacked pins are of type passive.)

The problem here is that the symbol editor does not like this. It changes all stacked pins to be equal if one changes any pin while they are stacked. This means if one wants to edit something in the pin properties one needs to move them apart, do the changes and move them back on top of each other.

maqifrnswa, 2017-05-04:
Something new consider, based on https://github.com/KiCad/kicad-library/pull/1207#issuecomment-299075457

For consistency in style: "When drawing symbols follow, the symbols should try to follow that used on the datasheet, except when the datasheet style conflicts with other components of KLC"

This is to avoid situations in the audio library where some devices look like op amps and some look like ICs. It just helps clear up some ambiguity and allows kicad to defer style decisions to the manufacturer  This may be hard because different manufacturers may use different styles, and the same datasheet may use different styles.

bobc, 2017-06-28:
**Keywords**

Keywords seem to be a bit random and not very consistent. perhaps we could have a list of suggested/recommended keywords, maybe on per library basis. 
I have a script to extract keywords, here is a snippet for 74xx.lib

```
3
3State
ADD
ALU
and2
And3
And4
ANDNOR
Arith
BIDI
buffer
BUS
CNT
CNT4
COMP
DECOD
DECOD10
DECOD16
DECOD4
DECOD7
DECOD8
DEMUX
DEMUX4
DEMUX8
DFF
DFF4
DFF6
DFF8
```
"3" is not a very useful search term. A lot of the "codes" are not very guessable, unless you already know what you are looking for.

SchrodingersGat, 2017-06-30:
@bobc I think suggesting a "style" for keywords is a good idea. It would be good to give some long-form description of what constitutes a good keyword (and what doesn't).

A lot of it is "common sense" but sometimes does need to be explained.

matthijskooijman, 2017-07-05:
I just read through the KLC and the FAQ for the first time, and thought to write down things that I noted. I'm not quite familiar with the footprint standards (and only moderately familiar with KiCad), so take my suggestions with a grain of salt :-)

> 1.4: When dimensions are used in part name, they are in millimeters, decimal places separated by a dot e.g. C_Rect_L13mm_W4mm_P10mm 

Having a decimal in the example might be helpful.

> 3.3: Manufacturer Part Number (MPN) should be used given preference

I think this sentence has extra words or is missing them, but I'm not sure.

> 4.6.i: Power and Ground pins should be set to either POWER INPUT or POWER OUTPUT

I believe ground pins are always POWER INPUT? If so, this should be stated. If not, some additional guidance could be given. In general, some additional guidance about electrical types wouldn't hurt.

> 4.10.i.a: For symbols with a default footprint, the human-readable footprint name should be appended to the description e.g. SOIC-8
> 4.10.iii: For parts with different symbols associated with different footprints, shortform footprint name should be specified in the description field (e.g. DIP8)

These rules look redundant to me.

> 4.11.ii: For components with a single default footprint, the Footprint field is filled with a valid entry of the format Footprint_Library:Footprint_Name
> 4.11.iii: For components with multiple possible footprints, the Footprint field should be left blank

When does a part have "a single footprint" exactly? The FAQ states that parts which are supplied with different footprints should become separate symbols, so the part a symbol refers to is only available with one footprint. It could also mean there is only a single appropriate footprint in the footprint libraries, but isn't this something that can change over time when alternative versions (e.g. handsoldering versions) are added. My hunch is that parts should pretty much *always* use wildcard filters (or at least by default).

In fact, the FAQ already contains a confusing example:

> For example, the INA19x operational amplifier is available in a SOT-23-5 footprint, and so the corresponding footprint field should be set to (for example) TO_SOT_Packages_SMD:SOT-23-5

However, there's two relevant footprints ([SOT-23-5](https://github.com/KiCad/TO_SOT_Packages_SMD.pretty/blob/master/SOT-23-5.kicad_mod) and [SOT-23-5_HandSoldering](https://github.com/KiCad/TO_SOT_Packages_SMD.pretty/blob/master/SOT-23-5_HandSoldering.kicad_mod) already, so a filter would be more appropriate.

The next example is also contridactory:

> Footprint filters use the asterisk (*) symbol as a wildcard to allow matching various footprint names. This allows matches for variants of the same footprint e.g. SOIC-8* will match SOIC-8 and SOIC-8_HandSoldering.

Firstly, the mentioned "SOIC-8" and "SOIC-8_HandSoldering" do not exist (anymore?), only "SMD_Packages:SOIC-8-N", "SOIC:SOIC-8-1EP_3.9x4.9mm_Pitch1.27mm" and "Housings_SOIC:SOIC-8_3.9x4.9mm_Pitch1.27mm".

Secondly, it contradicts with the KLC:

> 4.12.iii: Filters should not match the pin count - this is performed by KiCad's internal footprint association
> 4.12.iv: Filters should match dimension information (where required) to be as specific as possible, e.g.
    b. SOIC*3.9x8.7mm*Pitch1.27mm* to match SOIC-14_3.9x8.7mm_Pitch1.27mm

On a related note: Noting the "multiple footprints" -> "multiple symbols" rule in the KLC might be useful and help interpreting some of the other rules?

> 6.2: Package name and number of pins are separated using a hyphen. e.g. SOT-89, SOT-23-5. 

This sort of implies that SOT-89 has 89 pins. What's the point of this example?

> 8.1: Footprint Placement type should be set to Surface mount
> 9.1: Footprint Placement type should be set to Through hole

Perhaps this should note that older KiCad versions call this "Attributes":  "Normal+Insert" and "Normal" respectively?

> FAQ: Generic vs Atomic parts

Are parts like "2N7000", which are manufactured by multiple manufacturers atomic or generic parts? I suspect generic, but this might need an example.

suzizecat, 2017-07-06:
> When does a part have "a single footprint" exactly? The FAQ states that parts which are supplied with different footprints should become separate symbols, so the part a symbol refers to is only [...]

Sometimes, you can have a single component with multiple footprints and almost the same part number (see http://www.ti.com/lit/ds/symlink/cd74hc08.pdf for an example). In this case, you just let the user chose which one he wants. If you only got one FP possible, you'll add it in the component field.

In any case, you should fill the FP filtering fields, even if there is only one choice possible. With that, if someone wants to add a variant to the footprint (handsoldering for example) then it's possible and the user can easily change it.

matthijskooijman, 2017-07-06:
> Sometimes, you can have a single component with multiple footprints and almost the same part number (see http://www.ti.com/lit/ds/symlink/cd74hc08.pdf for an example). In this case, you just let the [...]

This is not how I interpret [the FAQ](https://github.com/KiCad/kicad-library/wiki/Frequently-asked-questions#symbol-variants):

> Where components have multiple footprint options, a separate symbol should be drawn for each footprint option. Each symbol should have the appropriate aliases added if required.

In your example, I would think this would need multiple symbols (not aliases), e.g.
 - CD74HC08E for the PDIP package
 - CD74HC08Mx for the SOIC package
 - CD74HC08Px for the TSSOP package

>In any case, you should fill the FP filtering fields, even if there is only one choice possible. With that, if someone wants to add a variant to the footprint (handsoldering for example) then it's possible and the user can easily change it.

That would also make sense to me, yes. But that's not how I interpret the current rules:

> 4.11.ii: For components with a single default footprint, the Footprint field is filled with a valid entry of the format Footprint_Library:Footprint_Name
> 4.11.iii: For components with multiple possible footprints, the Footprint field should be left blank

To me, it looks like 4.11.ii and 4.11.iii are mutually exclusive, but perhaps I'm misinterpreting and they are expected to both apply in a lot of cases (so a component has a "single default footprint" in the Footprint field, but also "multiple possible footprints" in the alias field).

bobc, 2017-07-06:
> symlink/cd74hc08.pdf for an example). In this case, you just let the user

My understanding is that parts with multiple footprints that nevertheless
have identical pin out can be a single component.
For parts that have different pin out in different footprints, then these
must be separate components.

4.11 ii and iii seem more like "recommended" rather than "mandatory".

poeschlr, 2017-07-06:
> To me, it looks like 4.11.ii and 4.11.iii are mutually exclusive, but perhaps I'm misinterpreting and they are expected to both apply in a lot of cases (so a component has a "single default footprint" in the Footprint field, but also "multiple possible footprints" in the alias field).

It does talk about different use cases. One applies where you have one specialized symbol per footprint. (Most ICs fall into this category.) The other rule talks about cases where this does not make sense. (Example: Resistors, Capacitors, Diodes, Connectors, ...)

hackscribble, 2017-07-06:
> 4.11.ii: For components with a single default footprint, the Footprint field is filled with a valid entry of the format Footprint_Library:Footprint_Name
4.11.iii: For components with multiple possible footprints, the Footprint field should be left blank

> To me, it looks like 4.11.ii and 4.11.iii are mutually exclusive, but perhaps I'm misinterpreting and they are expected to both apply in a lot of cases (so a component has a "single default footprint" in the Footprint field, but also "multiple possible footprints" in the alias field).

> It does talk about different use cases. One applies where you have one specialized symbol per footprint. (Most ICs fall into this category.) The other rule talks about cases where this does not make sense. (Example: Resistors, Capacitors, Diodes, Connectors, ...)

We should remove the word `default` from 4.11.ii.  

With it, it can be read as meaning " a preferred footprint but others are possible".   Without it, the two bullet points align with the use cases that @poeschlr described.

jkriege2, 2017-07-06:
Those rules were designed for the case where you have a component that is available in multiple packages (if there is only one package: all is clear). In such case we have several sub-cases:
1. all variants have the same pin-number and share the same pinout. Then you can have one symbol, which does not have a definite footprint (i.e. footprint field empyt), but has a list of FPFilters that maches (together with the pin-count filter) each available package ... this is often the case for TTL logic devices that are e.g. available in SO-14, TSOP-14, DIL-14, ... all with the same pinout
2. some variants have a different pinout: Then you need a separate symbol AT LEAST for those variants. The others can be summarized in a single symbol using the FPFilter strategy. The other variants can either have set footprint fields or also give several options by FPFilters.

This approach has several advantages: We don't need one symbol per package, i.e. we don't have 3 or for 7414, but only one. The symbol then represents the function of the device and the footprint is associated when the schematic is complete and you decide which variant to place on the board. Otherwise you will have to go back to the schematic delete all sub-symbols of that IC13, which you now want in DIP14, not TSOP-14 and add them all again. Also this makes it easier to manage the lib, as we need to fix only one 7414 symbol and not 3 if there should be a bug.

So yes, we should remove the word `default`.

JAN

matthijskooijman, 2017-07-07:
> all variants have the same pin-number and share the same pinout. Then you can have one symbol, (...)

This sounds reasonable to me, but perhaps the FAQ should be clarified. It gives two examples that need different symbols (different packages with different numbers of pins, and same package with different pinout), but it lacks an example where different footprints have the same pinout and can thus share a symbol.

Here's the two current examples I mentioned:

> As a first example, consider LTC4357 which is available in a 6-pin DFN package and an 8-pin MSOP. This requires two symbols to be added,
> (...)
> The last example is when a part has different pinouts in the same package. MCP6566 is available with 3 different pinouts all in the same SOT-23-5 package. Three symbols should be created, 

In hindsight, I think I originally read the first example as "Multiple footprints with the same pinout should still have multiple symbols", but re-reading I now see this talks about 6-pin vs 8-pin, which are by definition not compatible, so it's probably just a matter of adding an example, and perhaps improving this part of the FAQ:

> For parts that are available in different footprints, a unique part should be created for each variant. These variants should be named according to the manufacturer datasheet.

Also, I was thinking that the rules stated that FP filters should only be used when the Footprint field is empty, but re-reading the rules that understanding was completely wrong. The rules are even explicit about this:

> 4.11.v: Filters should be used even if a default footprint is given

I'm not quite sure where my confusion came from, so sorry for the noise :-S

I did notice a new issue:

> 4.11.iii: Footprint filters should be added as appropriate to match all possible footprints (see KLC 4.13 below)

This refers to 4.13, but that doesn't seem appropriate. Perhaps it should reference 4.12 instead? Same for:

> 4.9.3: The Footprint field is filled according to rule 4.12 (below) and is invisible

I think that should reference 4.11 instead.

bobc, 2017-08-31:
From #1580 

> we also don't fill transistor-circles with yellow ...

That might be our practice, but is not reflected by the KLC!

> 4.2 iii Fill style of symbol body is set to Fill background

Clearly according to the KLC transistors should be filled. There is no exemption allowed for transistor or other symbols. If we want to exempt transistors etc, that needs to be written in and not applied on an _ad hoc_ basis. 

We probably don't want "basic" symbols resistors and capacitors with filled body, but there are currently some symbols which I would not expect to be filled, e.g. opamps and crystals. 

The question is where do you draw the line, and do you define it? I would propose that an exception is made for symbols with a "distinctive shape" (term used by IEC). Only symbols with a rectangular shape should be filled, but we need to make a special case for the box resistor symbol. There is also a practical problem in KiCad for unusual shapes, they are impossible to fill unless part or all of the outline is decomposed to polylines (e.g. OR gate, valve symbol).

Some proposed wording:

4.2(iii)(a) Symbols primarily comprising of rectangular outline MUST be filled with background color (excluding 'box' resistor symbol)

4.2(iii)(b) Symbols with a non-rectangular outline MAY be filled with background color

4.2(iii)(c) Discrete components (i.e. not ICs) MUST NOT be filled with background color

I think that covers all of the current and intended practice. Op-amps and traditional logic gates would fall under 4.2(iii)(b). If we wanted to remove the optional 4.2(iii)(b), we could write 4.2(iii)(a) as "Symbols representing an integrated circuit MUST be filled..."

There are quite likely some exceptions that cannot be handled with generic rules and need to be written in explicitly.

jkriege2, 2017-09-01:
I second the proposal by @bobc above!
JAN

SchrodingersGat, 2017-09-17:
KLC 2.1 becomes KLC 3.0!

I've been reading back through this as a reference, here's my suggestion for the "background fill" issue.

(Note that this is now on the KiCad website :) )

[image]

bobc, 2017-09-17:
I think that covers two cases that everyone agrees on, but leaves the other cases open. I still don't understand how "hidden functionality" can be objectively applied, whether the function is "obvious" or "hidden" seems to depend on the knowledge of the user.

Some border cases:
1. op-amps, these are currently filled (should not be?)
2. crystals, some have a filled body (should not)
3. 'traditional' logic gates, they have a distinctive shape so should not be?
4. 'box' style IEC logic gates, not discrete, but not really hidden functions. Which fill?
5. flip-flops, do they have hidden functionality?

I would also encourage the consistent use of "must/may" for mandatory/optional qualifiers. Other words such as "should" can be ambiguous.

SchrodingersGat, 2017-11-21:
Closing this issue as [KLC 3](http://kicad-pcb.org/libraries/klc/) is now released.

closed 2017-11-21 by SchrodingersGat