KiCad/kicad-library #1145: KLC 4.8 symbol text size rule mandatory?

hackscribble, 2017-04-04:
I would like to update `nxp_armmcu.lib` to make it more KLC compliant.  

A common issue across many (all?) of its symbols is that the pin names and numbers are 35mil. Increasing to 50mil as per KLC 4.8 will mean increasing the symbol size and moving the pins (see image) to avoid the pin names overlapping.  I don't mind doing this but I want to check first if the rule is mandatory?

Thanks.

[image]

poeschlr, 2017-04-05:
I would not include all this information in a symbol. The human brain just can't handle it ;) (Or mine at least)
I would only include the port name (The PIOx_y part.) and remove the detailed description. (This information is either handled in the programming environment or at least in the datasheet. The user can also use local labels to show how a particular gpio pin is used.)

For this component it is not so bad but look at a typical cortex m4 micro processor. ([Example stm32f4 from the forum](https://forum.kicad.info/t/component-occupies-entire-page-newbie/5272/3)) 
There a symbol created in this manner would be larger than a A4 paper.

And at the end a quote from @SchrodingersGat  from within the same thread that gives an outlook of how this could be handled better in the new library format:
> One feature that will hopefully find its way into the new schematic file format (don't ask me when this will be completed) is the ability to assign multiple functions to a pin, and then select which one is actually in use.
This will drastically cut down on pin-label text, and will actually allow the various pin modes to be included in the ERC (e.g. MOSI is an output whereas ADC0 is an input)

hackscribble, 2017-04-05:
Thanks @poeschlr. Looks like I'll be able to shrink the symbols :smile:

closed 2017-04-05 by hackscribble