KiCad/kicad-library #1148: Use of aliases

hackscribble, 2017-04-05:
As part of updating `nxp_armmcu.lib` (see https://github.com/KiCad/kicad-library/issues/1145), I plan to rationalise symbols that are identical (apart from manufacturer variant suffix) and use aliases instead.  

Consider this subset of the parts which have identical pinouts and use the same package (the variants reflect different memory size etc).

```
LPC1111FHN33/101 
LPC1111FHN33/102
LPC1111FHN33/201 
LPC1111FHN33/202
LPC1112FHN33/101 
LPC1112FHN33/102
LPC1112FHN33/201 
LPC1112FHN33/202
```

I can think of a couple of ways of handling this.

1. Have one part called `LPC111xFHN33` and add the eight part numbers above as aliases.
2. Have one part called `LPC1111FHN33/101` and add the other seven part numbers as aliases.

I would welcome guidance on which approach to take, or a better way to do it.

SchrodingersGat, 2017-04-05:
Good idea. Options would be:

1. Create a generic named part that covers all options e.g. LPC111xFHN33 and does not have any aliases. The x indicates it is essentially a one size fits all.

2. Follow option 2 that you suggested above - I would suggest doing that.

hackscribble, 2017-04-05:
Thanks @SchrodingersGat 

I have just spotted that there are other parts that match the pattern `LPC111xFHN33` but have different pinout.  So option 2 is definitely the way.

closed 2017-04-05 by hackscribble