KiCad/kicad-library #1159: Pin type for multiple power output pins

hackscribble, 2017-04-07:
For devices with multiple power output pins that are intended to be joined together to increase current capacity, what is the preferred way of setting the pin type in the symbol to avoid ERC errors?

I've seen examples with:
- all power output pins set to `PASSIVE`
- one power output pin set to `POWER OUTPUT` and the rest to `PASSIVE`

Thanks.

PhirePhly, 2017-04-15:
I just tripped on this trying to use an 1117 voltage regulator in as SOT-223 package. How should this be resolved?
[image]

poeschlr, 2017-04-15:
@hackscribble i would go with option 2:
> one power output pin set to POWER OUTPUT and the rest to PASSIVE

I would specify it as:
If a component has multiple power output pins, these pins are stacked and only one pin is set as visible with electrical type power output. All other pins are invisible and electrical type passive.

But i found a problem with that. If you put two pins on top of each other, the symbol editor gives them the same electrical type. (At least if you change anything in one of the stacked pins. A workaround would be to make the pins non stacked, and move them in place as the last step.)

hackscribble, 2017-04-15:
Thanks @poeschlr

closed 2017-04-15 by hackscribble