KiCad/kicad-library #1170: Footprint filters for conn_??x?? symbols are very restricted should they include more footprints?

poeschlr, 2017-04-18:
We have a lot of footprints for different connectors. But the conn_??x?? have only pin headers and socket strips in their footprint filters.
Any suggestions how the footprint filters should look like? 
Should we add all applicable footprints manually or is there a better option?

SchrodingersGat, 2017-04-18:
This behaviour is in nightly only but it would pay to make use of this for 5.0.x

poeschlr, 2017-04-19:
How does the stable version react if there is a ":" character in one of the footprint filters?

SchrodingersGat, 2017-04-19:
It tries to match it just against the footprint and fails

SchrodingersGat, 2017-04-19:
It was exactly for the use case that you describe here that I implemented this filter feature :)

poeschlr, 2017-04-19:
Thanks  for implementing this feature and for the clarification.
If nobody else volunteers i would add this filters to the conn_ symbols. But i probably will not be able to do it before the end of may.

SchrodingersGat, 2017-04-19:
Should be easy enough to fix with a handy text editor and some regex :)

poeschlr, 2017-08-22:
@SchrodingersGat What nightly version did add this feature? I would like to test this but i think my nightly might be a bit too old. (I might need to update my operating system to get updates again.)

evanshultz, 2017-08-24:
@poeschlr Can this be closed now?

closed 2017-08-24 by poeschlr