KiCad/kicad-library #1211: many files have a CRLF DOS line ending

jonasstein, 2017-05-01:
a quick
find kicad-lib* -not -type d -exec file '{}' ';' | grep CRLF
on the kicad-library tree revealed that many files have a CRLF DOS line ending.

Mixing the file endings in large projects can cause trouble and most likely you want the unix line ending standard for all files.

```
$ find kicad-lib* -not -type d -exec file '{}' ';' | grep CRLF
kicad-library/modules/packages3d/Valves.3dshapes/VALVE-MINI_PENTODE_LINEAR.step: ASCII text, with CRLF line terminators
kicad-library/modules/packages3d/TO_SOT_Packages_THT.3dshapes/TO-8-3.stp: ASCII text, with CRLF line terminators
kicad-library/modules/packages3d/TO_SOT_Packages_THT.3dshapes/TO-8-2.stp: ASCII text, with CRLF line terminators
[..]
```

bobc, 2017-05-02:
Exactly what trouble? Given the fact there are lots of mixed line endings with no reported problems suggests that it is actually not a problem.
I get that some people obsess over details but the line ending "issue" is really not important. As soon as you "fix it" it will happen again, but it still won't matter.

jonasstein, 2017-05-02:
inconsistent file format can cause bugs, which consume time. Just a question of time. The gitclient is prepared to handle this automatically. https://help.github.com/articles/dealing-with-line-endings/ 
So it is just a misconfigured git client on a Win machine.

bobc, 2017-05-02:
> inconsistent file format can cause bugs,

Do you have proof for that statement? Programs processing text files are designed to expect different line endings and process them correctly.

I think there is no "question of time", we would have seen issues already if it was a significant problem. I fit hasn't causes a problem so far, it likely never will.

jkriege2, 2017-05-03:
This was already discussed and yes, line-ending SHOULD be UNIX on the GitHUB-Servers. I think there should be `.gitattributes`-files that give the rules for text-files. But this only works when the files are checked in (once) as Unix, then Git converts them to you local settings and converts them back when pushing.

But for now I would say we should have all the .lib and .kicad_mod-files consistent. About all the 3D models I'm not so sure ...

If you want/have time, you could make a PR to fix this problem ... (maybe start with the .lib/.dcm-files) ... and we go on discussing there.

Best,
JAN

BsAtHome, 2017-05-15:
The problem files are in the attached text file of .pretty libraries (simple clone and a git status).
[kicad_pretty_nl_crnl.txt](https://github.com/KiCad/kicad-library/files/1002259/kicad_pretty_nl_crnl.txt)

The list could be easily scripted and run through dos2unix. All files are identified by something like this:
`$ for i in *.pretty; do ( cd $i; git status | grep 'modified:' | THEDIR=$i awk '{printf("%s/%s\n", ENVIRON["THEDIR"], $(NF))}' ); done`

Run it in the directory where all the pretty repos are. You can easily modify it to run dos2unix on the identified files.

gauravjuvekar, 2017-07-09:
/me may have spammed a lot of people with pull requests just now.
Anyways, should we add this as one of the tests to Travis CI (if it isn't already)?

poeschlr, 2017-07-09:
A travis testcase might be a good idea.

bobc, 2017-07-09:
I see a good reason for not changing line endings is that it really messes up git diff!

A test like

```
""" return True if all line endings valid for current OS """
def test_line_term (filename):
    f = open (filename, 'U')
    s = f.readline()
    while s:
        s = f.readline()
    if f.newlines != os.linesep:
        return False
    return True
```

would ensure the line endings are valid for the user's current OS (or the Python environment).

SchrodingersGat, 2017-07-09:
Currently the [.gitattributes file](https://github.com/KiCad/kicad-library/blob/master/.gitattributes) is set to:

`* text = auto`

Which according to the [documentation](https://help.github.com/articles/dealing-with-line-endings/) is a "good option" (whatever that means).

From the above link it appears that this should be set instead to:

`text eol=lf` to **force unix line endings**

Would you all agree?

bobc, 2017-07-09:
Nope! That sets working files to LF on checkout, which is OK for Unix but not Windows.
I believe

> * text = auto
or
> *.kicad_mod text=auto

is correct for us.

Note that if files already in git have CRLF endings when .gitattributes is changed, those files get "stuck" with CRLF, which may explain why we ended up with some bad files in git, even though we now have text=auto set in .gitattributes. There is a procedure described in git manual for "normalizing" files and the setting [.gitattributes](https://git-scm.com/docs/gitattributes).

SchrodingersGat, 2017-07-09:
Thanks for the clarification @bobc . So after a purge of offending lines, git should take care of the rest?

poeschlr, 2017-07-09:
If nobody is using a buggy git client then yes. 
I'm also not sure if the gitattribute file is supported in older git clients so we have to see what the future brings.
A travis check might help us catch future problems before they enter the lib.

poeschlr, 2017-07-09:
Other problems might exist with very old pull requests where the gitattributes file is non existent in the branch of the contributor.

SchrodingersGat, 2017-07-09:
> Other problems might exist with very old pull requests where the gitattributes file is non existent in the branch of the contributor.

Good point

gauravjuvekar, 2017-07-09:
> Other problems might exist with very old pull requests where the gitattributes file is non existent in the branch of the contributor.

Can't we just rebase / merge master before?

poeschlr, 2017-07-09:
That would need to be done by the contributor. (We librarians don't have control over the contributors branch.)
Rebasing can be quite dangerous if not done right. (Data might get lost.)
Merging is more likely to be done correctly.
(Just remember that a lot of our contributors don't come from an informatics background but from an electronics background and therefore might make their first steps with git by contributing here.)

Also the changed files section explodes if you merge the master into an old pull request. (Makes it hard to see what the contributor changed.)
For this reason i would only suggest to merge the current master into the contributors branch if there is a conflict.

gauravjuvekar, 2017-07-31:
https://github.com/KiCad/Battery_Holders.pretty/pull/13

gauravjuvekar, 2017-07-31:
Added the .gitattributes file to remaining .pretty repos. Can we close this now? @SchrodingersGat @hackscribble @poeschlr @jkriege2

SchrodingersGat, 2017-08-01:
Thanks @gauravjuvekar , happy to close.

closed 2017-08-01 by SchrodingersGat