KiCad/kicad-library #1222: Wrong symbol for optocoupler / standard key/value pair as solution?

jonasstein, 2017-05-06:
Most optocouplers in the library have two zigzag arrows between the light emitter and light sensor:
[image]

IEC 60617-2: 05-06-08 looks like
[image]

The KiCad library symbols do not claim to follow any standard. 
Perhaps the only clean solution to get a schematic in one standard is to add a new tag: 

```
standard=xkcd # symbols as painted in xkcd comics
standard=peters-symbols # symbols as my boss peter wants them
standard=IEC 60617-2 # symbols as defined in some proprietary standard
```
A standard can include various symbols for the same object like
```
diode-EU
diode-US
or
transformer-detailed
transformer-minimal
```
Thus a second tag might be handy in the long term. The user could then switch between the standards.

evanshultz, 2017-05-06:
Hi @jonasstein 

KLC (https://github.com/KiCad/kicad-library/wiki/Kicad-Library-Convention) is the standard governing KiCad's symbols. While you're quite correct that there's a lot of work to do in order to refine the library and get everything in compliance, you can see that rule 4.2.iv specifies that IEC symbols are to be used unless some deviation is requested. So there is a standard, but admittedly the exact IEC doc isn't shown.

While any user can use KiCad to create symbols any way they like, integrating into the main library should be done in a consistent, unified way so the library supports KiCad users well. As noted above, improvements to the existing libraries, updated KLC standards, and limited librarian time mean the current libraries are not being met in many cases.

That being said, thanks for pointing out a place where the libraries can be improved. If you would be so kind as to create a PR with the symbol changes, they can be formally reviewed and considered for merging into the library. I may take some time to look at this more and others may also discuss here before a PR.

jonasstein, 2017-05-06:
Hi @evanshultz , thank you for the quick response. I still have to learn KiCad more and will not be able to make a PR for this soon. Perhaps it can be fixed with a script by someone who is a library expert. There are too many false optocoupler symbols for manual repair.

bobc, 2017-05-14:
I have written a script which changes most of the zigzags, I will submit a PR.

poeschlr, 2017-08-23:
I looked at the current symbols in the opto lib. None of them seem to have zig zag arrows any more. I assume this can be closed.

closed 2017-08-23 by poeschlr