KiCad/kicad-library #1225: DB25MC mechanical holes

christakahashi, 2017-05-08:
DB25MC has mechanical holes at Y= -2.54,  They should actually be between the rows of pins and Y= -1.27,  At least this is the case for all the connectors I have (2 brands)

evanshultz, 2017-05-09:
@christakahashi Hi! Thanks for this report.

I assume the footprint you're talking about is Connectors/DB25MC:
[image]

And the issue you're reporting is captured here (note that the origin is the blue "+"):
[image]

If that's right, you're looking at a right-angle, 25-pin, male, DB-style connector.

Can you provide any datasheet links against which we can check the footprint?

Here are some I found and their dimension:
- [TE 5747238-4](http://www.te.com/commerce/DocumentDelivery/DDEController?Action=srchrtrv&DocNm=5747238&DocType=Customer+Drawing&DocLang=English): 1.42mm
- [Amphenol/FCI D25P33E4GV00LF](http://portal.fciconnect.com/Comergent//fci/drawing/c01-8646-0151.pdf): 1.42mm
- [Amphenol D25P13A4GL00LF](http://portal.fciconnect.com/Comergent//fci/drawing/c01-8646-0164.pdf): 1.27mm
- [HARTING 09653626812](https://b2b.harting.com/files/download/PRD/PDF_TS/0965362X81X_BL05_R31892_100078415DRW002A.pdf): 1.27mm
- [Assmann ADS25A-KG-TASB5](http://www.assmann-wsw.com/fileadmin/datasheets/ADSxxA-KG-TASB5-R.pdf): Not given but looks like 1.42mm
- [Amphenol FCE17-B25PA-450](https://www.amphenolcanada.com/ProductSearch/drawings/AC/FCE17XXXXAXX0X.pdf): 1.42mm
- [Omron XM3-L/XM2-L/XM4K/XM4L](http://www.omron.com/ecb/products/pdf/en-xm3_l_xm2_l_xm4k_xm4l.pdf): 1.42mm

So what I see is that 1.42mm is most common, but there is plenty of variation. And none of the above match the footprint in the library. Perhaps we need to have a couple footprints that cover all the parts in the field?

I'd appreciate if you can add more parts into the mix so we have a better overview of the components that KiCad users might want to put on their boards.

In the future, it's best if you can file this type of issue with the relevant footprint library. For this item, it is at https://github.com/KiCad/Connectors.pretty.

christakahashi, 2017-05-09:
Ah! sorry i got the repo wrong.  Thanks for the correct pointer.  I know where to look now.  You are correct in your assumptions; I am referring to Connectors/DB25MC:.  I have the assmann part.  The other part I have is just in a bin so I don't have part numbers.  I used calipers and rounded to 50mil to verify but I couldn't really distinguish between 1.27 and 1.42.

Looking at the datasheets it looks like the pin pitch is always 2.77 give or take 10 microns, which makes since because the D-sub connectors have specified pin spacing but the rows can be anything and vary.  I suspect a part with 2.54 spaced rows and 2.84 spaced rows will probably work in the same footprint.  I think the key commonality though is that the mechanical hole is evenly spaced between rows.  

In the case of the kicad footprint the row spacing is 2.54 (which fits my Assmann part with 2.84 row spacing fine i have working PCBs with cutoff mechanical tabs)  so I think the most correct thing to do is to put the tabs at y=-1.27.

There is the question of 1 or 2 foot prints.  Given the similarity my opinion is to side with simplicity over absolute correctness and keep just one footprint with the row spacing as is, but perhaps the development community has guidelines or ideas on how to resolve that.

evanshultz, 2017-05-09:
Looking at the comments inside the [DB25MC.kicad_mod file](https://github.com/KiCad/Connectors.pretty/blob/master/DB25MC.kicad_mod), it's in French and Google tells me "Connecteur DB25 MALE couche" translates as "Connector DB25 MALE Layer".

There is also a DB25MD which has a comment that translates to "Connector DB25 MALE right".

I assume that "couche" and "droit" are the "C" and "D" in these footprint names, which needs to be corrected to English and better described by the footprint name, so thanks for incidentally pointing this out. It would only make sense that there is a part matching this footprint, but unfortunately the footprint description doesn't help us with that, so the best bet may be to create footprint(s) to support the parts identified above. If you can do some more searching online it would be helpful.

bobc, 2017-05-13:
This has been noted before https://github.com/KiCad/Connectors.pretty/issues/44. I was never able to find a part matching the DB25MC footprint.

The terminology "couche" = right-angle, and "droit" = straight (vertical) is used on several other connectors in Connectors.lib. There is also "CMS" meaning SMD in French.

SchrodingersGat, 2017-10-06:
I am closing this issue for now - any further issues with the DSUB footprints should be directed to https://github.com/kicad/connectors_dsub.pretty

closed 2017-10-06 by SchrodingersGat

jkriege2, 2017-10-18:
just a note: I'm currently working on a script-based rework of the DSUBs

Best,
JAN

evanshultz, 2017-10-18:
Great! That was on my list to do eventually. I had also marked https://github.com/KiCad/Connectors.pretty/issues/44 as something to be reviewed when these connectors were updated. Looking forward to your rework!

jkriege2, 2017-10-19:
Have a look at the results: https://github.com/pointhi/kicad-footprint-generator/pull/69

gnbl, 2017-11-09:
Nitpicking, but note it should be _D-Sub_ and not _Sub-D_: https://en.wikipedia.org/wiki/D-subminiature
It would be better to adhere to some standard in order to avoid confusion/duplicates/having to check two places.

I'm here because I can't seem to find any DB25 footprints in 4.0.6., but it seems it's outdated, so:
1. Footprint Editor
2. Footprint Libraries Manager
3. add GitHub library `${KIGITHUB}/Connectors_DSub.pretty`