KiCad/kicad-library #1243: extension of KLC 4.3 (pin stacks)

jkriege2, 2017-05-15:
To complement the discussion about an extension of the kicad library check scripts (https://github.com/KiCad/kicad-library-utils/pull/116) I'm proposing a change to KLC:

We were discussing how to deal with power-I/O and output pin stacks. There are several problems with these:
1. power-pins should always be visible, as invisible power-pins are reserved to power-flags (+12V, +5V, GND, ...)
2. no two output or power-output pins may be tied together as this causes an ERC violation (two outputs driving one signal).

Therefore I propose an additional rule in KLC 4.3:
> 4.3.vi stacks of output-, power-output- or power-input-pins are a special case for which the following strategy is required: One pin of the stack has the required type (power-I/O, output) and is visible. All other pins in the stack have the electircal type PASSIVE and are invisible.

The script mentioned above checks for such stacks and provides auto-fixing for violations.

Best,
JAN

SchrodingersGat, 2017-06-04:
Hi Jan if you look at the KLC I have just now updated to add this rule. I have also added a brief note to the FAQ page

jkriege2, 2017-06-05:
Thanks,
JAN

closed 2017-06-05 by jkriege2