KiCad/kicad-library #1259: Licensing of symbols and footprints (and current 3d models in this repo)

poeschlr, 2017-05-25:
Currently the libraries seem to be under normal gpl library.
Should we add a symbol exception similar to what we did with the new 3d models library to these repos as well?

bobc, 2017-05-25:
The license is Lesser GPL, which already has a "linking" exception.

I don't think engineering data can be covered by copyright, and LGPL is designed for software, so "linking" may not be relevant. But as far as there is any legal application, the user is allowed to incorporate symbols from the official library into his/her design without incurring any GPL obligations.

poeschlr, 2017-09-18:
I just checked. No it seems the libs are under normal GPL not LGPL.

So the question still stands. Should we make the symbol exception clear in the lib?

SchrodingersGat, 2017-09-18:
What serendipity - this just popped up on the forums - https://forum.kicad.info/t/default-libraries-gpl-licencing-vs-proprietary-designs/140/8

We *definitely* need to come up with a clear exception for all the libraries. I would make this a required task for the KiCad v5 / KLC v3 reboot.

Additionally, I'm not sure if we need to include a statement to the effect that "users contributing to the libraries implicitly agree to license their contributions under the above agreement". Otherwise each little touch-up of a file will require attribution and that'll become very hard to manage.

poeschlr, 2017-09-18:
That's why i posted here-

I also found this discussion you had with the developers:
https://www.mail-archive.com/kicad-developers@lists.launchpad.net/msg24570.html

It basically states we should use CC-BY-SA 4.0 with some weird exception that i do not really understand.

SchrodingersGat, 2017-09-18:
Yeah I just reignited that flame again.  However I just found this much older thread - https://lists.launchpad.net/kicad-developers/msg21732.html

Here Wayne sent the gEDA license to the FSF for input. I would prefer to use that. I have emailed him to see if he ever got a response. I'm hoping to short circuit the too-many-cooks of the developers mailing list :)

poeschlr, 2017-09-18:
I think this is what @easyw also mentioned some time back on the forum https://forum.kicad.info/t/3d-new-library-for-mechanical-cad-exporting-and-enclosure-design/1763/27

And this might be the reason we have exactly this license for the 3d model lib.

poeschlr, 2017-09-18:
I found something else:
https://www.mail-archive.com/kicad-developers@lists.launchpad.net/msg16268.html

SchrodingersGat, 2017-09-18:
Good find!

bobc, 2017-09-19:
@poeschlr "I just checked. No it seems the libs are under normal GPL not LGPL.

So the question still stands. Should we make the symbol exception clear in the lib?"

Where exactly did you check? It's quite clearly LGPL here  https://github.com/KiCad/kicad-library/blob/master/COPYRIGHT.txt#L1

Personally, I think the requirements to share and attribute are impractical, and we are never going to be enforce it, a BSD/MIT license would make more sense. But I also think engineering data files are not covered by copyright anyway so the whole question is rather moot.

poeschlr, 2017-09-19:
We would leave the Library GPL in place. But add a clear statement that Projects created using the files here are not required to use the same license.

And yes i learned in the mean time that the Library GPL is a precoursor for the Lesser GPL. (It is called lesser since the 2.1 release of the lib.)

poeschlr, 2017-09-19:
An idea about license change:
We could do it similarly to what we did with the 3d models.
Make a new lib under a new license and ask new contributions to be transfered there. Old stuff needs to be transfered slowly which would mean for some time kicad would still ship the old lib under the current license but we can offer a reduced set of symbols/footprints under a better license.

I think a lot of footprints can be transfered quite fast because they are script generated. And the scripts are written by me, oliver and @pointhi. Which limits the amount of people needing to agree on a license change.

This way we would be safe from problems regarding re licensing of old stuff.

jkriege2, 2017-09-19:
I think that way we'll loose a lot of footprints, becaus in principle we would have to ask every contributor (of a single footprint) whether he's OK with changing the license ... that does not really seem viable ...

Best,
JAN

PS: Not sure how to do this properly or correctly without loosing contributions ... In general maybe we can simply ASSUME that most contributors didn't really think about licensing and just published, assuming that there is an open source license in place which "makes sense" for the typical KiCAD usecases (i.e. commercial + non-commercial) ... then we could just put the footprints under a new license without asking ... but legally that's for sure AT LEAST sketchy

poeschlr, 2017-09-21:
It looks like the footprints currently do not have any license assigned. (None of them have a license* or copyright* file) So i think we might get lucky here. Leaves only the symbols as a problematic area. (Here the lgpl normally would not permit re licensing without the consent of all contributors.)

SchrodingersGat, 2017-09-21:
See https://github.com/KiCad/kicad-library/issues/1645

Shackmeister, 2017-12-03:
@SchrodingersGat can this be closed now?

SchrodingersGat, 2017-12-03:
Yep this has now been resolved.

closed 2017-12-03 by SchrodingersGat