KiCad/kicad-library #1271: MAX31855 thermocouple-to-digital converter: duplicate symbols

hackscribble, 2017-05-30:
`sensors.lib` has MAX31855 ...

[image]

`interface.lib` has MAX31855KASA plus aliases for other suffixes ...

[image]

If the librarians think this should be rationalised, I'm happy to raise a PR.  Also noticed that the MAX31820 temperature sensor (plus aliases) is in `maxim.lib` not `sensor.lib`.

jkriege2, 2017-06-10:
If you have time: Go ahead and submit a PR with the changes. I would suggest using the symbol from interface.lib, but maybe moving the footprint field to the bottom right, besides the GND-pin. The symbol in sensors.lib can then be removed (it's more like an interface, as it connects toa thermocouple but does not contain one itself). You could also move those temp-sensors from maxim.lib to sensors.lib.

Best & Thanks,
JAN

hackscribble, 2017-06-10:
Thanks @jkriege2 I'll work on it this week.

hackscribble, 2017-09-05:
Closed by https://github.com/KiCad/kicad-library/pull/1595

closed 2017-09-05 by hackscribble