KiCad/kicad-library #1287: KLC-modifications for graphical symbols/power-symbols

jkriege2, 2017-06-05:
Dear all (and @SchrodingersGat ):

as discussed earlier, I'm proposing here some changes to KLC to allow for graphical and power-flag symbols. This PR is accompagniied by changes to the KLC-chech-scripts: 

## graphical symbols (no pins, hidden REF, not in BOM)
These require:
- hidden REFDES
- REFDES with leading '#'
- may have no FPFilter and footprint
- may have hidden VALUE field, which causes the symbol name to have a leading tilde '~' (this is not stated explicitly in KLC, as it is an internal "Hack" of KiCAD, but the scripts should know about this!)

## power-flag symbols (1 pin, hidden REF, not in BOM, visible value)
These require:
- hidden REFDES is always `#PWR`
- may have no FPFilter and footprint
- VALUE field is visible
- may have only one pin

## required changes:

KLC 4.9.i should be changed to:
> The Reference field contains the appropriate Reference Designator (REFDES) and is _visible_. For power-flag symbols (e.g. `+5V`, `GND`, ...), the reference designator is always `#PWR`. For graphical symbols (no pins, not part of the BOM) the REFDES is _invisible_ and arbitrary, but starts with a hash-character `#` (e.g. `#SYM` or `#LOGO`).

KLC 4.9.ii should be changed to:
> The Value field contains the name of the symbol and is _visible_. For graphical (no pins, not part of the BOM) and power-flag symbols the value field may also be _invisible_.

please add to KLC 4.11:
> 4.11.vi For power-flag and graphical symbols, no footprint filter is required and the footprint field should be empty

add KLC 4.13:
> 4.13 Power-flag symbols
>  4.13.i power-flag symbols have a reference designator `#PWR` (cf. KLC 4.9.i)
>  4.13.ii power-flag symbols have exactly one pin of a POWER-type, which is _invisible_
>  4.13.iii power-flag symbols may have invisible value-fields (e.g. for Earth-symbols that are uniquely recognizeable by their symbol shape) (cf. KLC 4.9.ii)

Best,
JAN

SchrodingersGat, 2017-06-05:
Thanks Jan these are good changes. I have adjusted KLC accordingly.

poeschlr, 2017-06-05:
> Contain exactly one pin, which is invisible, and must be either Power Input or Power Output

Should we make it clear that only power input should be used for symbols that are intended to function as global labels? (I would even go so far to only allow invisible power input and make an exception for the power_flag. There is only one such symbol needed in the library.)

SchrodingersGat, 2017-06-05:
@poeschlr you are right, I have now corrected this.

jkriege2, 2017-06-06:
OK for me ... I adapted the scripts in my other PR to adapt to these rules.

JAN

closed 2017-06-06 by jkriege2