KiCad/kicad-library #1295: Prioritize library placement by manufacturer or component type?

brandoncurtis, 2017-06-06:
I'd like to add a bunch of Microchip parts, mainly digital potentiometers and digital-to-analog converters.  I see that there's a Microchip library, but there's also an `adc-dac` library and an `interfaces` library that contains a bunch of digital potentiometers.

The Kicad Library Convention document doesn't indicate how to prioritize library selection when multiple libraries could fit.  Where should I put these parts?

brandoncurtis, 2017-06-06:
According to response in #1296, component type should be prioritized over manufacturer.  I will prepare a pull request on the contents of the [KLC document](https://github.com/KiCad/kicad-library/wiki/Kicad-Library-Convention).

closed 2017-06-06 by brandoncurtis

evanshultz, 2017-06-06:
FYI, while the original topic of discussion has been diverted, I stated an issue on this topic at #1082.