KiCad/kicad-library #1328: Footprint for PLCC4 is incorrect

CountMurphy, 2017-06-21:
The footprint for PLCC4 (a ~2mm square LED) is showing up in Kicad as a 5mm square (actual footprint 5050).  See these [pictures (imgur)](https://imgur.com/a/hpBEX) for details

jkriege2, 2017-06-21:
Hi @CountMurphy !

please have a look at the datasheet linked in that footprint: http://www.cree.com/~/media/Files/Cree/LED-Components-and-Modules/HB/Data-Sheets/CLMVBFKA.pdf ... The footprint matches the package described therein. Also this is NOT A GENERAL PLCC4 package, but one for CREE, who seem to define the package dimensions differently (that's why the manufacturer name CREE is in the footprint name!).

Maybe `LED_WS2812B-PLCC4` fits your needs better ... but also that one is manufacturer-speific (even part-specific, note the part number in the symbol name). To me it seems there is currently no general PLCC4 with the dimensions you need ... Maybe you could provide one?

Best,
JAN

CountMurphy, 2017-06-21:
I just read the data sheet and it proves my point. The datasheet says "Size (mm):2.0x 2.0 "  and "The compact 2mm x 2mm package allows for a very high resolution  screen." However the footprint itself in kicad is 5mm (this is incorrect). In fact every PLCC4 I screen captured from digikey was a cree LED, further proving the footprint should be 2mm, not 5.  To clarify I'm not looking for a 5mm PLCC4, I'm looking to actually use a real PLCC4 in my board.

jkriege2, 2017-06-21:
Oh ... mixed that up, sorry ... would you provide the correct FP?

Best,
JAN

suzizecat, 2017-06-21:
if @CountMurphy isn't working on it, i can provide it.

CountMurphy, 2017-06-21:
I can certainly try to make the correct footprint, though I wont be able to get started on it for few weeks (currently rushing to finish various hardware projects for a wedding near the 4th). If there is pressing need for it now, or if @suzizecat simply wants to work on it, I say go for it. If not, I can work on it once I get some free time again.

ppelleti, 2017-10-04:
I independently discovered and fixed this bug [a few days ago](https://github.com/KiCad/LEDs.pretty/pull/46).

SchrodingersGat, 2017-10-04:
@CountMurphy has this been fixed to your satisfaction?

CountMurphy, 2017-10-04:
@SchrodingersGat I'll take a look at it later tonight.

Shackmeister, 2017-12-03:
@countmurphy can you close this issue if the issue has been fixed?

CountMurphy, 2017-12-04:
confirmed fixed. Thanks everyone

closed 2017-12-04 by CountMurphy