KiCad/kicad-library #1329: Polarity indication silk on LED footprints trimmed by pad clearance

PhirePhly, 2017-06-21:
Looking at the LEDs:LED_0402 through LEDs:LED_1206 footprints, it appears the default KiCAD pad clearance (0.2mm?) is large enough that any indication of which side of these LEDs is the anode/cathode is lost when the board is exported, with the possible exception of LEDs:LED_0606.

Possible solutions include:
1. Changing the default global pad clearance to something less than 0.2mm
2. Changing the pad clearances on these LED footprints from "use global" to something specific which doesn't trim the silkscreen.
3. Somehow indicate which pad is the cathode with silk outside of the default pad clearance area.

[image]

SchrodingersGat, 2017-06-21:
@PhirePhly for the offending footprints, the silkscreen should be expanded such that it does not overlap the default soldermask clearance. Are you able to make this change?

PhirePhly, 2017-06-21:
How concerned are we about violating the courtyard? I think the mask clearance goes pretty much all the way out to the courtyard.

Do we know why the default clearance is so large? I'm used to using something like 3thou instead of the 8thou in KiCAD. Are the cheaper fabs just that bad at registration?

SchrodingersGat, 2017-06-22:
In that case you could just change the global mask clearance values for your project - that would fix this without any change to the library, correct?

PhirePhly, 2017-06-22:
Well yes, and I've certainly already done that so I could enjoy other advantages like solder mask between pads on my ICs, but if my concern was only fixing it on my projects I wouldn't have opened an issue ticket, I would have just made my own footprint.

SchrodingersGat, 2017-06-22:
> if my concern was only fixing it on my projects I wouldn't have opened an issue ticket, I would have just made my own footprint.

Appreciated! 

I am not sure where the 0.2mm default was decided. It does seem very conservative. 

Given that this is the default, however, the silkscreen for these should be expanded so it is at least a bit visible.

evanshultz, 2017-06-23:
@PhirePhly Would you mind updating the footprints and submitting a PR? If not, that's OK and we still appreciate you bringing this to our attention.

PhirePhly, 2017-06-28:
I'm researching the issue and industry standards to try and propose an informed solution, barring someone else jumping in and solving the immediate issue at hand here.

poeschlr, 2017-07-25:
For now we have decided to have a 0.2mm clearance between pads and silk screen. I think this should be ok for most manufacturers. If you can find some standard for this it would be great if you could give us this information.

Shackmeister, 2017-12-03:
This has been corrected in the new footprint repo

SchrodingersGat, 2017-12-03:
Thanks @Shackmeister

closed 2017-12-03 by SchrodingersGat