KiCad/kicad-library #1355: Footprint pad numbering

SchrodingersGat, 2017-06-26:
Many footprints require non-standard naming for certain pins, beyond 1,2,3,4,...,n.

Examples:

* Packages with thermal pads
* Connectors with shield pins
* Multi-throw switches with common terminals

Currently this is handled by adding an extra number to the pads.

e.g. A 48-pin housing with an E-PAD connection has pin `49` added.

Should we develop a standard nomenclature for these extra pads, as we can use more than just numerical identifiers for pads?

Examples:

1. E-Pad connections could be marked `EP`, `EP1`, `EP2` etc
1. Shield / GND pins could be labelled `SHLD`, `GND`, etc.
1. Common pins on switches could be labelled `C` or `COM`

This will bring a greater level of consistency to the library. However we would need to develop a scheme that is extensible. 

a) Does anyone have any information on whether this level of naming is specified under IPC standards? 
b) Comments? Objections?

@jkriege2 @poeschlr @pointhi  @easyw @evanshultz et al :)

SchrodingersGat, 2017-06-26:
Another key example is the `TAB` pin on many SMT devices

rivimey, 2017-07-06:
If I may comment, I would prefer that non-standard pins were obviously so; for an exposed pad it seems perfectly sensible to call it EP  and ditto for tabs, etc. The line, for me, would be for 'pins' that are more than a specialised ground/chassis... would you agree?

poeschlr, 2017-07-26:
I don't think it is a good idea to use ep as pin number for the exposed pad. A lot of datasheets use numbers. (And we currently use a number, changing it would require us to change all affected symbols and footprints in one go. We then have the added problem that a lot of people get up do date footprints via the github plugin but don't get the new symbols.)

I would only use non numerical pin numbers for specialized footprints that are used for only one symbol. (example are relays or switches) In short: non numerical pin numbers only for atomic parts.

closed 2017-11-16 by SchrodingersGat