KiCad/kicad-library #1358: More deprecated libraries to remove

evanshultz, 2017-06-26:
Along with https://github.com/KiCad/kicad-library/pull/1349...

I believe these can be safely removed:
https://github.com/KiCad/EuroBoard_Outline.pretty
https://github.com/KiCad/Filters_HF_Coils_NEOSID.pretty
https://github.com/KiCad/NF-Transformers_ETAL.pretty
https://github.com/KiCad/TerminalBlocks_Phoenix.pretty
https://github.com/KiCad/TerminalBlocks_WAGO.pretty
The various https://github.com/KiCad/Transformers_CHK.pretty
https://github.com/KiCad/NF-Transformers_ETAL.pretty
https://github.com/KiCad/Transformers_SMPS_ThroughHole.pretty

https://github.com/KiCad/Sockets_MOLEX_KK-System.pretty has parts that appear to be the same as the KK 6410 parts at https://github.com/KiCad/Connectors_Molex.pretty. It seems that the same physical parts are covered in the KiCad library, but the deprecated library is an older footprint version. Can someone please check me on that?

https://github.com/KiCad/Oddities.pretty has a few footprints for net shorting that may be useful. Go ahead and close this too if you want, but I'd like to submit a PR for matching schematic symbols and find a nice place for these footprints (after appropriate updating to meet KLC).

Lastly, the connectors/pin headers/terminals/etc. seem overly confusing. Perhaps there's a better way to organize all of those similar parts?

evanshultz, 2017-06-26:
@SchrodingersGat  @poeschlr @jkriege2 
Forgot to alert you fine folks.

poeschlr, 2017-06-26:
I'm not sure deleting repos is a good idea.
Because of the current setup of the lib system cvpcb crashes for all users!
https://forum.kicad.info/t/cvpcb-crashes-on-deprecated-footprint-library-files/6872/2

It might be a better idea to just remove them from the fp-lib-tables and let them slowly die. (They can be removed a few releases in the future.)
I would even suggest to recreate the 3 libs that have already been deleted. (maybe only have one dummy footprint in them that is call this_lib_is_deprecated_and_will_be_deleted_soon)

evanshultz, 2017-06-26:
Uh oh. See the PR I linked above where Olivier only removed them from the fp-lib-tables. That was only my intent, to continue that work. Actually deleting the repo was not my intent, though I see my wording was confusing if one wasn't thinking about continuation from the linked PR.

poeschlr, 2017-06-26:
Yea but these 3 repos have already been deleted. (This is why i brought it up. Just wanted to make sure it will not happen again.)
I will open a second issue about that specific problem.

SchrodingersGat, 2017-06-27:
@evanshultz the `TerminalBlocks_xxx` libraries are actually new ones and shouldn't be removed

evanshultz, 2017-06-27:
Edit for clarify and removed the `TerminalBlocks_xxx` from the list.

evanshultz, 2017-06-27:
@SchrodingersGat The `TerminalBlocks_xxx` libraries have nothing in them, which was why I originally thought they should be removed. Is there a collection of footprints that are intended to go there so those libraries aren't empty?

SchrodingersGat, 2017-06-27:
@evanshultz they were created recently to reduce the library size of some of the connectors libraries. However there have been no additions yet.

evanshultz, 2017-08-24:
Only https://github.com/KiCad/EuroBoard_Outline.pretty still exists in the current fp-lib-table, and there are relatively recent commits there. (And talk of converting them into templates.)

closed 2017-08-24 by evanshultz