KiCad/kicad-library #1362: missing USB3-MicroB connector.

davygrvy, 2017-06-27:
The Micro-B has a different pinout with pin 4 being the OTG line.

davygrvy, 2017-06-27:
[usb3_microb.zip](https://github.com/KiCad/kicad-library/files/1106643/usb3_microb.zip)

evanshultz, 2017-06-27:
Hi @davygrvy!

I believe the libraries to have this covered. See the USB* symbols at https://github.com/KiCad/kicad-library/blob/master/library/conn.lib.

Unfortunately, symbol libraries aren't immediately updated (the footprint libraries are) and you need to wait for a new release of KiCad. Or you can individually update your symbol libraries.

Please take a look to see if the symbol you want is included in the latest library already.

davygrvy, 2017-06-27:
[image]

It's missing, sorry.  Thanks for the link

evanshultz, 2017-06-27:
Alright. Since it seems you've already whipped up a symbol, please fork the library and submit your symbol with a pull request. Then the librarians can review the symbol and, once ready, add to the KiCad library. Thanks again for bringing this up and helping to improve KiCad!

davygrvy, 2017-06-29:
My git chops are icky.  The connector graphic should be changed to represent the microB connector from the type B it currently is and I can't seem to get the library editor to do what I want for that.  Also, a 3.0 Micro AB is missing, too.  If someone could fix these, that'd be great.

davygrvy, 2017-06-29:
Both of these are missing [image]

poeschlr, 2017-06-30:
If symbols are missing, create them and if you want create a pull request to share them with other users. (Be the change you want to see.)

>  The connector graphic should be changed to represent the microB connector from the type B it currently is and I can't seem to get the library editor to do what I want for that.

What do you mean by that? maybe provide screenshots to show what you want to do. (And what part of it does not work.)

suzizecat, 2017-06-30:
@davygrvy you will find out how to contribute with some explanations here : https://opensource.guide/how-to-contribute/

poeschlr, 2018-01-20:
As this is about not yet contributed stuff and we already transferred the lib over to the v5 repo https://github.com/KiCad/kicad-symbols/, i will close this. If anybody involved in this discussion feels this needs to be reopened, feel free to reopen it on the v5 repo

closed 2018-01-20 by poeschlr