KiCad/kicad-library #1364: Socket strips and pin_header first pin in 2x version are inconsistent

suzizecat, 2017-06-29:
Hi,

I found out that the first pin of all `Socket_Strips` are made to have pin 1 in the top left corner when the footprint is horizontal when all `Pin_Headers` got it when the footprint is vertical.

So, as it's displayed, the `Socket_Strips` aren't KLC compliants but there is a problem for the user when you need a male header horizontally : The pin 1 isn't at the right corner (and vice-versa for a vertical female header)

But, since there is a ton of these connector, i don't want to fix that in a PR 😅  (and that's useful to have a component with pin 1 in top-left corner horizontally)

Shackmeister, 2017-06-29:
All of these are scripted and Im currently overhauling the male part https://github.com/KiCad/Pin_Headers.pretty/pull/24 and might have a look at these afterwards :) but since they are scripted you should never change them manually, would be a waste of time :)

jkriege2, 2017-06-29:
They are oriented this way so you can easily place pairs ... but as said: I scripted them, so a rotation could easily be fixed. Note thet the double-row-headers are made, so the pin-numbers match those of the male connectors, when connected to one of theose with pin1-pin1-match!

JAN

suzizecat, 2017-06-29:
I thought about the fact that is may be scripted, i wouldn't do it myself 😅 
@jkriege2  Oh ! So that's why it's inverted that's a nice feature !

closed 2017-07-04 by suzizecat