KiCad/kicad-library #1378: Repository for PCB antennas

aliteke79, 2017-07-03:
Hi guys, i have a PCB antenna footprint to share. Can you tell me the right repository name to contribute?

SchrodingersGat, 2017-07-03:
Hi @aliteke79 we currently have the [RF Modules](https://github.com/KiCad/RF_Modules.pretty) repository which is for RF front-end modules. However this is probably not appropriate for antennae.

Do you have a link to the particular datasheet for the antenna? This might influence our decision.

Perhaps a new repository named `RF_Antenna.pretty` would be acceptable? @jkriege2 @poeschlr any thoughts on this?

jkriege2, 2017-07-03:
I would vote for such a repo ...
JAN

pointhi, 2017-07-03:
I would also vote for such a repo.

Any thoughts about printed pcb-antennas, which would only consist of some copper? Would supporting such footprints in the library make sense? (they are to some extend dependent on pcb properties)

aliteke79, 2017-07-03:
yes, http://www.ti.com/lit/an/swra117d/swra117d.pdf

aliteke79, 2017-07-03:
[image]

suzizecat, 2017-07-03:
@pointhi Since it could be quite painful to make, i'll vote to support them as a standard footprint.
BTW i already have an antenna into RF_Modules but i vote for an antennae specific repo

SchrodingersGat, 2017-07-03:
Such items can be considered footprints, I'd support adding them.

SchrodingersGat, 2017-07-03:
https://github.com/KiCad/RF_Antenna.pretty

SchrodingersGat, 2017-07-03:
PR for lib-table entries here - https://github.com/KiCad/kicad-library/pull/1379

hackscribble, 2017-07-03:
Most (all?) of the other footprint repos have plural names (e.g. `Housings_xxx`, `Connectors_xxx`).  Maybe this new one should be `RF_Antennas` for consistency?

aliteke79, 2017-07-03:
Thank you all, i'm going to fork and add it

SchrodingersGat, 2017-07-03:
Thanks for raising the issue @aliteke79

closed 2017-07-03 by SchrodingersGat