KiCad/kicad-library #1385: Allwinner BGA parts

mkschreder, 2017-07-04:
Does anybody know where I can find schematic symbols and footprints for Allwinner SoCs? H3, A10, A20, A13 in particular..

poeschlr, 2017-07-04:
If a google search does not show anything maybe ask over at the kicad forum. If no user has such symbols you might need to design them your self.
(We would like to see a pull request that adds such parts.)

suzizecat, 2017-07-04:
I don't think that there are in KiCAD, but you can find them by Googling it

In example : `Allwinner H5 datasheet` first result will give you the datasheet. Then, look in the table of content for "packaging" stuff  (here in the appendix, page 68.)

This footprint looks quite annoying to make, but it isn't pretty hard (just a little long...) 

(I made a mistake and used H5, but H3 will give you the exact same result)

Have fun 😅 

Best, Julien

poeschlr, 2017-08-23:
Closing this because the question of the original poster was answered. (And no further question was asked.)

closed 2017-08-23 by poeschlr