KiCad/kicad-library #1397: How to make a "multi-part" footprint ?

suzizecat, 2017-07-07:
Hi all !

I need to make the footprint for a LoRA PCB antenna. I found the  [http://www.taoglas.com/wp-content/uploads/2016/07/ILA.09.pdf](Taoglass ILA.09 (datasheet)) but i run into a difficulty...

At the end, the antenna should be linked with two wires : one to the ground, the other to the transceiver.
The problem is that these two wires should link to the same pad, which isn't possible on KiCAD

Since the component seems to be polarized, should i make a "sneaky pad" with two different pads which would be side by side ? I'm pretty sure the DRC system won't like that at all... 

Plus, there is some others components which positions looks pretty well-defined within the keepout area. Should i include pads for these components or left to the user the responsibility to add them later ? The problem is that i don't think its possible to make a copper wire properly within a footprint...

Thanks for your help !
Julien

SchrodingersGat, 2017-07-07:
I'm not quite sure what you mean, but can you draw directly onto the copper layer instead of making a pad?

To achieve this, you will have to:

1. Draw the shape on a different layer (e.g. F.SilkS)
1. Edit the file manually and change the layer of the drawing to F.Cu
1. Don't edit that shape again in the KiCad editor
1. Attach the two actual pads at the end of this "wire"

If you do it this way, I **think** that the DRC won't freak out. (But you will have to test this)

suzizecat, 2017-07-08:
Wow ! That's quite a trick 😅 I'll try that sunday or during the week

evanshultz, 2017-08-24:
@suzizecat Can this case be closed?

suzizecat, 2017-08-25:
It worked fine, thanks

closed 2017-08-25 by suzizecat