KiCad/kicad-library #1402: RFC - Rearranging Symbol Libraries

SchrodingersGat, 2017-07-10:
Following on from the work by @jkriege2 [here](https://github.com/KiCad/kicad-library/issues/1398#issuecomment-313969813)

| Library Name | Description | Previous Name | Notes |
| ------------------ | --------------- | --------------------- | ------------ |
| **Basic Devices** 
| devices.lib | Symbols for simple devices e.g. Resistors / Capacitors / etc. | device.lib | |
| elec-unifil.lib | ??? | | | 
| leds.lib | LED | device.lib |
| motors.lib | Eletric motors | | |
| power.lib | Power symbols (global) | | special symbols library |
| pspice.lib | Spice symbols | | |
| switches.lib | Buttons and switches | | | 
| **Connectors** |
| connectors.lib | Connectors, general purpose | conn.lib | Could be expaned in future e.g. *connectors_usb.lib* / *connectors_JST.lib* |
| **Virtual** |
| graphical_symbols.lib | | | |
| logo.lib | | |
| mechanical.lib | | | 
| **Semiconductors (simple)** |
| diodes.lib | Diodes | | |
| esd_protection.lib | ESD protection devices | ESD_Protection.lib | |
| transformers.lib | Transformers | | |
| transistors.lib | Transistors | | |
| triac.lib | Triacs, thyristors | | |
| valves.lib | Valves | | |
| **Displays** |
| displays_lcd.lib | LCD displays | display.lib | |
| displays_led.lib | LED displays | display.lib | |
| **Opto** |
| opto_isolators.lib | Optocouplers / optoisolators | opto.lib |
| **Sensors** |
| sensors.lib | Sensors that do not fit other categories | sensors.lib |
| sensors_multiple.lib | Sensors that sense multiple things e.g. temperature / pressure | sensors.lib |
| sensors_current.lib | Current shunts / transducers / etc | sensors.lib |
| sensors_magnetic.lib | Hall effect, etc | sensors.lib |
| sensors_motion.lib | Accelerometers / gyro | sensors.lib |
| sensors_photo.lib | Light sensors, RGB, photoresistors | sensors.lib |
| sensors_pressure.lib | | sensors.lib |
| sensors_temperature.lib | | sensors.lib |
| **Audio / Video** |
| audio_amplifiers.lib | Audio amplifiers | audio.lib / digital_audio.lib |
| audio_adc_adc.lib | | digital_audio.lib |
| video.lib |
| **Linear Devices** |
| amplifiers.lib | Amplifiers | linear.lib | These could be further split by function e.g. 'instrumentation', 'transconductance', 'current_sense' |
| comparators.lib | Comparators | linear.lib |
| **References** ||
| references_current.lib | Precise current reference | references.lib |
| references_voltage.lib | Precise voltage reference | references.lib |
| **Power Conversion** |
| regulators_linear.lib | Linear regs | regul.lib |
| regulators_switching.lib | Switching regs | dc-dc.lib | These could be broken out further e.g. internal / external FET |
| ac-dc_modules.lib | Modular ac-dc devices | ac-dc.lib |
| dc-dc_modules.lib | Modular dc-dc devices | dc-dc.lib |
| **Mixed Signal** |
| analog_switches.lib | Analog switches |
| adc.lib | Analog -> Digital | adc-dac.lib |
| dac.lib | Digital -> Analog | adc-dac.lib |
| adc-dac.lib | Devices with both ADC and DAC |
| **Drivers** |
| drivers_led.lib | LED drivers |
| drivers_gate.lib | FET gate drivers |
| drivers_motor.lib | Motor control ICs |
| **Timers** |
| oscillators.lib | | Oscillators.lib |
| timers.lib | | linear.lib | e.g. NE555 |
| plls.lib | | linear.lib | e.g. NE567 |
| rtc.lib | Real time clock | |
| **Logic** |
| logic_cmos40xx.lib | 4000 series logic | cmos4000.lib / cmos_iee.lib | Previously split into IEC / IEEE variants. New library format could allow different graphical representation for each symbol, so users can choose which symbol they want |
| logic_74xx.lib | 74 series logic | |
| **CPLD / FPGA** |
| cpld_xilinx.lib | | xilinx.lib |
| cpld_altera_intel.lib | | Altera.lib |
| fpga_xilinx.lib | | xilinx.lib |
| fpga_altera_intel.lib | | Altera.lib |
| fpga_actel.lib | | actel.lib |
| fpga_atmel.lib | | atmel.lib |
| **Microcontrollers / Processors / DSP** |
| microcontrollers_8048.lib | | |
| microcontrollers_8051.lib | | |
| microcontrollers_atmel_avr | Atmel AVR series | atmel.lib |
| microcontrollers_atmel_avr32 | Atmel 32-bit AVR series | atmel.lib |
| microcontrollers_atmel_arm.lib | Atmel ARM micro | atmel.lib |
| microcontrollers_cypress.lib | | cypress.lib |
| microcontrollers_cypress_arm.lib | | cypress.lib |
| microcontrollers_microchip_pic10 | PIC10 series | microchip_pic10mcu.lib |
| microcontrollers_microchip_pic12 | PIC12 series | microchip_pic12mcu.lib |
| microcontrollers_microchip_pic16 | PIC16 series | microchip_pic16mcu.lib |
| microcontrollers_microchip_pic18 | PIC18 series | microchip_pic18mcu.lib |
| microcontrollers_microchip_pic24 | PIC24 series | microchip_pic24mcu.lib |
| microcontrollers_microchip_pic32 | PIC32 series | microchip_pic32mcu.lib |
| microcontrollers_microchip_dspic33 | DSPIC33 series | microchip_dspic33dsc.lib |
| microcontrollers_freescale_hc.lib | Freescale HC series | |
| microcontrollers_freescale_coldfire.lib | Freescale coldfire | motorola.lib |
| microcontrollers_infineon_c166.lib | | |
| microcontrollers_nxp_arm.lib | NXP microcontrollers | | nxp_armmcu.lib |
| microcontrollers_parallax.lib | | |
| microcontrollers_renesas.lib | |
| microcontrollers_st_stm32.lib | STM32 micro | stm32.lib |
| microcontrollers_st_stm8.lib | STM8 micro | stm8.lib |
| microcontrollers_texas_msp340.lib | MSP340 micro | msp430.lib |
| processors_intel_x86.lib | | |
| processors_amd_x86.lib | | |
| processors_motorola_68000.lib | | motorola.lib |
| processors_motorola_powerpc.lib | | motorola.lib |
| processors_zilog_z80.lib | | Zilog.lib |
| dsp_texas.lib | Texas DSP (digital signal processors) | dsp.lib |
| dsp_freescale.lib | Freescale DSP | dsp.lib |
| soc_texas_arm.lib | Texas ARM SOC (system on chip) |
| **Micro / DSP (Generic)** |
| microcontrollers.lib | Micros that don't fit anywhere else |
| processors.lib | Processors that don't fit anywhere else |
| dsp.lib | DSP chips that don't fit anywhere else |
| soc.lib | SOC chips that don't fit anywhere else |
| **Memory** |
| memory_vram.lib | Volatile RAM (DRAM / SRAM) | memory.lib |
| memory_nvram.lib | Nonvolatile RAM (MRAM / FRAM) | memory.lib |
| memory_eeprom.lib | EPROM / EEPROM | memory.lib |
| memory_flash.lib | FLASH memory | memory.lib |
| **Interface (Comms)** |
| interface_bluetooth.lib | Bluetooth | rfcom.lib |
| interface_can_lin.lib | CAN / LIN | interface.lib |
| interface_ethernet.lib | Ethernet | interface.lib |
| interface_lvds.lib | LVDS | interface.lib |
| interface_rf.lib | RF comms that don't fit elsewhere | rfcomm.lib |
| interface_rs232.lib | RS232 interface | interface.lib |
| interface_rs422.lib | RS422 interface | interface.lib |
| interface_rs485.lib | RS485 interface | interface.lib |
| interface_usb.lib | USB | 
| interface_wifi.lib | WiFi | rfcom.lib |
| interface_zigbee.lib | Zigbee / XBee | rfcom.lib |

-----------------------------------------

## Unnassigned

The following libraries are yet to be assigned.

- periphery.lib
- periphery_portexpanders.lib
- periphery_timer.lib
- periphery_rtc.lib
- periphery_uart.lib
- periphery_video.lib
- periphery_network.lib
- bus_i2c_smbus.lib for I2C drivers, isolators, ...
- bus_usb.lib for USB hubs, drivers, ...

- modules.lib

- ic_misc.lib (formerly powerint.lib)

Left-overs, where the contents should be resorted ...

allegro

- .lib
- analog_devices.lib
- bbd.lib
- bosch.lib
- brooktre.lib
- contrib.lib
- ftdi.lib
- gennum.lib
- graphic.lib
- infineon.lib
- intel.lib
- intersil.lib
- ir.lib (moved to sensors_current.lib and gate_drivers.lib)
- LEM.lib
- maxim.lib
- microchip.lib
- nordicsemi.lib
- nxp.lib
- onsemi.lib
- philips.lib
- powerint.lib (moved to smps_controller.lib and ic_misc.lib)
- RFSolutions.lib
- silabs.lib
- siliconi.lib
- supertex.lib
- texas.lib
- wiznet.lib
- Worldsemi.lib
- Xicor.lib
- zetex.lib
-

SchrodingersGat, 2017-07-10:
Some notes on the (currently unassigned) libs above

1. I don't like the word "periphery" - I don't think it is technical enough
1. The remaining unassigned libs are mostly manufacturer specific. We will have to go through them and decide what functions to assign.

SchrodingersGat, 2017-07-10:
Rather than making new tables below, please make any adjustments (after discussion) to the table in the comment at the top.

SchrodingersGat, 2017-07-10:
## Plurals

Plural naming does not work well for everything. However I think that non-plural naming can work in all cases. Should we enforce a non-plural naming scheme is anyone really passionate about plural names?

Examples of plurals that just sound wrong

* memories_flash.lib  _(memory_flash.lib)_
* interfaces_ethernet.lib _(interface_ethernet.lib)_
* dacs.lib _(dac.lib)_

I propose we name *function_extra_sub*, otherwise we get a weird combo of *function**s**_extra_sub*, _function_extra_sub**s**_

evanshultz, 2017-07-10:
powerint.lib has most parts that are internal FET regulators, and also a few line discharging parts that I think go into `ic_misc.lib`.

You need `audio_misc.lib` or something like that for SRCs, mic preamps, etc.

Where do CODECs go? I assume `audio_adc_dac.lib`, but what else would go there? Why not call it `audio_codec.lib`?

More libraries are needed for SMPSs more. Break out DC-DC internal FET (regulators) from DC-DC external FET (controllers). We already have a lot of each. And generic SMPS controllers like UC3842 and NCP1200 deserve their own library. They're a different ballgame than integrated DC-DC parts. Lastly, where do PFC controllers go? They're a totally different type of IC (unless you're talking about single-stage flyback or oddball topologies like the Clark converter of PFC buck) and I think they should be in their own library.

Lastly, transformers are a total wasteland. I'd like to find a paradigm that works. I've tried 3 categorization schemes and none of them worked well for me. Does anybody have anything that works? It may take multiple libraries to cover AC line frequency, SMPS, gate drive, ethernet, and all other kinds of magnetics. Additionally, do coupled inductors go here or elsewhere? What about common mode chokes? Do those have their own libraries?

jkriege2, 2017-07-10:
Some more  comments:
- singular is fine for me
- `audio_codec.lib` is fine with me
- if we split DC-DC, then we should also have a separate lib for switched-cpacitor voltage converters (e.g. LTC660) ... name could be `regulators_switching_capacitance.lib`
- For simple magnetics like coils we don't need a separate lib (the footprints should simply use the pin-numbers 1 and 2 and they match to the generic symbol L, ...). 
- For common-mode chokes we should have a separate lib ... could we have something like `filters.lib` that could also contain other "filtering" components (most of them again fit the generic symbol, but some just might be different)??? Not sure about that
- For transformers also not sure: maybe `transformer_power.lib` (for larger ones, 220V->xyV) and `transformers_small_signal.lib` or `transformers_pulse.lib` (for audio, ethernet, ... is that the correct english name? I translated from the german "Überträger" using Wikipedia ;-)
- several libs were lost I feel (battery_management, relay_drivers.lib transistor_arrays.lib ...) I will try and add them this evening ...

Best,
JAN

hackscribble, 2017-07-10:
+1 for singular.

There were suggested libraries starting `quarz...` that I don't see in the table above.  Would these better as `piezo...` since the material used might be crystal or ceramic?  Whether it's a filter or a resonator, it is the piezoelectric effect that they have in common.  But if we stick with the suggested name, it should be `quartz...` (I'm being very picky :wink:)

What will be left in `device` if inductors move to a separate library as @jkriege2 suggests above?  Just resistors and capacitors?  If so, why not split into two libraries for clarity?

How about shortening `microcontroller_` to `mcu_`? Anyone using one ought to recognise the acronym.

`peripheral` instead of `periphery`?

SchrodingersGat, 2017-07-10:
1. Alright, singular names it is (I have updated the table accordingly).
1. I also prefer `mcu_` as a prefix (we already have `soc_` and `dsp_`
1. Regulators (switching). Agreed that we need to arrive at some categorizations here
1. Transformers - my knowledge here is lacking
1. `peripheral.lib` - the suggested contents of this library still seem very broad to me. Can we narrow this down?

hackscribble, 2017-07-10:
> piezo / quart / etc - what is the function of this library (not the material it is made of!) Is it oscillators? Resonators? Other?

OK, I see that there have been edits on the other thread that prompted my comment :smile: : https://github.com/KiCad/kicad-library/issues/1398#issuecomment-313926433

That now suggests two libraries:

> oscillators.lib (formerly: Oscillators.lib)
quarzes_filters.lib (formerly: Oscillators.lib)

If `oscillators.lib` is passive piezoelectric oscillating devices (i.e. standalone crystals, crystal resonators with built-in capacitors and ceramic resonators), that's good.  But it should not IMHO include oscillator ICs.

Piezoelectric filters can be made of quartz crystal or ceramic material.  So `quartz_filters.lib` does not sit well with me.

pointhi, 2017-07-10:
After a quick look
* ```audio_adc_adc.lib``` -> ```audio_adc_dac.lib```
* ```sensor_multiple.lib``` -> ```sensor_combined.lib``` is probably a better name. Probably someone has a even better one

@SchrodingersGat probably ```kicad-symbols``` would be a more appropiate name for the repo than ```symbols```

nbxmike, 2017-07-10:
You may as well combine the NXP and Freescale processors to NXP or you will need to do it later.

evanshultz, 2017-07-10:
@pointhi Can you please check you first bullet? It looks like the same lib name is repeated.

Unless someone objects, I volunteer to propose the divesting of the following libs:
- analog_devices
- infineon (which purchased IR)
- intersil
- on semi
-zetex

evanshultz, 2017-07-10:
@SchrodingersGat Is the above a "all edit" table? Or are we making suggestion with one person (you, I assume) as mediator and editor of the table?

evanshultz, 2017-07-10:
Finished with analog_devices.lib, infineon.lib, and zetex.lib.

I like the idea of separate diode libraries IF the symbols are different. For example, regular Si diodes, Schottky/SiC, zener/TVS, etc. This will result in the least amount of unique symbols so it's the easiest to maintain.

diggit, 2017-07-10:
In _my_ opinion, there is no need for symbols of specific symbols for such simple devices. Generic symbols are IMO enough. There is only several permutations with so few pins. Then splitting them does not make sense, they could all got to device.lib.
At least my workflow is:

1. find online which components suits the purpose and is available
2. choose correct generic symbol (it is fast, keywords like: gds, zener, tvs, diode, fet, npn, .... works well)
3. rename symbol to match selected device
4. assign correct FP

I have similar workflow for linear regulators, because in 70% cases, the one I choose to use is not in libs.

SchrodingersGat, 2017-07-10:
> @evanshultz - @pointhi Can you please check you first bullet? It looks like the same lib name is repeated.

Read it again ;) It's a very subtle mistake that I made in the original table, which I have now fixed.

SchrodingersGat, 2017-07-10:
> @SchrodingersGat Is the above a "all edit" table? Or are we making suggestion with one person (you, I assume) as mediator and editor of the table?

It's open to editing, but only after discussion here :) Please add edit comments at the end of the table as I have been doing, if you edit anything!

SchrodingersGat, 2017-07-10:
### Amplifiers

Currently the amplifier organization is pretty rough. What are some broad categorizations for amps?

* amplifier_instrumentation
* amplifier_current_sense
* amplifier_chopper
* amplifier (everything else)

gauravjuvekar, 2017-07-11:
@SchrodingersGat Er, I don't think others can edit the post. That's not how github works.

SchrodingersGat, 2017-07-11:
Sorry @gauravjuvekar you are completely correct - please ignore that.

Update : Make suggestions and I will add them to the table :)

jkriege2, 2017-07-11:
I think you need to be Member or Owner to edit a non-own post.
JAN

gauravjuvekar, 2017-07-11:
Can we create separate `rf_*` libs. I know that currently there aren't many rf components in the libs, but there are a large number of categories in rf out there.

Eg, see https://www.digikey.com/products/rf-wireless/en
I guess rf_{trasmitters, receivers, transceivers} could go in `interface_rf`, but there are a lot of non-interface components like amplifiers, mixers, couplers, active balun ICs, switches, etc.

SchrodingersGat, 2017-07-11:
@gauravjuvekar this is a good idea, I was only just thinking where antennae would go. I agree there is a place for `rf_` libs separate from RF ICs

gauravjuvekar, 2017-07-11:
I think antenna should also be a separate library. It also has a lot of subdivisions (type: patch, chip, wire)

jkriege2, 2017-07-11:
@SchrodingersGat: some comments about amplifiers:
- `amplifier_current_sense` shouldn't/couldn't that go to `sensor_current`?
- I'm missing `amplifier_difference` (for components like these: http://www.ti.com/lit/ds/symlink/ina157.pdf) ... basically OPAMPS, but with (calibrated) resistors inside
- Also maybe `amplifier_power` and/or `amplifier_highvoltage`? (not so sure about these ... but there we could put e.g. opamps that also work at 100V ...)

Just ideas ...

JAN

SchrodingersGat, 2017-07-11:
> amplifier_current_sense shouldn't/couldn't that go to sensor_current?
I think that's fair.

> I'm missing amplifier_difference (for components like these: http://www.ti.com/lit/ds/symlink/ina157.pdf) ... basically OPAMPS, but with (calibrated) resistors inside
> Also maybe amplifier_power [...]

I've added `_difference`, `_power`, and `_rf` but I don't think that `_highvoltage` is a category all on its own...

SchrodingersGat, 2017-07-11:
Added transformer categories.

jkriege2, 2017-07-11:
will people know that `amplifiers.lib` or `ic_amplifiers.lib` contains OPAMPS? ... i.e. are all amplifiers OPAMPS or one of the others ...

JAN

poeschlr, 2017-07-11:
What about (multi purpose) signal level translators and signal isolators? (currently they might go into the logic libs but i think giving them a separate lib might be a good idea.)

The symbols of the LEM lib could go into the sensor_current lib.

evanshultz, 2017-07-11:
> amplifier_current_sense shouldn't/couldn't that go to sensor_current?

I disagree. The sensor is the part that actually senses the current. ICs like we see in `linear.lib` now that are "current sense" parts are simply amplifying an external sensor, like a resistor or transformer/inductor winding. I do not think those ICs belong in `sensor_current.lib`.

I also do not like having so many sub-categories for opamps. Why would you break out chopper-based amps? Their benefit is low input offset voltage, but you can get non-chopper amps with excellent performance here as well. It seems that we'd be splitting up parts which are otherwise interchangeable, cutting users off from seeing all opamp possibilities they might want to choose.

poeschlr, 2017-07-11:
Gennum  is now semtech
the `gennum.lib` would need to be split up into multiple libs. (non of the symbols have a description which makes this a bit harder.)

`graphic.lib` does not exist in the current master branch. It has been renamed to `graphic_symbols.lib` (or has it been deleted and the graphic_symbols lib is a completely new lib?)

`intel.lib` mostly into a new `mcu_intel.lib` but also some into `oscillator`, `interface*`, ...

`intersil.lib` either `driver_gate` or `driver_motor`

`maxim.lib` split up in many parts. (many temp sensors but also a some other stuff.)

`microchip.lib` mostly interface chips. (some gate drivers)

`nordicsemi.lib` only one chip is in there. should go to `interface_bluetooth`

`nxp.lib` some i2c led controllers and i2c multplexers

`onsemi.lib` mostly `esd_protection` but also `transistor`

`philips.lib` mostly interfaces

---

Where should digital potis go?
Where should i2c, spi, ... io expander go?
should h-bridge drivers go into `diver_gate` or `driver_motor`?

nbxmike, 2017-07-11:
At the rate at which sub-categories are being proposed, the library revision better have some kind of search tool across ALL libraries.  

I am no op-amp expert and rely on others to tell me that part of a design, so I would not necessarily know which of the categories to look in for a part number I've been provided.  I doubt I am alone in allowing other people with more specific expertise to dictate portions of my designs, I could see us getting very frustrated with so many finely sliced parts libraries.  Right now, grep is my go to solution for this but I think less command line experienced people would prefer a built in tool.

poeschlr, 2017-07-11:
When placing a part in eeschema the search field gives results in all libs.

poeschlr, 2017-07-11:
The note for connector lib:
> Could be expaned in future e.g. connector_usb.lib / connectos_JST.lib

I don't think generic connectors need a specialized symbol.
The only symbols we could add is for multi row connectors where each row has a separate connector on the wire side. Or for connectors that don't follow a pure numeric numbering scheme.

jkriege2, 2017-07-11:
@poeschlr 
I would say:
- dig. potis in `ic_digpotentiometers` or `digital_potentiometers` ... there are so many that I think they deserve their own lib ... anotherpossibility (although not completely accurate would be `dac.lib`)
- h-bridges I would put into driver_motor?
- for IO-expanders, RTC, ... I proposed the periphery_-tree ... which was not accepted ... as RTCs are in `rtc.lib`, this would call for `io_expanders.lib` ... 

@SchrodingersGat :
- what about timers like 8253, which are specifically designed as µC-periphery ... do they fit in `timer.lib`?

JAN

jkriege2, 2017-07-11:
About OPAMPs: I wouldn't separate out the chopper-amps ... as far as I know they are (to the outside) like OPAMPs (they have In+, In- and VoltageOut .. possibly+control pins) ... but OTA-amplifiers, or transconductance amplifiers have a current output, not a voltage output ... shouldn't they be separated out?

SchrodingersGat, 2017-07-12:
@poeschlr I have applied your suggestions to the table, thanks :)

@jkriege2  I have added `digital_potentiometer.lib` :) I think `driver_motor` contains standalone motor controller ICs, while `driver_gate` contains things like half/full-bridge drivers etc (they're used for more than just motors)

SchrodingersGat, 2017-07-12:
> @jkriege2 what about timers like 8253, which are specifically designed as µC-periphery ... do they fit in timer.lib?

Yes, I'd expect so.

jkriege2, 2017-07-12:
@SchrodingersGat about drivers: Just starting from the lib-name in `driver_gate` I would expect driver-ICs that are used to drive the gate of a MOSFET, not half or full bridge drivers (although both might share the same functioning principle ...) ...

I scanned a bit the webpages of semiconductor manufacturers:

- http://www.ti.com/lsds/ti/analog/motordrivers/overview.page
- https://www.infineon.com/cms/de/product/power/channel.html?channel=db3a304319c6f18c011a154646852706
- http://www.nxp.com/products/automotive-products/smart-power-drivers:SMART-POWER-DRIVERS
- http://www.onsemi.com/PowerSolutions/taxonomy.do?id=210
- http://www.st.com/en/motor-drivers.html

Most of them separate (at least) between:
1. MOSFET/IGBT drivers
2. Motor drivers/controllers (with sub-cats like stepper, bldc, DC, ...)
3. some also have load/relay drivers

I think we should use the same categories, e.g.
1. `driver_motor` (for all motor-related products)
2. `driver_gate` (for ICs that are supposed to drive high-power MOSFETs/IGBTs)
3. `driver_load_other` or `driver_load` for all others
4. maybe we could think about having `motor_controller` for motor-controll ICs like stepper-controllers or BLDC-controllers that do not necessarily contain drivers for the motor.

jkriege2, 2017-07-12:
Also I just stumbled about MCU-companion chips like RESET-ICs and voltage supervisors ... where to put those? `periphery`?

What about the IO-expanders and other periphery that does not fit in any existing lib (yet)?

JAN

SchrodingersGat, 2017-07-12:
Reset ICs should go in `power_supervisor.lib` - this lines up with how they are categorized on [digikey](https://www.digikey.com/products/en/integrated-circuits-ics/pmic-supervisors/691).

I/O expanders (etc) are yet to be allocated. As no better name has been suggested, perhaps a series of `peripheral_` libraries is needed.

jkriege2, 2017-07-12:
+1 on both ;-)

jkriege2, 2017-07-12:
- if IO-expanders are the only ones left ... maybe `interface_parallel.lib` would also be an option? + then we don't have to open up another "tree"
- We could still have a lib `periphery.lib` for everything that does not fit in any other category ...
- Also I think we don't have any place for I2C transceivers/drivers/hubs and SPI switches/transceivers/... maybe `interface_spi.lib` + `interface_i2c.lib`

gauravjuvekar, 2017-07-12:
most of `ftdi` will also go to `interface_usb` or `interface_rs232`

poeschlr, 2017-07-12:
`opto_isolator.lib` might not be the best name. (And opto maybe not the best category)
Opto is more the how is an isolator implemented than a functional description.

Maybe better would be `isolator_opto` and further categories for each type of isolator 'isolator_induction' 'isolator_capacitiv'. 
But i don't like categorizing by internal technology. Yes inductive isolators tend to be used for signals with higher frequencies but there might be opto-isolators out there that can handle similar frequencies to cheap induction based isolators. (or there are specialized induction based isolators for low frequency/ low power applications)

I would not split them up to be honest. maybe just `isolator_digital`? (I don't think there are analog isolators. If they are ever invented or if a contributor finds one we can simply add a new lib called isolator_analog)

Another lib still missing is the `level translator`. 

I personally would place the level translator and the isolator stuff either under logic or under their own signal translation category. (The later might be more future proof.)

poeschlr, 2017-07-12:
Something similar with the display libs.
under `display_lcd` there will be monchrome and color displays. Under display led i think there mitht be led bars, 7 segment, oled displays and in future maybe full color displays (TVs use leds already maybe in future there will not be lcds. also lcd display has a redundant display in it. it should be display_lc).

i would suggest the following function based lib names:
- display_bar (or display_led-bar but this would again be technology based not function based)
- display_segmented (7, 9, 14 and 16 - segment displays) 
- diplay_dot-matrix
- display_monochrom
- display_color

SchrodingersGat, 2017-07-12:
@poeschlr regarding the `display_` libraries:

I think there are few enough symbols that `display_led` and `display_lcd` are OK (at least for now). We have to ensure that we don't over-specialize, creating too many libs.

If a certain library becomes too large we can always split further down the road.

SchrodingersGat, 2017-07-12:
> opto_isolator.lib might not be the best name. (And opto maybe not the best category)
> Opto is more the how is an isolator implemented than a functional description.

Fair point. What about just `isolator.lib` - if we find this has too many symbols we can later expand by sub-category.

SchrodingersGat, 2017-07-12:
> Also I think we don't have any place for I2C transceivers/drivers/hubs and SPI switches/transceivers/... maybe interface_spi.lib + interface_i2c.lib

@jkriege2 can you give me some examples of what you mean by such components? :) It will help me to better understand their categorization.

SchrodingersGat, 2017-07-12:
> I personally would place the level translator and the isolator stuff either under logic or under their own signal translation category. (The later might be more future proof.)

Should we add `logic_translator.lib` for e.g. parts that are not covered by the 40xx and 74xx series?

poeschlr, 2017-07-12:
@SchrodingersGat wrote:
> I think there are few enough symbols that display_led and display_lcd are OK (at least for now). We have to ensure that we don't over-specialize, creating too many libs.

I really think led and lcd are bad names. there are lcd 14 segment displays.
If you only want to libs maybe the following might be a good split:
- `display_graphical` (Pixel based. Pixels spaced evenly across the display area. Can later be split into `display_graphical_monocgrom` and `display_graphical_color`)
- `display_simple` (segment based, dot matrix, bars, ... Can later be split into `display_bar,` `display_segmented`, `display_dot-matrix`)

SchrodingersGat, 2017-07-12:
How about `display_matrix` (pixel based displays), `display_segment` (bar, 7-seg, etc) then?

poeschlr, 2017-07-12:
dot-matrix displays fall (for me) more in the same category as 7 segment (or better 14 segment) displays because the have a clear separation between there "letters"

I think "graphical" describes the capabilities better. Simple might not be the best name. But would you search dot matrix displays or bar displays under `display_segment`?

SchrodingersGat, 2017-07-12:
Should we just leave it as `display.lib` and then see where the categorization falls as the library grows?

poeschlr, 2017-07-12:
TlDr: I would split the display libs along the line of graphical/non graphical

I would make it a bit more future proof. I would aim for something where we only need to split libs again when v6 comes out. (In other words, if we think this lib might be split as soon as more symbols are added we should split it now.)
If @Misca1234 carries on as he does now we will shortly reach that point. (one display symbol, footprint 3d model a week -> that sums up over the next 2 to 3 years until v6 comes out.)

But i might be wrong. Your call. It is hard to predict where the contributors take us. (And i'm sure we will be forced to split a few libs during the lifetime of v5 while others stay "empty". I just hope to limit this as far as possible.)
I understand why limiting libs might be a good idea. (Some people would like one large lib. I'm more the fine grained type. I don't think from a maintainance point of few there are problems as long as it is easy to see where a symbol should go. There is of course the danger that there exists a component that falls into multiple categories.)

---

Side note, (explains my stance a bit i think):
I hope that the contribution frequency stays at the current rate or even increases further. (The last year saw much more activity then the year before that. That year saw more activity than the year before, ...)
A large part of that is a reduced response time between initial PR opening and first feedback (even if it is only by travis) And i think the clear KLC also helps. A clear library structure might further increase interest by contributors. (Or everybody hates us for putting the bar too high and they stop contributing. God i hope not.)

SchrodingersGat, 2017-07-12:
@poeschlr I appreciate your points - if we can predict future use it is better to make the change now rather than later. I'll ponder over this one (it's way too late at night here already :) )

jkriege2, 2017-07-12:
@poeschlr There actually are analog isolators (usually called isolation amplifiers, see e.g. http://www.analog.com/en/products/amplifiers/isolation-amplifiers/ad215.html ... and we even have some in the LIB) ... and there are parts that combine an LED and an LDR, which can also be used as an analog isolator (although slow).

jkriege2, 2017-07-12:
@SchrodingersGat here are some I2C examples:
- http://www.ti.com/lsds/ti_de/interface/i2c-multiplexer-switch-products.page
- http://www.ti.com/lit/ds/symlink/iso1541.pdf
- http://www.ti.com/lsds/ti_de/interface/i2c-hub-buffer-repeater-products.page, http://www.ti.com/lit/ds/symlink/pca9518.pdf
- http://www.ti.com/lit/ds/symlink/pca9515a.pdf
- http://www.nxp.com/products/interfaces/ic-bus-portfolio/ic-bus-repeaters-hubs-extenders:MC_41849
- http://www.nxp.com/docs/en/data-sheet/P82B715.pdf

and for SPI:
- http://www.ti.com/lit/an/slyt441/slyt441.pdf

jkriege2, 2017-07-12:
@SchrodingersGat What about the name `logiclevel_translator.lib` ... level is maybe a bit generic?

jkriege2, 2017-07-12:
About displays: I would like to see
- `display_character` (7seg, typical LCD-displays for text ...)
- `display_graphic` (graphic displays maybe also LED DOT-matrix in here)

One could think about splitting out the two special cases:
- `display_dotmatrix` (dot-matrix, bars, ...)
- `display_segmented` (7-segment, ...)

but not really sure ... what is best

Best,
JAN

SchrodingersGat, 2017-07-12:
Ok I have updated `graphic_` and added `interface_i2c` and `interface_spi`. Time for bed :) Thanks for the suggestions.

[55 more replies]

closed 2018-01-20 by poeschlr