KiCad/kicad-library #1406: Hirose connectors

G33KatWork, 2017-07-10:
Hi,

Commit cf9ec51825a0a36b1d2495a3f25fd717bfe45081 added a bunch of Hirose connectors which I used on my PCBs. And I just wondered where the models that I used in the past had gone.
I found that 13e8762dd361e8cb05795e032a171ca54e7a090c reverted a whole lot of stuff including the addition of the Hirose connectors in an earlier commit. Was this intended or by mistake? Because I'd really like to see these models to be readded so that I don't have to create my own library with all the models in them.

Thanks,
- Andy

evanshultz, 2017-07-10:
Hi Andy,

I think you will want connectors at https://github.com/KiCad/Connectors_Hirose.pretty. This library was added to break out the footprints since we had so many connections. Sorry for any confusion.

You'll need to update your fp-lib-table file (https://github.com/KiCad/kicad-library/tree/master/template) to see the new libraries in KiCad. I believe I've seen an automated utility to do it some time in the past, but I can't find it now. If you go to Preferences > Footprint Libraries Manager in Pcbnew you will see the path at the top of the dialog.

Please reply if you run into more trouble!

G33KatWork, 2017-07-10:
Thanks for your answer. I have the footprints, the 3d shapes are what is missing.
The footprints from https://github.com/KiCad/Connectors_Hirose.pretty even reference the old 3d shapes:

https://github.com/KiCad/Connectors_Hirose.pretty/blob/master/Hirose_DF13-02P-1.25DSA_02x1.25mm_Straight.kicad_mod#L45

evanshultz, 2017-07-10:
Andy, my mistake. I didn't catch that you meant the 3D models.

@SchrodingersGat It looks like you merged. Can you take a look?

SchrodingersGat, 2017-07-10:
@G33KatWork you have stumbled upon my shame.

I accidentally merged my personal libraries with the official KiCad libraries. I think perhaps you have been using the 3D models from those? The Hirose 3D models are not part of the official KiCad library dataset, due to licensing issues (all models must be submitted by the creator of the model).

If you are willing to develop these models for us? :) We have a powerful set of scripting tools for generating 3D models. Have a look at the examples in the [packages3d repo](https://github.com/kicad/packages3d)

G33KatWork, 2017-07-11:
Are you sure we are talking about the same commit? There is a pull-request which deliberately added them: https://github.com/KiCad/kicad-library/pull/249, so at least the addition of the Hirose connectors don't look like an accident to me.

No offense, I just want to be sure that we are really talking about the same thing before I maybe try (no real experience in MCAD) to model them by myself.

SchrodingersGat, 2017-07-11:
@G33KatWork you're right. It appears that the other commit erroneously removed these. Let me look into it.

SchrodingersGat, 2017-07-11:
I have just submitted a PR to re-add these shapes - https://github.com/KiCad/kicad-library/pull/1407

closed 2017-10-23 by SchrodingersGat