KiCad/kicad-library #1411: RFC - Legacy 3D models

SchrodingersGat, 2017-07-12:
I think we should mark the [current packages3d] directory as legacy, and stop accepting PRs here. We are duplicating efforts which could be better spent improving the quality of the new 3D models repository.

With the effort to refresh and reorganise the symbols repository, this will eventually be separated from the 3D models in any case.

Any objections?

poeschlr, 2017-07-12:
The main question is, what do we do with models that currently only exist in the legacy lib. 
We have two options:
- reduce the number of available 3d models but keep the new repo clean. (only step/wrl pairs, correctly scaled)
- relocate the files missing files to the new repo. (the new repo would end up beeing not as clean as it currently is.)

I'm a bit uncomfortable reducing functionality for users.

Maybe we should check how much files are affected and make a call as soon as we know the scope.

Shackmeister, 2017-07-12:
I totally vote for option to,  I think the new repo is pretty solid by now.

poeschlr, 2017-07-12:
Another question: Do we overstretch ourselves by the large symbol lib reorg and the 3d files reorg?

SchrodingersGat, 2017-07-12:
We should firstly assess scope of changes. My thinking would be to leave the current (legacy) 3D models untouched - users who are using them "as is" can continue to do so but new models will not be added.

Any new models and improvements will only be made to the new improved repository.

> Another question: Do we overstretch ourselves by the large symbol lib reorg and the 3d files reorg?

No - one thing at a time. However locking the legacy 3D models at their current state requires no extra work - and reduces the work that 3D model contributors have to do.

poeschlr, 2017-07-13:
What if we add models here until the 4.0.7 release is out? (This way 4.0.7 still gets a few more models and we have a clear date where we stop adding stuff to the legacy lib)

Shackmeister, 2017-07-14:
If the new repo isnt gonna be used before 5.0 I think it makes perfect sense to keep both synced until then. Another thing I would like to discuss here before writing the devs,  (or perhaps someone a bit more recognised should write) is how the 3D file endings should be handled in 5.0 footprints,  since step and iges is now supported.. For most/all 3D models in the official repo,  the WRL model is generated from the step or same source as the step,  meaning the auracy should be as good as using steps which are internally converter. So the question is how the different file types,  should be handled in kicad/the libraries. They both have advantages so being able to seemlesly switch between them would be very much in favor. A few ideas so far on how it could work. 
1. the Stepup approach: having KiCad mainly use the WRL models for the 3D viewer, but search for a .Step with the same name when exporting step. Models. 

2. Removing file endings in the libraries: my idea here would be to have a setting in the viewer where you could set you prefered priority of file types ie. Wrl, step, iges. From here the viewer should try and find the file available. 

3. Adding both a step model and a Wrl model to the footprint. Again from here have some filtering in the viewer/exporter to select the appropriate type. 

I might be the only one considering this a potential Issue. But I've see n quite a few people trying to add step files to the library,  and it would be a shame to have fight a fight between Mcad and raytracing people since the library beautifully supports both :) maybe @easyw have something to add here

SchrodingersGat, 2017-07-14:
@Shackmeister this is a very good question and I have been giving this issue some thought. I think the way that the @easyw StepUp plugin has handled this issue for a long time was a very clever solution, and **maybe** it is the way to go for KiCad.

I imagine an option in the 3D model export e.g. "Substitute component models with STEP files". Then KiCad searches for step files in the same path. (Same as StepUp)

This would be a change that would need to happen to KiCad, and I imagine it would be well received (if it was very easy to use, and not "hacky").

easyw, 2017-07-14:
I wouldn't consider useful to add both STEP and WRL to a fp.
Footprints already accept more than one 3D model, letting the user to add i.e. mechanical parts or whatever...
Having configured both type of the same symbol would be a bit confusing and would need to have also an option to hide one of the two models... and wouldn't be
back compatibile.
The simplest solution is a search for same name but extension STEP (STP) in the same path. Still this solution can give some troubles to users if the library was not built in the correct way (i.e. some users own custom parts).
Anyway this solution is the simplest and can be adopted if clearly stated in the manual and menus.
Maurice

SchrodingersGat, 2017-07-14:
If only we knew the creator of the StepUp plugin - they would be the perfect candidate to implement this feature in KiCad itself!

Shackmeister, 2017-07-14:
Perhaps @poeschlr and @hackscribble has something to chime in with? 
I agree the stepup approach is probably the easiest and cleanest to implement. Also I think the packages3d readme should be changed from "wrl or step models accepted" to "wrl and step pairs accepted"

easyw, 2017-07-14:
> Also I think the packages3d readme should be changed from "wrl or step models accepted" to "wrl and step pairs accepted"

it should be already stated that 

> This repository contains 3D models for rendering and MCAD integration, for use with KiCAD EDA software. The model needs a mechanical model source (a manually-modelled or script-generated file); its WRL counterpart file must be obtained as a conversion from the MCAD model

I cannot find the phrase _"wrl or step models accepted"_ in the readme.md

Shackmeister, 2017-07-18:
@easyw my bad. that has already been changed :)

hackscribble, 2017-07-19:
About the first question - to freeze the legacy directory or not. 

A few days ago, I compared the list of all `wrl` files in the legacy folder with the `wrl` files in the new repo.  There were about 1,600 models "missing" from the new repo.  There were some big clumps (as well as many individual items) - mainly THT resistors, capacitors, inductors and LEDs. 

I would prefer to keep the new repo clean and not copy these old models over.  So maybe the mitigation is around documenting clearly (a) how to download and use the new models and (b) how to search for and "cherry pick" old models from the legacy folder?

Shackmeister, 2017-07-19:
@hackscribble I have picked up the work on the THT capacitors again so it can hopefully be added to the new repo. Could you perhaps modify the script to work the opposite way around, so checking if the WRL file is mentioned in any footprint. Im suspicius quite a few wrl models are not being used anymore

hackscribble, 2017-07-19:
@Shackmeister I've got a script that does that already.  I will clone fresh versions of the footprint repos and library and run it again.

Shackmeister, 2017-07-24:
@hackscribble could you please add the folder/library to the modelname in unused.txt? This way it's easier to find and manage the model

hackscribble, 2017-07-24:
Folder names added ...

[unused2.txt](https://github.com/KiCad/kicad-library/files/1169205/unused2.txt)

Shackmeister, 2017-07-24:
I was actually thinking the .3dshapes folder :)

hackscribble, 2017-07-24:
Doh :cry: 
 
Those are the `.3dshapes` folders - I typed `.pretty` into the format string by mistake I'll fix the script.

poeschlr, 2017-07-26:
I don't think it is a good idea to not ship new models. Especially because there is no 3d shapes downloader in stable and the users get the footprints with 1:1 scaling via the github plugin. If the footprints would be installed locally i would agree but because a standard installation uses the github plugin we need to give users updated 3d models. (or we need to freeze the footprints as well)

Shackmeister, 2017-08-01:
@hackscribble can you please rerun the script to check differences in the repos?

hackscribble, 2017-08-01:
@Shackmeister you mean comparing the lists of wrl files in `kicad-library` and `packages3d`? That was command line stuff, not the script.  I'll try to recreate it and post the results.

hackscribble, 2017-08-01:
3,311 `wrl` files in `kicad-library`.  1,969 in `packages3D`.

Biggest groups of missing files are (in alphabetical order) crystals, THT diodes, DIP packages, THT inductors, THT LEDs, potentiometers, THT resistors, socket strips and certain TO packages.

Here is a patch file (`-` is missing in `packages3D`).

[comparison.txt](https://github.com/KiCad/kicad-library/files/1191921/comparison.txt)

SchrodingersGat, 2017-08-01:
All of those sound like good candidates for scripting!

@jkriege2 IIRC you scripted a lot of the existing .wrl models, yes? It would be worth reassessing these scripts to see if we can get MCAD models out of them.

Shackmeister, 2017-08-02:
@hackscribble thanks!
@SchrodingersGat thats pretty much what I have been working on for the last couple of weeks. most of the THT capacitors have been based on Jans models. I found it easier to port the models over to the CQ framework where all the export tools (WRL properties, scaling etc) has already been made by @easyw. Since all the footprint models was scripted I could also adapt Jans script to output the parameters for CQ which made everything soooo much faster and cleaner. once I finish the SMD LED's Ill probably make port the THT resistors and THT LED's

SchrodingersGat, 2017-08-02:
Fantastic :)

Shackmeister, 2017-11-01:
Hi guys a bug has been reported regarding the step substitute for WRL. Wayne says the release is coming quite soon so it should get started really soon to make it into V5. 
https://bugs.launchpad.net/kicad/+bug/1710796
Would Cirilo be the main guy to ask if he could be willing to add this feature?
I think it's pretty important to get this in to avoid a war between wrl and step as the default type, since the current best way is to use a text editor to change it . (or use KSU, but the internal should be supported as well )

closed 2017-12-03 by Shackmeister