KiCad/kicad-library #1415: Questions about the process for migrating the symbol lib.

poeschlr, 2017-07-13:
This thread should manage the question of how the migration process should be handled.

We need to know how long we can discus, when we stop accepting new pull requests on the legacy lib and when we migrate what lib. (We also need someone responsible for each lib i think.)

0. )How long do we have until v5? Will the first release already be a stable release or is it more like a beta release? (can the testing phase be after the release or should it be before that?)

1. ) will the lib for v5 live in a separate repo or will it be a separate branch on the current repo?
(Or to we simply tag the lib as 4.0.7 and start the migration process?)

2. ) how much support do we give to 4.x series libs?
I would say we should allow bug fixes. (not klc fixes, Example: fix wrong pin number for a pin)

3. ) When do we stop accepting pull requests on this lib? When do we stop merging? What should happen to open pull requests?
I would say the latest when we stop merging is when the first lib gets migrated. (Would mean we have some time when no contributions are accepted aside from help by the migration process. I hope our contributors are understanding in this regard.)
I would guess we will try to merge as many of the currently open requests as possible but when the time comes we will close all pull requests on the legacy lib with a note that no new features are accepted in the legacy lib.

4. ) How do we communicate what libs are already migrated, who is responsible, ...
One option would be a google doc. Anything where multiple people have write access would do i think.
Another option would be to have a separate source branch where we delete symbols that have already been migrated. (libs are deleted once they are empty.)

5. ) Do we care that a lot of symbols break klc?
No i think we should move them as they are. Fixing klc errors should be done separately. (otherwise it gets confusing.)

6. ) When do we accept pull requests on the new lib?
I would suggest merging only as soon as all libs are migrated. Possibly when the most active libs are migrated. Or when target lib is not expected to get touched any more by the migration process. (The first option is the easiest for us.)
I would just tell everyone who opens a new pull request that we are in the middle of migrating and that their PR will be looked at once the migration has finished. (Again, hopefully our contributors are understanding in this regard.)

7. ) testing. how long?
No matter what we do there will be some errors. So we need some time after the last symbol is migrated until v5 comes out. I guess we could ask our contributors and the guys over at the forum to help us there.
I would reserve at least 3 weeks here.

gauravjuvekar, 2017-07-13:
Noob question. Will the new lib be "usable" in the current stable - i.e. if I want to add a new component, will there always be a path from the official libs by which that component can be used in the current stable release.

poeschlr, 2017-07-13:
As far as i know the library format does not change from 4.x to 5.x.
But the library structure will be vastly different. 

In short: iff i am right that the lib format does not change, 4.x users can open 5.x libs. But they need to put in extra work because we will not publish this new lib structure for 4.x (too much change. People do not expect such a change in a minor release.)

gauravjuvekar, 2017-07-13:
Isn't the S-expr format new in v5? It's in the roadmap http://docs.kicad-pcb.org/doxygen/v5_road_map.html#v5_sch_sexpr

SchrodingersGat, 2017-07-14:
@gauravjuvekar I'm unsure as to how much of the new schematic format will make it into v5. AFAIK it is only the "symbol table" which improves library loading but not the new s-expr format (.sweet).

SchrodingersGat, 2017-07-14:
@poeschlr - nice zero-offset indexing for your points ;)

> 0. How long do we have until v5? Will the first release already be a stable release or is it more like a beta release? (can the testing phase be after the release or should it be before that?)

Wayne keeps saying "End of Summer" (by which he means northern hemisphere summer, I assume). 

> 1. Will the lib for v5 live in a separate repo or will it be a separate branch on the current repo?
> (Or to we simply tag the lib as 4.0.7 and start the migration process?)

Good question. This is a question for the package maintainers, I will ask over on the developers mailing list. This is a great first step into the master plan for library management improvements. I would think we should mark `kicad-library` as legacy and start `kicad-symbols` (and remove the 3d-packages)

> 2. How much support do we give to 4.x series libs?
> I would say we should allow bug fixes. (not klc fixes, Example: fix wrong pin number for a pin)

We would continue support until release of v5, assuming points 0 and 1 above are covered.

> 3. When do we stop accepting pull requests on this lib? When do we stop merging? What should happen to open pull requests?
> I would say the latest when we stop merging is when the first lib gets [...]

There will have to be a period of side-by-side development, where 4.x improvements are also merged into the new libraries.
At the transition point where we discontinue support for old libraries, we will have to mark each outstanding PR as closed but redirect the contributor to the new libraries.

> 4.  How do we communicate what libs are already migrated, who is responsible, ...
> One option would be a google doc. Anything where multiple people have write access would do i think.
> Another option [...]

Either we have a new repo or a new branch. In either case, we could keep a spreadsheet in the repo itself, which librarians can update as the transitions are made.

> 5. Do we care that a lot of symbols break klc?
> No i think we should move them as they are. Fixing klc errors should be done separately. (otherwise it gets confusing.)

No, quality improvements are a separate issue.

> 6. When do we accept pull requests on the new lib?
> I would suggest merging only as soon as all libs are migrated. Possibly when the most active libs are migrated. Or when target lib is not expected [...]

We will have to decide a "close enough is good enough" point after which it is open for business.

> 7. testing. how long?
> No matter what we do there will be some errors. So we need some time after the last symbol is migrated until v5 comes out. I guess we could ask our contributors and the guys over at the forum to help us there.
> I would reserve at least 3 weeks here.

The more help we can get the better. As most of this work is recategorization, *hopefully* there is not too much burden.

SchrodingersGat, 2017-07-14:
I think the point at which we stop accepting PRs for old symbols is also the point at which we discontinue support for old 3D models. This means we duplicate 3D model pull requests until then.

bobc, 2017-07-14:
> )How long do we have until v5? Will the first release already be a stable release or is it more like a beta release? (can the testing phase be after the release or should it be before that?)

The release process for v4 took from about March (code freeze) to 4.0.0 RC1 in September 2015 and final 4.0.0 released in December 2015, so I would expect similar for v5. Wayne mooted the idea of a v5 freeze around March 2017 I think, but that seems to have slipped indefinitely.

If v5 was frozen today, I would expect to see 5.0.0 RC1 in Jan 2018.

gauravjuvekar, 2017-07-14:
> The more help we can get the better. As most of this work is recategorization, hopefully there is not too much burden

Assuming that we can use https://github.com/KiCad/kicad-library-utils/blob/master/schlib/move_part.py for the actual work, is it possible to use it in a batch-process mode (or just a shell script that does the job). We can just create a google spreadsheet with `old_lib`, `part_name`, `new_lib` columns, and prefill the old columns.
Then we can just ask people to fill the `new_lib` column, and once everything looks good, run the `move_part.py` script on the columns. This way, we can get the help of even the most non-technical people - it's just filling a spreadsheet - anyone could help.

SchrodingersGat, 2017-07-15:
Here's an idea:

The kicad-library-utils scripts currently have a "move part" functionality, we can extend this to do most of the work for us.

Create a script which has a set of patterns, which define which parts need to move to which new libraries.

e.g.

````
{
'microchip_pic10mcu:*' : 'mcu_microchip_pic10',  # Move entire library
'sensors:mcp9700*' : 'sensor_temperature',  # Move all MCP9700 parts to the sensor_temperature library
}

The filters can be as loose or as tight as they need to be. 

Any parts that are not matched by the filters are printed at the end of the script (so we know which ones we have missed) - these can be then added to the patterns.

The script can be updated in conjunction with any PRs to the current repository. This way, when it comes time to change, we can just run the script! We can to trial runs as we go but there's now no need to make continuous updates.

I have a barebones version of this underway, I'll post when It's workable (the patterns will still need to be filled out).

lorem-ipsum, 2017-08-17:
As someone who is just adding PRs to the repo: Please keep everything in the same repository. Moving to a new one will make all issues / PRs useless without any benefit.

A new branch should be perfectly fine. You can first do the migration in a separate branch and merge it into master when everything is done. Right before that you can also spit a new branch for maintenance of the 4.x series.

SchrodingersGat, 2017-12-11:
Closing this issue as the transfer is now under way

closed 2017-12-11 by SchrodingersGat