KiCad/kicad-library #1422: KLC - Pad-to-silk clearance isn't specified

evanshultz, 2017-07-19:
I was thinking about this on a recent PR of mine, and also saw this topic pop up at https://github.com/KiCad/RF_Modules.pretty/pull/22. It seems we don't specify this distance. KLC 7.3.ii only mentions silk can't overlap, but it doesn't give a minimum or recommended distance.

I typically do about what Rene wrote and use ~0.2mm since I didn't find a specific number in KLC.

I think this should be added to KLC. While I don't see it in any IPC docs I can find online, page 13 of https://www.cskl.de/fileadmin/downloads/PCBLIBRARIES/Documentation/What-is-New-in-IPC-7351C_.pdf mentions that the spacing should be equal to the line width. For KiCad, that would typically be 0.12mm but we do allow variation in silk line width for some special cases.

I think clause 7.3.ii of KLC could be amended to `Silkscreen to pad clearance is equal to the silkscreen line width and may not be placed over pads.`. Even though mentioning clearance should be clear that they cannot overlap, it's probably best to explicitly state that.

Agreed? Add this to KLC? Any other comments?

suzizecat, 2017-07-19:
` Silkscreen to pad clearance is equal to the silkscreen line width and may not be placed over pads.` I would rather write 
` Silkscreen to pad clearance is at least equal to the silkscreen line width and may not be placed over pads.`

Since having a 1.2mm space could be bothersome (imperial-unit based FP, 1mm grid FP and so on)

evanshultz, 2017-07-20:
@SchrodingersGat 
I assume you can take a look at this and update KLC?

If/when you do, two other things:
1. 3.3 and 3.7 are still there and we discussed they're redundant.
1. 1.8 uses bold and italic to describe the two different line endings.

Shackmeister, 2017-07-23:
@evanshultz shouldn't it be 0.2mm since that is the default pad clearance?

evanshultz, 2017-07-23:
@Shackmeister 
You mean the pad-to-pad clearance? If so, then I haven't seen there's any correlation. Printing silk has a variety of methods that can give a variety of results, but I know it to be unrelated to the process of etching away copper.

Shackmeister, 2017-07-23:
@evanshultz it was the soldermask clearance I meant

SchrodingersGat, 2017-07-23:
@evanshultz thanks for pointing that out, I have fixed those KLC issues.

@Shackmeister KLC 7.3/ii 

```
Silkscreen clearance:
a. Silkscreen must not be placed over pads
b. Silkscreen must have at least 0.2mm clearance around pads
```

poeschlr, 2017-08-23:
@evanshultz is this resolved to your satisfaction?

suzizecat, 2017-08-24:
> b. Silkscreen must have at least 0.2mm clearance around pads

Since this 0.2mm comes from the default pad to soldermask clearance, what should be done when the footprint specify a non-standard pad to mask clearance (like in KiCAD/Housings_LGA.pretty#6 )? 

IMHO, this should be specified by also giving a silkscreen to soldermask clearance (or just manage specifically in those case since that's not very common)

poeschlr, 2017-08-24:
I use 0.2mm for silk to copper because this is what the fab i mostly use recommends. It can have solder mask. The minimum solder mask clearance is much smaller. (50µm)
I assume this is similar for other fabs. (Much larger clearance between copper and silk than for the soldermask.)

suzizecat, 2017-08-24:
@poeschlr If you are replying to me, I think there is a misunderstanding... AFAIK, you shouldn't have silkscreen on copper, so if a pad have an extended soldermask clearance, which may be over 0.2mm, you should put away the silkscreen to avoid overlapping.
In this case, how should it be handled ?

poeschlr, 2017-08-24:
Ok then how about:
Silkscreen has a minimum copper to silk clearance of 0.2mm. In addition it is not allowed to overlap with masked areas. (= areas that do not have solder resist on the final pcb)

suzizecat, 2017-08-24:
Well, it's seems fine, but this imply that there may be no clearance at  all between the silkscreen and the copper (without overlapping, just 0 mm). I don't know if this is fine, but if it is, i'm ok with it.
(Sorry to quibble about that but, hey, gotta be precise 😉 )

evanshultz, 2017-08-25:
@SchrodingersGat Can you update the [KLC page](https://github.com/KiCad/kicad-library/wiki/Kicad-Library-Convention) again? 3.3 has a typo. It says "Manufacturer Part Number (MPN) should be used given preference" but we need to remove either "used" or "given preference".

@suzizecat Are you only asking that silkscreen not be allowed in areas where there is no mask (or course exceptions may apply)? 7.3 clearly says silkscreen must avoid copper by 0.2mm, so that shouldn't be in question.

suzizecat, 2017-08-25:
@evanshultz The 7.3 don't say explicitly that silkscreen must avoid _copper_ (it's about _pad_ that is, for me, a little bit different...) so, when  you got an extended clearance on the pad. IMO, it may be confusing.

poeschlr, 2017-08-25:
I would make it in this way for your footprint. I can't think of a way to clearly specify this in the KLC. But i think some special cases will always need a bit of discussion.

[image]

suzizecat, 2017-08-25:
Ok but, do we agree that there is a 0mm clearance between the silkscreen (which i believe is the green line) and the masked area (so, potentially the copper) ? 

In this case, the extended clearance specify 0.2mm but what could have happened if it was more than 0.2mm ?

poeschlr, 2017-08-25:
My thing is not drawn to scale! And 0.2 is the minimum clearance. You can increase it if you think it is necessary.

suzizecat, 2017-08-25:
The problem doesn't come from the scale, i'm writing about the fact that the silkscreen touch the mask limits. 
But, since 0.2mm is specified as a minimum clearance and if it's clearly stated that the silkscreen must not overlap any copper, it  solves my concern.

closed 2017-09-06 by evanshultz

SchrodingersGat, 2017-09-06:
I have updated 3.3 and 7.3 in the [KLC](https://github.com/KiCad/kicad-library/wiki/Kicad-Library-Convention)