KiCad/kicad-library #1423: Cirrus CS8420: Multifunction pins on symbols

evanshultz, 2017-07-19:
I was ready to add [CS8420 ](https://d3uzseaevmutz1.cloudfront.net/pubs/proDatasheet/CS8420_F4.pdf)to the library, but I found a symbol already existed named `CS8420_SOFT`:
[image]

This symbol only covers the pin configuration of CS8420 when in software mode (where digital communication is provided for an external processor to configure the IC).

This is shown in the datasheet on page 51:
[image]

In addition, there are not one but 6 (yes, really, six) hardware modes. Page 57:
[image]

Page 61:
[image]

Page 65:
[image]

Page 69:
[image]

Page 72:
[image]

And finally, page 76:
[image]

Is there already a known good method to handle this situation? CS8420 is obviously a special case with 7 different pin configurations, so maybe it's just an outlier and we haven't come across this before?

On microcontrollers I've seen we use the "base" pin number, like `PA1` instead of `PA1_UART5_RTC_ADC0` or some other long pin name. That wouldn't seem to apply here since there isn't a singular name for each pin (only 3 pins are common in all modes) that can be referenced in the datasheet.

Having separate symbols seems a bit lame but maybe that's the best option? If we were to combine the pin names for all the hardware modes into one symbol, we'd end up with these pins names:
```
1: COPY/C
2: DFC0
3: ~EMPH/U/V
4: RXP/SMFT0
5: RXN/SFMT1
6: VA+
7: AGND
8: FILT
9: ~RST
10: RMCK/APMS
11: RERR/~LOCK/TBCLD
12: TCBLD/ILRCK/RCBL
13: PRO/C/ISCLK
14: MUTE/SDIN/CHS
15: TCBL/NVERR
16: OSCLK/CEN
17: OLRCLK/V
18: SDOUT/U
19: ~AUDIO/V/CUVEN
20: S/~AES/PRO/C
21: OMCK/APMS
22: DGND
23: VD+
24: H/S
25: TXN/U
26: TXP/C
27: DFC1
28: ORIG/U
```

That seems rather ugly to me. But so does having 7 symbols for this part. Any suggestions?

suzizecat, 2017-07-19:
You can have a look at any STM32 MCU (which are script-generated) to see that it's handled with pins combined. Since any combinaison is possible, this is a little bit different but you get the thing.

IMHO it would be fine that way.

SchrodingersGat, 2017-07-19:
The STM32 libs are the logical extension of this approach, and (IMO) they are ridiculous. Some take up almost an entire sheet of (A4) paper.

But as you state @evanshultz these are the two options - one symbol with large names or multiple symbols.

For this I would go with large names:

a) Less to rework later
b) When we (eventually) convert to the new library format we can enumerate each pin function

evanshultz, 2017-07-20:
Another option would be to only give pin numbers, not names, but that's unlike any other IC I've seen and just seems bad practice. I only mention it since even a bad solution is still a solution.

@SchrodingersGat So you're OK if I rework the existing symbol to make it work for all modes?

SchrodingersGat, 2017-07-20:
@evanshultz I think so - the names are not "too" long, for this one.

suzizecat, 2017-07-20:
@SchrodingersGat Got to say that STM32 symbols are absolutely insanes (need A2 paper for STM32L4...) ^^'

SchrodingersGat, 2017-07-20:
@suzizecat yes I have it on my list to fix but my list is very very long.. If anyone else gets around to doing this... ;)

closed 2017-07-23 by evanshultz