KiCad/kicad-library #1425: What to do with vactrols?

evanshultz, 2017-07-20:
I'm creating a symbol and footprint for the VTL5Cx family of vactrols. These are still used in audio and hobbyist applications because they have some unique characteristics such as very smooth attack and release (not discontinuous), they're relatively linear as diode current changes, etc. They're cool parts.

Here is the catalog page and datasheets for the single-cell and dual-cell versions:
- http://www.hofoo.com.cn/FileDoc/200552441243879.pdf
- http://www.qsl.net/wa1ion/vactrol/vactrol.pdf
- http://www.alliedelec.com/m/d/ecb1ed91959ae54d97cf67f5103d9b2b.pdf

So there is a 4-pin (single) and 5-pin (dual) version, and it's basically an LED with a photo-responsive resistor in one package.

And some information on them:
- http://denethor.wlu.ca/pc300/optoisolators/analogoptoisolatorintroduction.pdf
- http://www.perkinelmer.com/CMSResources/Images/44-3429APP_AnalogOpticalIsolatorsAudioApps.pdf

So now to the task at hand. Questions:
1. How should the photoresistor be represented? Everything I see (app notes above, [Google search](https://www.google.com/search?q=vtl5c2&source=lnms&tbm=isch&sa=X&ved=0ahUKEwjfqYSt7pbVAhXmylQKHQdUDiAQ_AUIBygC&biw=2134&bih=1095#tbm=isch&q=vtl5c2+schematic), [QSC MX schematic](https://qscservice.com/files/6613/6329/3814/EX4000.pdf)) looks like a resistor, but we've gone with a box due to IEE 315-1975. That being said, even it shows similar isolators with a squiggly-line resistor symbol in section 8.10. Here is what I have made:
[image]
1. How to name the symbols? There are `VTL5Cx` and `VTL5Cx/2`. However, we can't use a `/` in the symbol name and having two `x` would be weird and confusing. Perhaps `VTL5C` and `VTL5Cx2` or `VTL5Cx` and `VTL5Cx2` or `VTL5Cx` and `VTL5Cx_DUAL` or `VTL5Cx_SINGLE` and `VTL5Cx_DUAL`?
1. What should the refdes letter be? IEEE 315-1975 says that "photon-coupled isolator" parts should start with a `U`. Any objections?
1. Where should the footprints live? I don't find a .pretty repo that seems suitable.
1. How should the footprints be named? Same as the symbol names?

Here is my symbol code so far:
```
# VTL5C
#
DEF VTL5C U 0 40 Y N 1 F N
F0 "U" 0 200 50 H V C CNN
F1 "VTL5C" 0 -200 50 H V C CNN
F2 "" 50 -250 50 H V C CNN
F3 "" 50 -250 50 H V C CNN
$FPLIST
 VTL5C*
$ENDFPLIST
DRAW
S -200 150 200 -150 0 1 0 f
S 90 70 150 -70 0 1 0 N
P 2 0 1 0 -170 30 -90 30 N
P 2 0 1 0 -50 30 10 -30 N
P 2 0 1 0 -50 30 10 -30 N
P 2 0 1 0 0 30 60 -30 N
P 3 0 1 0 10 -10 10 -30 -10 -30 N
P 3 0 1 0 10 -10 10 -30 -10 -30 N
P 3 0 1 0 60 -10 60 -30 40 -30 N
P 3 0 1 0 120 -70 120 -100 200 -100 N
P 3 0 1 0 120 70 120 100 200 100 N
P 4 0 1 0 -230 -100 -130 -100 -130 100 -230 100 N
P 4 1 1 0 -170 -30 -130 30 -90 -30 -170 -30 N
X - 1 -300 100 70 R 50 50 1 1 P
X + 2 -300 -100 70 R 50 50 1 1 P
X ~ 3 300 100 100 L 50 50 1 1 P
X ~ 4 300 -100 100 L 50 50 1 1 P
ENDDRAW
ENDDEF
#
# VTL5Cx2
#
DEF VTL5Cx2 U 0 40 Y N 1 F N
F0 "U" 0 200 50 H V C CNN
F1 "VTL5Cx2" 0 -200 50 H V C CNN
F2 "" 50 -250 50 H V C CNN
F3 "" 50 -250 50 H V C CNN
$FPLIST
 VTL5C*
$ENDFPLIST
DRAW
S -200 150 200 -150 0 1 0 f
S 90 70 150 -70 0 1 0 N
P 2 0 1 0 -170 30 -90 30 N
P 2 0 1 0 -50 30 10 -30 N
P 2 0 1 0 -50 30 10 -30 N
P 2 0 1 0 0 30 60 -30 N
P 2 0 1 0 200 0 150 0 N
P 3 0 1 0 10 -10 10 -30 -10 -30 N
P 3 0 1 0 10 -10 10 -30 -10 -30 N
P 3 0 1 0 60 -10 60 -30 40 -30 N
P 3 0 1 0 120 -70 120 -100 200 -100 N
P 3 0 1 0 120 70 120 100 200 100 N
P 4 0 1 0 -230 -100 -130 -100 -130 100 -230 100 N
P 4 1 1 0 -170 -30 -130 30 -90 -30 -170 -30 N
X - 1 -300 100 70 R 50 50 1 1 P
X + 2 -300 -100 70 R 50 50 1 1 P
X ~ 3 300 100 100 L 50 50 1 1 P
X ~ 4 300 0 100 L 50 50 1 1 P
X ~ 5 300 -100 100 L 50 50 1 1 P
ENDDRAW
ENDDEF
#
```

jkriege2, 2017-07-20:
Hi!

a few comments:
1. footprint should go to https://github.com/KiCad/Opto-Devices.pretty ... look at  	Optoisolator_NSL-32.kicad_mod for reference (it is a coupler with LED+LDR)
2. your symbol looks fine ... please note that the box-line (and potentially also the resistors box) should be 10mil thick
3. REFDES seems fine to me!

Best,
JAN

SchrodingersGat, 2017-07-20:
Opto-Devices seems like a good fit to me also.

evanshultz, 2017-07-20:
Thank you, gentlemen! PR submitted.

closed 2017-07-21 by evanshultz