KiCad/kicad-library #1429: Wrong pin numbering - *78M05_TO252

bumcyk, 2017-07-20:
Pin numbers are inproperly assigned for the following parts:
- KA78M05_TO252
- LM78M05_TO252
- MC78M05_TO252

Function - current pin number - correct pin number
Input - assigned pin 3 - should be pin 1
Ground - assigned pin 1 - should be pin 2
Output - assigned pin 2 - should be pin 3

The part is incorrectly bound with LM1117-x, where the pin numbering is correct.

Sources:
http://www.ti.com/lit/ds/symlink/lm341.pdf
http://www.farnell.com/datasheets/1855685.pdf

evanshultz, 2017-07-20:
@bumcyk 
Thanks for pointing this out. Yes, indeed, this is an error. I'm quite familiar with both parts. Generic 78xx in DPAK have the same pinout as in TO220.

Are you up for making this change and generating a PR to merge it into the library?

bumcyk, 2017-07-20:
@evanshultz 
Sure, here you go. I'm new to Github and I don't know yet how to publish changes, so I attach both modified library and the diff file.
[regul.txt](https://github.com/KiCad/kicad-library/files/1163034/regul.txt)
[regul.zip](https://github.com/KiCad/kicad-library/files/1163035/regul.zip)

evanshultz, 2017-07-20:
I had forgotten that the regulator library has been updated recently by @jkriege2. If you're looking at the library that came with KiCad 4.0.6 you may see something different than in the latest library hosted here.

Anyway... all the `*M05_TO252` parts are an ALIAS of `NCP1117-12_TO252`.

[NCP1117 datasheet](https://www.onsemi.com/pub/Collateral/NCP1117-D.PDF):
[image]

[MC78Mxx datasheet](https://www.onsemi.com/pub/Collateral/MC78M00-D.PDF):
[image]

So yep, definitely a different pinout.

Here's the schematic symbol:
```
# NCP1117-12_TO252
#
DEF NCP1117-12_TO252 U 0 10 Y Y 1 F N
F0 "U" -150 125 50 H V C CNN
F1 "NCP1117-12_TO252" 0 125 50 H V L CNN
F2 "TO_SOT_Packages_SMD:TO-252-2" 0 225 50 H I C CNN
F3 "" 0 0 50 H I C CNN
ALIAS NCP1117-1.5_TO252 NCP1117-1.8_TO252 NCP1117-1.9_TO252 NCP1117-2.0_TO252 NCP1117-2.5_TO252 NCP1117-2.85_TO252 NCP1117-3.3_TO252 NCP1117-5.0_TO252 LF15_TO252 LF18_TO252 LF25_TO252 LF33_TO252 LF50_TO252 LF60_TO252 LF80_TO252 LF85_TO252 LF120_TO252 LP2950-3.0_TO252 LP2950-3.3_TO252 LP2950-5.0_TO252 MC78M05_TO252 LM78M05_TO252 KA78M05_TO252 LF47_TO252 LF90_TO252
$FPLIST
 TO?252*
$ENDFPLIST
DRAW
S -200 75 200 -200 0 1 10 f
X GND 1 0 -300 100 U 50 50 1 1 W
X VO 2 300 0 100 L 50 50 1 1 w
X VI 3 -300 0 100 R 50 50 1 1 W
ENDDRAW
ENDDEF
#
```

The 78M05 parts should be an ALIAS of `LM7805_TO220`.

And lastly, here's the footprint (which is fine for both parts):
[image]

evanshultz, 2017-07-20:
Thanks @bumcyk. I've submitted a PR for this.

Git allows you to fork (copy) a repository (in this case, the schematic symbol library), make changes, and then submit a pull request to merge your changes into the original repository. That's the workflow we use rather than emailing libraries. If you're interested in making more fixes or submitting new parts please ask away. We depend a lot on community support.

At any rate, thanks again for catching this and letting us know!

evanshultz, 2017-07-25:
Closing since the PR to fix it was merged.

closed 2017-08-05 by jkriege2