KiCad/kicad-library #1435: Footprint TO_SOT_Packages_SMD:SOT-23 and 3D model pins does not match

vpelaez, 2017-07-22:
I have just installed version 4.0.6 64 bits in Windows 10 and found that pins in the 3D model of the TO_SOT_Packages_SMD:SOT-23 footprint does not match the footprint representation. Please see the following image.
[image]

I think that the 3D figure needs to be rotated clockwise to match the pads.

When using this footprint in a PCB the 3D viewer shows the same problem.

This problem is not present in the TO_SOT_Packages_SMD:SOT-23-5 footprint.

Shackmeister, 2017-07-22:
Hello I can see from the screenshot you are using the old 3D model, you should update them to the newest :) see more here: https://forum.kicad.info/t/incorrect-component-models-in-3d-viewer/7011/5

Shackmeister, 2017-07-22:
Also make sure you have the latest footprints while you are at it :)

Shackmeister, 2017-07-23:
@vpelaez did you have ny luck solving it?

vpelaez, 2017-07-24:
Hello @Shackmeister, you are very right, I have just downloaded the new 3D models and it is working as expected now. Thank you very much for your help.

closed 2017-07-24 by vpelaez