KiCad/kicad-library #1438: Pinhead 3D model not matching footprint

antoniovazquezblanco, 2017-07-23:
The 3D viewer is not placing my 1x6 pinhead properly.

[image]

poeschlr, 2017-07-23:
3d models are not updated automatically. If you use the github plugin for your footprint libs, the footprint is updated everytime kicad needs a footprint.
This means if we update anything in the connection between 3d models and footprints, users get incompatible libs that require manual update. (updated footprints but the old 3d model)

Yes this is a bad situation. We library managers are not happy about it. We do not like that the default kicad installation uses the github plugin. This setup should only be used by users who know what the downsides of it are.
To get a local footprint lib either use the [footprint lib download script](https://github.com/KiCad/kicad-library-utils/blob/master/download_pretty_libs.py) found in the kicad-library-utils repo or use the library wizard as described in this [tutorial](https://github.com/bobc/kicad-doc/wiki/How-to-install-local-Footprint-libraries) by @bobc.
This will not solve your current problem, but it might be a good idea anyways. (If you use a local setup you are in control when you get new footprints.)

To solve your current problem, try to download the current 3d models from this repo. Or from or new [packages3d repo](https://github.com/KiCad/packages3D). (The later has fewer models because it only includes models for which we also have a step model)
I would suggest you clone the repo into your user directory and set the path variable KISYS3DMOD to point to the correct location. (setting path variables is done in the kicad main window->configure paths)
This way updating 3d libs can be done with a simple `git pull`

If the problem still exists after you download the newest 3d models then something is really wrong.

There could also be the opposite problem. You could have 3d models that are too new for what is stored in your pcb_new file. (footprints are stored in the header of the pcb_new file after importing the netlist.)
To check if this is the case, simply open the footprint with the footprint editor and check the 3d view there. (Do not open the footprint editor from within pcb_new right click menu. Open the editor from the main window and load the footprint via the footprint browser.)
If this is the case, you can solve it by updating the footprint on your board. For that open the footprint properties and press the change footprint button to open the change footprint dialog.
In this dialog you have radial boxes to select what to exchange. Either choose all footprints or footprints with the same footprint name.
Be aware that updating footprints resets the position of the reference and value text fields to what is defined in the library.

antoniovazquezblanco, 2017-07-23:
Using my local repositories for the 3D models solved the issue.

Thank you very much!

closed 2017-07-23 by antoniovazquezblanco

poeschlr, 2017-07-23:
@SchrodingersGat We really should make sure that 4.0.7 gets the new 3d models. Otherwise this problem will persist. (Or get even worse for future 4.x releases if more of them are planned until 5.x)

pointhi, 2017-07-23:
@SchrodingersGat does someone know the current state of the symbol schematic liibrary with 4.0.7? The last time I tried a new installation, half of them were not enabled by default. (for example switches.lib)

Shackmeister, 2017-07-23:
@poeschlr shouldn't we also make sure all footprints has KISYS3DMOD added so its possible to use another path?

SchrodingersGat, 2017-07-23:
4.0.7 has been marked as a 'minor update' that won't feature a new release of the libs. There's not enough time to switch to the new 3D models repo. We should certainly make this a priority for next release.

Shackmeister, 2017-07-23:
@SchrodingersGat kicad-library should be up to date?

poeschlr, 2017-07-24:
@SchrodingersGat  because a lot of 3d models got updated with the 1:1 scaling we should definetly supply the new 3d models. (I assume the footprints are still supplied with the github plugin.)
The easiest way to achieve this would be to tag the current master and ship not only the new 3d models but also th new libs.

poeschlr, 2017-07-24:
@Shackmeister the KISYS3DMOD requirement is already part of the KLC. So yes it would be nice to have all footprints use it. 
Not sure if it is really a problem. Kicad might interpret it as a relative path. (relative to KISYS3DMOD)