KiCad/kicad-library #1443: How should keepout areas be shown in footprints

poeschlr, 2017-07-25:
Keepout zones can not be defined in footprints. How should we show that a footprint has an area where no copper should be placed.

I have seen the following options:
 1. dwgs.user layer for the zone outline plus text on cmts,user (Example: https://github.com/KiCad/RF_Modules.pretty/pull/16)
 1. everything on the fab layer. (Example: https://github.com/KiCad/Connectors_Hirose.pretty/pull/17)

We should fix it in the klc such that all footprints follow the same rule and the user of our libraries knows what to expect.

I vote for option 1. (Fab layer is for the documentation. If we put stuff like keepout areas on this layer, it gets very crowded.)

hackscribble, 2017-07-25:
+1 for option 1 

Would we also want to specify the visual style of the keepout?  For example, one of the following ...

[image]

evanshultz, 2017-07-25:
I like option 1a: hatched lines on Dwgs.User.

poeschlr, 2017-07-25:
I would say a or b plus maybe a text on comments that specifies what the drawing means.
Some footprints say no copper, others might specify no vias (no bare copper), others might specify no tracks except ground, ...
Or should we define different styles for different meanings?

hackscribble, 2017-07-26:
My preference would be for just one style with the meaning explained in comments.

This example is hatched on Dwgs.User and text on Cmts.User ...

[image]

SchrodingersGat, 2017-07-26:
Option a) works best IMO, especially for irregular regions. Is there any reason that the explanatory text (on User.Comments) should not be overlaid on the same space? It makes it a bit harder to read but it's obvious in high-contrast layer mode

hackscribble, 2017-07-26:
This is with high-contrast button enabled (4.0.7 on Ubuntu in OpenGL). 

[image]

SchrodingersGat, 2017-07-26:
Are they both on the same layer? Here's what I get, OpenGL on Ubuntu:

[image]

[image]

hackscribble, 2017-07-26:
Mine are definitely on Dwgs and Cmts.  Here is the view in OpenGL without the contrast button enabled.

[image]

hackscribble, 2017-07-26:
Also, can I check that it is not possible in footprint editor to write directly to either of these layers?  One needs to write onto another layer and then edit the properties (or edit the file) to move the lines and text to the .User layers?

poeschlr, 2017-07-26:
@hackscribble set the active layer different to either commends or drawings while in highcontrast.
@SchrodingersGat set the active layer equal to drawings in the first screenshot and equal to comments in the second.
That's why they look different.

poeschlr, 2017-07-26:
@hackscribble Yes one can not directly draw on these layers. (But this might be something we can request from the developers. Maybe we can even get a dedicated layer in v5?)

SchrodingersGat, 2017-07-26:
@poeschlr I would like to have a working keepout, rather than a layer dedicated to keepout. 

This would require that keepouts can be assigned multiple layers, I think. I have a code branch where this is possible (spawned from this discussion) but it requires more work.. Hopefully I get time to work on this pre v5

suzizecat, 2017-07-26:
I would also point out that IMHO it'normal that FP designer can't write on these layers since it's "User" Comment and "User" drawings... There should be something dedicated

poeschlr, 2017-07-26:
If keepout zones can be used in the footprint that would be even better. But if this does not get finished in time, it might be a good idea to include a special graphical layer for keepouts on top and bottom. (Yes dirty workaround but better than nothing.)

suzizecat, 2017-07-26:
> it might be a good idea to include a special graphical layer for keepouts on top and bottom. (Yes dirty workaround but better than nothing.)
Well, in this case, you should add one keepout layer by copper layer...
In example, in RF design you often have to clear copper on all layers (or all layers except the last one)

poeschlr, 2017-07-26:
Lets hope oliver finishes the implementation of keepout support for footprints.

hackscribble, 2017-07-26:
> Lets hope oliver finishes the implementation of keepout support for footprints.

In the meantime, is there still value in adding something to KLC to specify a purely graphical and text note representation?  If we (improperly) use the .User layers, at least it would be easy to trawl all footprints later to identify those that need upgrading to the new implementation.

Or are there relatively few contributions with keep-out zones, so this is not a big issue?

poeschlr, 2017-07-29:
I would at least fix the style we want to use until v5 comes out.

If i read all your posts correctly i think we all agree with version 1a (cross hatched area on drawings layer plus text on comments layer.)

SchrodingersGat, 2017-07-29:
I will update the KLC with such a rule - I'm away until Tuesday so not until then.

SchrodingersGat, 2017-07-31:
I have added this to the KLC, new rule 7.8

poeschlr, 2017-08-04:
Thanks. I assume i can close this now. (If anyone disagrees simply reopen it.)

closed 2017-08-04 by poeschlr