KiCad/kicad-library #1451: What should we do with dual (or multi) row connectors that do not follow the odd/even pin numbering

poeschlr, 2017-07-26:
There are dual row connectors that do not follow the even odd pin ordering used in the current library symbols.

Examples:
There are connectors that have the classic dip style pin ordering (counter clockwise numbering).
https://github.com/KiCad/Connectors_Harwin.pretty/pull/8
Here the pins stay normal numbers but the order is different. I would use the suffix _DIP-numbering to signal this.
(Not sure if this would be clear enough. The description will of course have the info that this symbol has pin ordering similar to the dip packages. The current dual row connectors would get odd/even mentioned in the description.)

There are phoenix multi row connectors where the mating connector is not one part for all rows. (It is one connector per row)
https://github.com/KiCad/Connectors_Phoenix.pretty/pull/46
Maybe a good pin ordering here would be similar to BGAs using a letter for the row and numbers for the pin number within this row. (This also works for 3 an 4 row connectors. Phoenix manufactures such connectors.)
It might even be a good idea to create these connectors as multi unit symbols.
I really don't have a good naming convention for such symbols. Maybe suffix _BGA-numbering

bobc, 2017-07-29:
>both rows start at the same side and count up in the same direction

We already have connectors that follow this convention, e.g lots of Molex connectors [Mini-Fit Jr](http://www.molex.com/pdm_docs/sd/039288100_sd.pdf)

Not sure there is a problem, surely we should follow the manufacturer? In this case of Molex, they mark pins 1, 2, 3 with stripes and a digit, so it would be *very* confusing to adopt a different convention.

[image]

Some connectors use a letter for rows, e.g. 1a, 1b, or A1 B1.

I'm not even sure that the even/odd ordering is defined anywhere, although it seems to be a common convention. For dual row 0.1 headers, a nominal pin 1 is marked for orientation purposes, but the pin numbering is up to the user I think.

bobc, 2017-07-29:
ERNI connectors, where row 1 is a1..an, row 2 is b1..bn. Also marked on connector as such \:

[image]

poeschlr, 2017-08-03:
What is the purpose of the SIL and DIL connectors? They have the same pin ordering as the conn_PPxRR connectors. But they use inverted pins. This makes no sense to me.

Forum topic that made me aware of this fact:
https://forum.kicad.info/t/generating-new-connector-header-lib-for-eeschem-in-python/7315

poeschlr, 2017-08-04:
In the forum there is a generator script posted.

If we can find a naming convention i will create the following variants:
1. The current conn_RRxPP variant (odd/even)
1. The DIP pin order which is currently used by the DIP_PP symbols (counter clockwise pin numbering)
1. One where row one is from one till num pins/2 and row tow starts at num_pins/2+1 up to num_pins (the harwin connectors in the pull request linked in my first post use this pin numbering scheme.)
1. The variant where there is a row index (letter) and the pins are numbered separately for each row (Picture of erni connector by @bobc)
1. Maybe even two of these (letter first/number first)

I would remove the SIL/DIL and DIP symbols. (duplicates of other symbols, very long pins, inverted pins)

---

my suggestion for names:
PP stands for pins per row.
1. conn_02xPP_odd-even (maybe keep the current name without suffix? I would say for clarity adding the suffix might be better.)
1. conn_02xPP_counter-clockwise
1. conn_02xPP_top-bottom (very unsure about this naming. Only works while the connector is not rotated.)
1. conn_02xPP_row-letter-first
1. conn_02xPP_row-letter-last

I am open for suggestions. (Really not sure about the naming.)

poeschlr, 2017-08-04:
I just noticed that the pins for connectors are not alligned to the 100 mil grid (for connectors with an even number of pins per row.)

Should i change this when i generate the new symbols? (I would just move even numbered connectors down by 50 mil)

poeschlr, 2017-08-05:
Another thing. Might it make sense to have male and female connector symbols?
(Suffix _male and _female)

jkriege2, 2017-08-05:
Fixing the 100mil grid would be great!

About the male/female: I'm not sure ... we have a few special connectors where we do that (I think a few basic ones + the D-SUBs) ... but doubling the amount of symbols (I presume you'll create wuite a bunch) is maybe over the top?

JAN

poeschlr, 2017-08-05:
@jkriege2 any opinions for the naming convention?

poeschlr, 2017-08-05:
For now the symbols look like this.
I need to find a way to incorporate longer pins while still staying on 100mil grid to accommodate the longer pin names for the letter-pin number combinations. (see pins b10 and a10 of row letter first symbol in the screenshot below.)
[image]

poeschlr, 2017-08-05:
How to resolve the pin length problem:

1. One easy option would be to increase the pin length to 200. (The symbol would get a lot larger.)
1. Or increase the pin lenght to 150 and make the body wider by 100 mil (50 in each direction)
1. I could also increase the pin size to at least 135 which would mean the body can be at most 130mil wide. (Currently it is 200 mil) This leaves very little space for artwork within the connector body.
1. Another option would be to decrease the font size to 35mil. (That would violate the KLC though.)
1. Or increase only the pin lenght to 150 and leave the body as is now. With this option pins are no longer on the 100mil grid.

jkriege2, 2017-08-05:
My opinion on names:

- I think everyone is used to the odd-even ordering with the CONN_02xPP connectors, so I wouldn't rename them and see the others as variants.
- The for the DIP-styles I would use `Conn_02xPP_DILNumbering´ (DIL=Dual-In-Line), as all DIL/DIP packages that I know use that row-ordering and people will immediately recognize what is meant (also, when they want to use such a symbol to represent e.g. a Textool-test-socket
- instead of `conn_02xPP_top-bottom` maybe `Conn_02xPP_row-row` or `Conn_02xPP_MultiRowNumbering`?
- Instead of `conn_02xPP_row-letter-first` maybe `Conn_02xPP_RowLetterNumbering` ... and I think one variant would be sufficient? If a second, would we need `Conn_02xPP_ColLetterNumbering` (column=letter, row=int)? WHat's the difference between letter-first and letter-last anyways? I don't see one in the screenshots above!

Also I would capitalize the first letter of the name, i.e. `Conn` instead of `CONN`. I made a PR to change this (in advance?) and also remove/move some other parts and do some renaming stuff: https://github.com/KiCad/kicad-library/pull/1487

Best,JAN

jkriege2, 2017-08-05:
BTW: Should we have some extra Conn-Symbols for DIP-sockets that look more like such a socket? I think there would be some use-cases (e.g. test-circuits for ICs, EPROM/µC-programmers ...). In some cases it would be nice to be able to say explicitly in the schematic "This is a test-socket" ...

What do you all think?

JAN

gauravjuvekar, 2017-08-05:
@poeschlr 

> 5. Or increase only the pin lenght to 150 and leave the body as is now. With this option pins are no longer on the 100mil grid.

Perhaps I didn't get it, but wouldn't increasing *both* left and right pins by 50mil still keep all pins on 100 mil grid? The pins will remain on 100 mil grid - the symbol center anchor would be offset by 50 mil to left/right.

evanshultz, 2017-08-06:
@poeschlr @jkriege2 
Rene, you've been thinking about this longer than me, and done a lot of great work, so I'm sorry if I'm missing something and write something dumb here. This is a really important topic to make sure we have symbols for all connector footprints. I'm just going to stay typing as thoughts come:
-I completely agree that all pins should be on a 100mil grid.
-I also think, especially with hobby projects, that changing connectors may be common. So IMO it's best to make the pin positions match for all connector symbols.
-Would it be better to use capital letters in pin names instead of lowercase (`A1` instead of `a1`)? Regardless of what the manufacturer uses (like the ERNI part above), capital letters look better and are less ambiguous.
-If the symbols stay the same visually as they are now, but are scripted, what about using a square instead of multiple polylines for the pin indicators inside the symbol. A square can put 4 text lines on a since line, which can save space since the symbols for larger CONN_xxXxx symbols are very long right now!
-Here's another option to your list of possible resolutions: 6. Put the pin numbers inside the symbol. I made a quick mock-up and, after changing the pins to 100mil and making the body 100mil wider it looks like there's room for 3 characters in each pin name:
[image]
Since there are generic connectors, I think it's worth considering if they could have 50mil pins if you need more room. We do this in other places for non-atomic parts.

poeschlr, 2017-08-06:
I really dislike showing pin names for such a symbol. It is redundant information to the pin number.

Your suggestion of a wider body does not help for the problems with linger pin numbers. (Chang pin 10 to a10 and test your symbol again.)
I think I will use the suggestion by @gauravjuvekar and use 150mil pins with the symbol moved 50mil towards left or right. (This way we can keep the small symbol and are still on grid.

Exchanging symbols with each other should not be a problem. All symbols will have the same pin spacing. (They are identical except for pin numbering.)

evanshultz, 2017-08-06:
My example was to shown that three characters can fit into the symbol for each pin with the pin length shrunk to 100mil, hence why I called it a "quick mock-up".

poeschlr, 2017-08-06:
Answer to naming opinion by @jkriege2:

> I think everyone is used to the odd-even ordering with the CONN_02xPP connectors, so I wouldn't rename them and see the others as variants.

applies only to veterans of kicad. The lib will not get better if way say it should stay that way because it was this way. (Understandable opinion but if it is the only reason for not using a more descriptive name i would use a more descriptive name.)

> The for the DIP-styles I would use `Conn_02xPP_DILNumbering´ (DIL=Dual-In-Line), as all DIL/DIP packages that I know use that row-ordering and people will immediately recognize what is meant (also, when they want to use such a symbol to represent e.g. a Textool-test-socket

This would rely on the future use of DIL packages. To be honest i am not sure we should rely on that. (Also DIL would suggest it is a ic connector. Not a generic all purpose connector.)

> instead of conn_02xPP_top-bottom maybe Conn_02xPP_row-row or Conn_02xPP_MultiRowNumbering?

not sure what of these 3 is the best. (all 3 are kind of ugly.)

> Instead of conn_02xPP_row-letter-first maybe Conn_02xPP_RowLetterNumbering

I can change my names to cammel case. This would violate the KLC though. Correct would be Row_Letter_First or Row_Letter_Numbering.  

>  WHat's the difference between letter-first and letter-last anyways? I don't see one in the screenshots above!

Well that's because my pin naming generator function was wrong for the row letter last version. (difference is a1 vs. 1a)

For now i kept the names as i had them (too tired to do anything new today.)
Here are my updated symbols. I added the 1 row connectors (neutral, female, male and screw terminal)
[image]

If someone wants a look at the current symbols:
https://drive.google.com/open?id=0B4cFFYaHy_b3eGJqa0lEZnU1ajA

poeschlr, 2017-08-06:
**Status Update**

For now i kept the names as i had them (too tired to do anything new today.)
Here are my updated symbols. I added the 1 row connectors (neutral, female, male and screw terminal)
They can all be generated as dual row connectors as well. (While still looking very similar)
[image]

If someone wants a look at the current symbols: (reduced pin options for easier handling while reviewing. The generator will later generate 1..40 pins per row as it is now.)
https://drive.google.com/open?id=0B4cFFYaHy_b3eGJqa0lEZnU1ajA

poeschlr, 2017-08-07:
@evanshultz wrote:
> My example was to shown that three characters can fit into the symbol for each pin with the pin length shrunk to 100mil, hence why I called it a "quick mock-up".

I think there was a misunderstanding. The pin names are not shown in connector symbols. I was "talking" about the pin numbers. (for them to fit i need longer pins not shorter pins.)
For now i used 150 mil long pins but ignored the center requirement. (I already need to violate the center requirement for symbols with an even number of pins per row. So it should not be that big if a deal.)

jkriege2, 2017-08-07:
Please have a closer look at the _MALE-connectors in the lib. They are mirrored (i.e. pins to the right), so they can be paired by default while keeping the numbering (i.e. connect 1-1, 2-2, ...)

Best,
JAN

poeschlr, 2017-08-09:
Update:
mirrored male connectors
Changed names to follow KLC.

[image]

poeschlr, 2017-08-23:
The new symbols have been merged. Thanks at all of you for your valuable input.

closed 2017-08-23 by poeschlr