KiCad/kicad-library #1473: Analog switch schematic symbols

evanshultz, 2017-07-30:
I would like to define how we will draw analog switch symbols in the official KiCad libary.

I see two different types of symbols for multi-gate switches:
- A single large gate with switches, logic/control, and power pins all together
- Multiple gates for each switch and logic/control, each having all power pins

There are also single gate switches with all pins together in one symbol. Like with single opamps, I suppose a single symbol is the way to go. Personally I still like having all power pins separate for grouping, but I'm more strongly of the belief that opamps and analog switches are similar in this respect.

Back to multi-gate switches. Here are images:
[image]

(It's obvious that this library needs additional help, as well, which I plan to address with a library overhaul.)

My understanding is that neither of the multi-gate symbols described above are OK. Like with multiple-gate opamps, we would want separate symbols for each switch, and then power pins are a separate gate. I'm not quite sure how to handle the logic/controls, though I've typically had one symbol for each switch and one symbol for logic/power and that worked out nicely.

Is that a wrong assumption? Has this been discussed before? Is there already a known best practice for analog switch symbols?

jkriege2, 2017-07-31:
I would propose to use symbols that look like the set of switch symbols that we already have ... maybe inside a box + extra subsymbols for the power-supply. I'm not sure whether generic symbols would work, as I'm not aware of a generic pinout-standard (same discussion as with single opamps), so I would be against such generics.

I would draw them like this:
[image]

Best,
JAN

jkriege2, 2017-07-31:
... or maybe like the ones in the overview there: https://www.eeweb.com/company-blog/nxp/nx3-high-performance-analog-switches

One could also think of integrating a trinagle for the input "driver", as e.g. there:
http://www.intersil.com/en/products/switches-muxs-crosspoints/medium-voltage-switches/switches-muxs-crosspoints/ISL43210.html

... Also the pin-location (input on the left) in the latter case would fit nicely with the existing KLC.

Best,
JAN

PS: In my drawing above, pins 1,2 should be passive (I think)!

SchrodingersGat, 2017-08-01:
The angled lines in the current symbols are pretty ugly IMO. The example that Jan has drawn above is good (at least for SPST switch). 

I like the style of the first link than Jan posted, but I think adding a triangle (as per the second link) is a little too complicated.

jkriege2, 2017-10-22:
I'm doing a rework of analog_switches.lib with new symbols in the next few days.

Best,
JAN

jkriege2, 2017-10-26:
OK, PR is posted: https://github.com/KiCad/kicad-library/pull/1773

SchrodingersGat, 2017-10-26:
Thanks Jan. I'll close this issue now.

closed 2017-10-26 by SchrodingersGat