KiCad/kicad-library #1513: Alias 'HIP2100IB' not found in library

pointhi, 2017-08-11:
Opening the library results with the following error message:

```
16:25:28: Alias 'HIP2100IR4' not found in library:

'/home/thomas/Projekte/Kicad/kicad-library/library/intersil.dcm'

at line 21 offset 16
16:25:28: Alias 'HIP2100EIB' not found in library:

'/home/thomas/Projekte/Kicad/kicad-library/library/intersil.dcm'

at line 27 offset 16
16:25:28: Alias 'HIP2100IR' not found in library:

'/home/thomas/Projekte/Kicad/kicad-library/library/intersil.dcm'

at line 33 offset 15
16:25:28: Alias 'HIP2100IB' not found in library:

'/home/thomas/Projekte/Kicad/kicad-library/library/intersil.dcm'

at line 39 offset 15
```

evanshultz, 2017-08-11:
@pointhi PR submitted and linked above. Thanks for the report.

closed 2017-08-11 by pointhi

kharar, 2017-12-07:
I reported a similar issue over at #1092