KiCad/kicad-library #1532: TMLM 3D models

antoniovazquezblanco, 2017-08-13:
I have found that TRACOPOWER provides 3D models for some of their components [in their webpage](http://www.tracopower.com/products/browse-by-category/show/ac-dc-power-supplies/encapsulated-modules/tmlm/1/#panel2) under the documents tab. 
I've asked if they could be used in this project and received a positive answer:

> Dear Mr. Vázquez,
> 
> Yes, generally all our 3D models are allowed for further use in any other software, there are no restrictions on these files.
> Of course if the model is altered in a way so that the original dimensions or attributes are not preserved we won't hold any responsibility for the file.
> 
> Thanks for your efforts to get a confirmation on this matter.
> 
> Best regards,
> Simeon Tremp
> 
> > To whom it may concern,
> > 
> > I have found that among other documentation your webpage provides 3D models for your products (particularly I am interested in a TMLM04103 model). I was unable to find any copyright or usage license for these 3D models. I was wondering if these models could be added to the KiCAD component library or if there may be any kind of legal difficulty in doing so.
> > 
> > Thank you very much,
> > Antonio Vázquez

I understand that there is no legal/licensing issues in adding this 3D models to the repository. Also, I have never added a 3D model to the KiCad library and provided models are already in STP format which I believe it is a supported format. Could anyone please add them or point me to some documentation on how to do so?

Thank you very much!

poeschlr, 2017-08-24:
With the current licensing approach of the lib i think we need to go even further than what this letter allows for. Currently all our models are shipped under GPLv2 (with symbol exception).
Adding the models would mean they are not only shared but re licensed. (The current license holder needs to explicitly agree to this re licensing.)
We could add a separate repo for models like this and note it in the readme where to find them.
(I would not mix licenses in our current repo. It would be a nightmare to manage that.)

antoniovazquezblanco, 2017-09-20:
After a while I sent a third mail to TRACO POWER and they finally reached to me again:

> Thank you Mr. Simeon for the fast response to that last mail,
> I am available for communication with the KiCad team although I think it would be better for you if you participated directly in the ticket. It will reduce communication time and it will also avoid possible communication errors. If you prefer, once registered, the ticket can be followed by mail without the need to visit GitHub webpage. Please also note that the ticket is public.
> Now, moving to what KiCad needs. Currently all the models in KiCad are shipped under GPLv2. This means that if KiCad ships your models they would not only be shared but also means that they are relicensed from your current license to GPLv2. In order to do that, the current license holder (TRACO POWER) needs to either explicitly agree with the re-licensing to GPLv2 or directly release the models under the GPLv2 license.
> Thank you again for taking some time to answer my emails,
> Best regards,
> Antonio Vázquez
>
> > Dear Mr. Vázquez,
> > I'm very sorry for the delay, I had to follow up on some other topics after my vacation and couldn't give you a response earlier. We are actually very interested to get this going we are just a bit short on staff in this particular area. It would help very much if the support team from KiCad could tell us what exactly they need us to do to resolve the licensing issue.
> > On another note are you still available for the communication to KiCad since you opened the ticket on this matter or do you want us to handle this issue directly with them? Either way I want to thank you for what you have done so far on this matter, it is much appreciated.
> > Best regards,
> > Simeon Tremp

Shackmeister, 2017-09-20:
That sounds really interesting! perhaps @easyw has something to pitch in
I guess the main thing is to send them the very exact licensing documentation (has it changed for 3D models since the new license decisions? @SchrodingersGat)
Im not sure if we have an official way of documenting the agreement with the manufacturer, would an email qoute do?.

SchrodingersGat, 2017-09-20:
@Shackmeister interesting question. 

I am exploring currently the [GitHub CLA functionality](https://cla-assistant.io/) which looks pretty interesting. I'm not sure whether a company will be willing to sign off on such a thing!

Mixing licenses is going to be terrible. Asking users to sign a CLA (integrated easily into GitHub, I tested it [earlier today](https://github.com/KiCad/kicad-symbols/pull/3) - might be sufficient. 

In this case, it would *probably* require that Traco (or a representive thereof) actually make the PR. Plus there would have to be a note in the library directory stating that a license transfer has taken place.

I'm not that interested in spending time and effort chasing contributions of third-party data that is already available from the manufacturer site. If it's not going to be easy and legally bulletproof, it's a non-starter.

I don't mean to be blunt, but I've had way more than enough of licensing dramas over the last week :)

antoniovazquezblanco, 2017-09-22:
> I don't mean to be blunt, but I've had way more than enough of licensing dramas over the last week :)

I've started to follow KiCad development more closely and for some reason I had not seen that discussion until now. It seems that the legal issue is bigger than what I had thought.

From my point of view there are companies that are interested in collaborating in the development of this program but it is not always clear as to what the legal consequences of doing so are. This legal issues help to clarify the situation.

I hope the issues of the last week have been useful in order to reorganize the licensing and the libraries despite the drama. I hope a new and easier way of contributing from the legal point of view is adopted so that this issues get solved.

Thank you for your work.

antoniovazquezblanco, 2017-11-02:
I have recently seen CLA enabled in some repos. How is it going? Have you decided on wether to enable it for every repo?

SchrodingersGat, 2017-11-02:
It will only be enabled for the new library repositories that accompany v5 release (i.e. not this one)

antoniovazquezblanco, 2017-11-03:
Does CLA support 3rd party signing a PR? i.e. I open a pull request with TRACOPOWER 3D models and a representant from that company also signs the CLA to comply with licensing.

Thank you @SchrodingersGat for providing a solution to the licensing issue.

SchrodingersGat, 2017-11-03:
I think for external libraries (e.g. manufacturer 3D data) it is best to have a page with links to these external libraries. There is little point in duplicating this data when it is readily available elsewhere. The licensing issues only complicate things further.

http://kicad-pcb.org/libraries/klc/M1.2/

easyw, 2017-11-03:
Only issue with manufacturers data is placement and orientation of the provided 3d model in respect to what kicad footprint would need... 
Sometimes users get lost in trying to align models to footprint... 
For this reason in some cases it would worth to have some commonly used parts replicated...

SchrodingersGat, 2017-11-03:
@easyw I feel that this is valid, and covered by the `The contributed files provide some advantage over the data supplied by the manufacturer` exception

SchrodingersGat, 2017-11-03:
Also, hopefully with your scripts and guidance, smaller size models can be generated :)

easyw, 2017-11-03:
@SchrodingersGat 
size of STEP files is a big topic...
The main reason is the nature of STEP file format... 
1. it is a text format
2. it describes the objects with geometries
3. it must be reliable to be the base for a industrial production
I did a lot of effort in improving FreeCAD STEP interoperability
you can see a recap at the FC forum
https://forum.freecadweb.org/viewtopic.php?t=23371
There **file size** is covered by a deep analysis... but not only file size, are also covered:
- file handling performance
- step import performance
- step export performance
- step hierarchy import and export
ATM FreeCAD 0.17 can handle huge STEP files or assembly with similar and in many cases better than commercial software and **it is the only open source or free package that can do it with this level of quality and compatibility**.
File size is just a small part of the all vision... and also size is much better know compared to situation of one year ago and with other commercial software.
MCAD world is comfortable with huge STEP assemblies (typically a MCAD hardware is very powerful and rich of resources)... but then a library of STEP accurate models can just be very heavy...
IMO the only way to manage this kind of data is a way to download only what is needed for the project that a user is developing...
here a sample test real world file of an open source project
https://github.com/sgrogan/FreeCAD/releases/download/STEP/steptest.7z
it is 20MB zipped and 200MB unzipped STEP file... before those enhancements that was just NOT hand-able at all in FC...
I already suggested that a python pcbnew macro or action script could create a list of needed STEP/WRL models containing also their respective folders...
Then the macro could download with a wget only what is missing to build what a user would really need ... Someone also suggested a way to check if the already present needed models would be up to date or not, and in case update the required ones.
I don't think that having an online HUGE library would be a problem if Kicad would download only what is needed for the project that a user is developing... Maybe a basic library containing most common parts could be offered with the basic installation, but other models would be just downloaded as requested.
 
Please consider that FreeCAD MCAD kernel is OCC and that is also the one used by well known and expensive commercial ECAD ...
ATM I think KiCAD is doing a great job for ECAD world and it has the right way to collaborate with the mechanical environment... Some commercial packages just followed what we have done in now more than two years of developing...
This is what I have learned in my MCAD experience since I started to need a mechanical collaboration for electronic designs.
Thank you again for your great effort in making the library really affordable and complete. I managed to develop almost all my internal use parts, but if I had started now in using Kicad, I don't think I would have to create nothing...
And thank you also for the rigid selection of the library that you @poeschlr @Shackmeister  @pointhi  and other librarians managed to have in Kicad... 
I agree so much with Renè when he pointed out that a library quality is measured by its weakest part...
Maurice

antoniovazquezblanco, 2017-11-03:
Given the statements from @SchrodingersGat this issue can then be closed as the point was to clarify if there is a way for the manufacturer to collaborate into suplying their models with KiCad just by relicensing (knowing that there may be other people willing to prepare the files for them to sign).

The answer is no.

In my opinion I know it now is less hastle for maintainers but in the long run renouncing to good 3D models from manufacturers that may be willing to contribute to the open source software community is not the best choice. I know it is a lot of work and it may not seem worth it.

I see that this has been a trigger to licensing clarification and hope it was positive for everyone. I think that for anyone really willing to put some effort into collaboration there is a clearer extent to what the current legal/licensing situation is.

Thank you again everybody for the effort put into this.

closed 2017-11-03 by antoniovazquezblanco