KiCad/kicad-library #1546: Welcome message for new contributors

evanshultz, 2017-08-18:
Librarians, is there a way to make an introductory message? It would be best if this could key off of the status of the account submitting the case, such that a first-time contributor would get a message explaining what KLC is and how to find it along with the FAQ. It would also go over how to use the Travis output. There's probably some other worthwhile messaging to add too.

bobc, 2017-08-23:
I don't believe it is possible to send messages like that, unfortunately that sort of feature gets abused by spammers.

It is possible to create templates for PRs and issues, these would be presented to every user though. https://github.com/blog/2111-issue-and-pull-request-templates

poeschlr, 2017-08-24:
We have something similar i think. When creating a new pull request a yellow box appears that asks the contributor to read the community guidelines. I'm not sure where this is configured.

bobc, 2017-09-03:
>  a yellow box appears that asks the contributor to read the community guidelines. I'm not sure where this is configured.

That particular box appears because of the presence of the file "CONTRIBUTING.md" in the repo. The other templates are triggered in a similar way.

poeschlr, 2017-12-04:
I think this has been added with the following commits:
- https://github.com/KiCad/kicad-symbols/commit/ce443f0ba3741b5c41b99c76f826e4fe78f3abfd
- https://github.com/KiCad/kicad-footprints/commit/8cbe11a023650d7e4e0518a2b3f0fdaa9b7aba6d
- https://github.com/KiCad/kicad-symbols/commit/cf0098bb3581883bc88cbdb95e7a8791892b674a
- https://github.com/KiCad/kicad-footprints/commit/3d9acd1b63a09def33195dfb9e88b5a5016f15d3

SchrodingersGat, 2017-12-04:
Yep I think this can be closed now @evanshultz

evanshultz, 2017-12-05:
Yep! Thanks!

closed 2017-12-05 by evanshultz