KiCad/kicad-library #1550: Taging the lib for 4.0.7 release

poeschlr, 2017-08-19:
Is there a way to tag all repos at once in preparation for the 4.0.7 release?
https://lists.launchpad.net/kicad-developers/msg30330.html

poeschlr, 2017-08-19:
@jkriege2 or @SchrodingersGat could someone answer this question? I think we don't have a lot of time left.

jkriege2, 2017-08-19:
Hi!

unfortunately I don't know ... I think @SchrodingersGat made the tags (at least they exist in the few repos I checked randomly) ... I think he had a script for that ...
@SchrodingersGat are you back yet and could give a hint?
@CarlPoirierCould you comment on that?

Best,
JAN

poeschlr, 2017-08-19:
@jkriege2 forgot a space so the notification to @CarlPoirier might not have gone out correctly. (Made this comment to create the correct notification.)

SchrodingersGat, 2017-08-20:
I've only done it once, for 4.0.6 - I used a simple script that cloned each repo, added a tag, and pushed to master. I have since evolved this script into the kicad-library-utils one that downloads all the .pretty libs. However the version in master does not currently do the tag and push. 

Has Wayne now indicated that he wants the libs tagged for 4.0.7? I'm still away on vacation for another week. If it needs to happen before then, it is simply a matter of someone with write access tagging each .pretty repo and pushing to master.

poeschlr, 2017-08-23:
I asked about it in the mailing list. (The answer was linked above.)
Wayne answered twice. Both answers where basically we should tag the libs.

I added a tagging functionality to the library utils script. (I also added a checkout function.)
This way the script can be used to download the libs, tag a specific state of the lib and checkout any tag (or branch)
I also enabled the possibility of pushing the tag to github. (And therefore gave the option to use the ssh url during cloning.)

I used this script to tag all footprint libs. (Wayne mentioned he wants an official release this weekend so it would be too late if we wait for your return.)
But it seems i have no write access for the following repos: (And could therefore not set the tag)
- Terminal_Blocks_Wago
- Terminal_Blocks_Phoenix
- Connector_Card
- RF_Antenna

I will also tag the kicad-library, packages3d and packages3d-source repos. (Within the next hour or so.)

evanshultz, 2017-08-23:
@poeschlr If you want to incorporate your new connector symbols I can do it in a few hours. If you want to tag a bit later, or tomorrow, then we can at least get those nice symbols in the library. Since the schematic symbols aren't connected to GitHub we won't have any way for most users to see the symbol updates until 5.0.

poeschlr, 2017-08-23:
In this case i will wait a few hours. I will also take a look at your audio lib cleanup but i fear that contribution is too large to rush in just before a release.

poeschlr, 2017-08-23:
The kicad-library repo, packages3d ans packages3d-source repo are now tagged. Which means every repo for which i have write access is tagged.

poeschlr, 2017-08-24:
All libs are now tagged.

closed 2017-08-24 by poeschlr