KiCad/kicad-library #1575: Move / link .pretty repos into kicad-library

gauravjuvekar, 2017-08-26:
It's a bit cumbersome to keep all the .pretty repos synced individually. Can we create a new `footprints` folder in this repo and put all of the `.pretty` repos inside that using `git submodule`?

Pros
- Nothing much changes for existing users --- they will have to explicitly do `git submodule update --init --recursive`
- People who want to use the 'bleeding edge' git repos can do so more easily with a simple repo.

Cons
- Lib maintainers will have to frequently update the submodule commits (though someone could use a script for this).

poeschlr, 2017-08-26:
There is a script that does all the work for you:
https://github.com/KiCad/kicad-library-utils/blob/master/download_pretty_libs.py

poeschlr, 2017-08-29:
@gauravjuvekar to clarify. I don't agree with moving repo libs into the symbol repo. In future I the setup will be 3 main repos: symbol-library, footprint-library, packages3d
With support repos for templates, packages3d-source, ...

But this is some time in the future. It does not look like we can get it for v5. (the github plugin needs the current setup. There is no sign this will change in v5.)

gauravjuvekar, 2017-10-19:
Closing since this is addressed a lot elsewhere on dev mailing list and on forums.

closed 2017-10-19 by gauravjuvekar