KiCad/kicad-library #1580: Overhaul of valve.lib

Misca1234, 2017-08-29:
What about vavle.lib (and the corresponding footprints)?
I was thinking of starting to fix it up, if it will be done, which type of symbols do we prefer

Old symbol (of http://www.r-type.org/pdfs/ecl82.pdf)
[image]

New version 1
[image]

New version 2
[image]

The "issue" with version 1 is that the H pins, which is the power pins so to speak, can not really be placed ontop and beneth, it will look strange, must be of the type "input" and not "power input". If they are, travis will complain.

jkriege2, 2017-08-29:
please also have a look at the IEC standard for electronic symbols ... as we try to follow that in other parts of the lib!

I think we can also stick to the old design, which does not have the problem of the strange hor. red lines inside the yellow backrgound + we also don't fill transistor-circles with yellow ...

Best,
JAN

bobc, 2017-08-31:
> we also don't fill transistor-circles with yellow ...

That might be our practice, but is not reflected by the KLC:

> 4.2 iii Fill style of symbol body is set to Fill background

There is no exemption for transistor or other symbols. Having to follow written rules (the KLC), as well as unwritten rules (common practice) which conflict with each other is highly frustrating for submitters!

In this case, it is not possible to fill the shape easily without the extra horizontal lines, so practically speaking it is better without the fill.

On the power pins, I would suggest keeping them horizontal, and request a 'legacy' waiver to the KLC rules on power pins.

jkriege2, 2017-09-01:
The rules in KLC are in my understanding mostly made for IC-style symbols (that should be written more explicitly there ;-) ... We mostly do exceptions for "small" symbols (R,L,C,D,transistors...) and I think that valves fit into this category. the same goes for the power-pins (whatever a power-pin is for a valve ... the electron source? ...).

Best,
JAN

bobc, 2017-09-03:
> whatever a power-pin is for a valve

For the heater... I guess not many people are familiar with valves nowadays!

SchrodingersGat, 2017-09-03:
> The rules in KLC are in my understanding mostly made for IC-style symbols (that should be written more explicitly there ;-)

It was my understanding that the "background fill" should only be for black-box symbols. The different color is used to indicate that there is something more going on inside the symbol than is readily apparent from the drawing. Discrete devices (R / L / C / Q / D) are completely described by their symbol (for the most part) and thus should not need extra highlighting.

I see now that this is not properly described in the KLC.

If y'all agree with my synopsis above, (or otherwise) please let me know and I will update the KLC to match this.

bobc, 2017-09-03:
I had no idea that certain parts of the KLC are only intended for ICs or black box symbols, I always assumed it should apply everywhere. I can't see how the average reader would know that. Anyway, clarification would be good... 

> The different color is used to indicate that there is something more going on inside the symbol than is readily apparent from the drawing.

I agree with that in principle, although it's a bit subjective. There are some test cases:
1) some crystals have the body filled - should not be.
2) op-amps have a special shape, and are not really black box? However, we have filled those.
3) flip flops are represented with a box, but the function is well known . Should we fill those?
4) logic gates, AND OR etc are surely not black box in any sense, However in #1578  I was specifically asked to fill all the logic gates which is small challenge, but perhaps we should not fill those?

I wrote more about this in #1143 . Personally, I am not that bothered whether symbols are filled or not, although some people find it quite controversial. I can easily build rules into my script, but I just need to know what the rules are! :)

SchrodingersGat, 2017-09-03:
> I can easily build rules into my script, but I just need to know what the rules are! :)

Fair point - and they are still in a state of flux :)

> I agree with that in principle, although it's a bit subjective.

Are you aware of any standard here that is less subjective? (I do not mean this facetiously, it would be good to follow best practice if such a thing is available).

Our options are:

1. Mandate that *all* symbols should be filled
1. Require that *some* symbols are filled, based on a simple ruleset e.g. `only black-box symbols`.

@poeschlr @jkriege2 any input on this?

evanshultz, 2017-09-03:
I had no idea, either. From seeing the symbols, my interpretation was that filled was used everywhere except for discrete parts (diode, resistor, capacitor, FET, etc.). I also think it looks best that way, but that's subjective. Having logic gate unfilled looks ugly to me, and my HEF4093B was merged with filled symbols and a separate power symbol some time ago (https://github.com/KiCad/kicad-library/pull/1450).

bobc, 2017-09-03:
> Are you aware of any standard here that is less subjective? (I do not mean this facetiously, it would be good to follow best practice if such a thing is available).

I'm not aware of one. Existing practice seems to be quite _ad hoc_, e.g. MCUs and connectors might be filled but nothing else is. Possibly, connectors are important because they are an external interface? Really, it seems to be at the whim of the designer. It would be nice if users could set the colors in the schematic on a per symbol basis.

> only black-box symbols.

Right, but what is black-box? It's not a rigorous definition you can readily apply. From my point of view, most ICs are fully defined by the datasheet, there is nothing hidden. Programmable devices like FPGA or MCU I would regard as a black-box - ie. with functions not shown.

IEC symbols show the function in the symbol, which you can read if you know the symbology. So does that mean they are not black-box?

It may too difficult to define generic rules, we could have a defined list of component types. Anything not on the list would have to be discussed.

poeschlr, 2018-01-20:
@Misca1234 if this is still relevant in the new repo https://github.com/KiCad/kicad-symbols please reopen it. (provide a short summary of this discussion and point back to this issue.)

closed 2018-01-20 by poeschlr