KiCad/kicad-library #1592: KLC: Pin text offset is not captured

evanshultz, 2017-09-04:
See https://github.com/KiCad/kicad-library/pull/1591. Here's an example black box (rectangular) symbol using the default 40mil pin text offset. To me, it looks excessive and gives a greater chance of text to collide inside the symbol body.

Could KLC capture this? I generally use 20mil but occasionally 10mil if that's even better.

The triangles used for opamps should be no more than 20mil except in an odd case.

For symbols with other shapes, I think we handle them case-by-case. But certainly for the two classes of symbols mentioned above, I feel we would get more consistency and better-looking symbols if we add this small tweak to KLC.

And on a related point, can this feedback be provided to the dev team. If we use 20mil most common in the KiCad libraries, why not make that the default in KiCad? Our text size is 50mil, but KiCad defaults to 60mil. Have different defaults already been requested of the dev team?

hackscribble, 2017-09-04:
Here are examples with 10mil, 20mil and 30 mil.  My preference would be 20mil.

[image]
[image]
[image]

hackscribble, 2017-09-05:
Maybe the default is 40mil to fit clock graphic style? ...

[image]

evanshultz, 2017-09-07:
@hackscribble 
In your example, 20mil looks best to me as well.

I don't know about fitting the clock pin type. I am sure that assuming the developers made the defaults suitable for the official KiCad library is wrong. By putting lines over the pin text (for example `~RESET`), IMO we capture all required info. Using clocks and other pin types isn't necessary. Again, my opinion. 4.6 doesn't really give tremendous guidance on this point. Perhaps another area to clarify?

@SchrodingersGat What do you think? This may fit nicely in section 4.2.

SchrodingersGat, 2017-09-07:
10mil looks a bit crowded IMO. I would generally agree that 20mil looks pretty good but there are cases where some leeway is needed.

KLC 4.2 is indeed a good spot for such a rule.

Should we mandate 20? Or a range of values...

jkriege2, 2017-09-07:
Hi!

especially for logic symbols, the use of clock-pins and the sort makes sense (IMHO) ... Therefore I would put a rule that say 20 as a default and then states that smaller or larger values can be used to accomodate e.g. clock-pins and other specialities.

Also: Do you think the developers can be convinced to change that default in KiCAD, because otherwise we will run into loads or PRs where the pin-ofset is set at the default and we need to ask people to change that (Also the place to change this value is - at least to me - quite non-obvious, so a screenshot might help in the KLC?

Best,
JAN

evanshultz, 2017-09-07:
I say we mandate 20mil in KLC unless another value is more suitable. Something like: `Pin Name Position Offset is set to 20mil on symbols with a body outline unless this results in graphical problems.`. I'm not totally thrilled with the end of that sentence, so maybe there's a better wording.

I also mentioned getting this to the dev team. I'll open an issue in Launchpad about this and the default text size if asked. Oliver, you might be the best person to do it, so please delegate if you want me to do it.

SchrodingersGat, 2017-09-18:
I have added this rule to the new KLC (3.x) so I'll close this issue for now.

closed 2017-09-18 by SchrodingersGat