KiCad/kicad-library #1598: KLC: Clarify fields mentioned in 4.8-4.10?

evanshultz, 2017-09-07:
@SchrodingersGat  et al

In https://github.com/KiCad/kicad-library/pull/1593#issuecomment-327110550 a contributor is asking about the differences between the fields mentioned in 4.8-4.9 and 4.10. I can see how this is confusing because the names use in KLC are similar.

In actuality, there are totally different places where info is captured and saved into two different files.

Would it be helpful to include more info about these sections to clarify they're not the same? Connecting them to the field names in the LIB and DCM file may be useful for only a select set of users, but at least some note to be clear that these are separate fields may be helpful for all users.

Thoughts?

SchrodingersGat, 2017-09-07:
I agree that more clarity is needed here. I have just tweaked 4.8/4.9/4.10 in the KLC, let me know suggestions for any further improvements that could be made.

evanshultz, 2017-09-07:
@SchrodingersGat Thanks! Looking better. A few questions:
- Since the tab name and one field name are both **Description**, and both in a red box in the screenshot, perhaps add **Description tab** to 4.10 or else get another screenshot with the box on the tab?
- Since I've seen this question a few times, perhaps give an example of multiple packages in 4.10.i.a. Keep `SOIC-8` as an example and add this sentence: `If the part is available in multiple packages, list them all with forward slash separators e.g. DIP-8/SOIC-8/TSSOP-8`.
- Should 4.10.i state that words in the Description should be capitalized?
- I'm confused by 4.10.iii. What exactly is this saying that's different than 4.10.i.a?
- 4.10.v should perhaps match the UI and use **Documentation FIle Name**?

evanshultz, 2017-09-07:
One more thing. In that linked PR here are the script results the user saw:
[image]

They noted that one of the fields is called "datasheet" by the check scripts, but the UI calls it "Documentation File Name". Between the UI using "documentation" in two area and having "datasheet" in the check scripts which isn't present in the UI, is there a way to clarify? Or is it worth changing?

hackscribble, 2017-09-07:
I've opened a PR with proposed improved wording to address @evanshultz' point.

https://github.com/KiCad/kicad-library-utils/pull/153

closed 2017-09-18 by SchrodingersGat