KiCad/kicad-library #1599: Arduino schematic symbols

giogziro95, 2017-09-07:
So, a couple of months ago, I've made a library that includes schematic symbols for Arduino Nano and Arduino Uno boards. They are also available as individual libraries. There are also the symbols for some other components. You can have a look at them [here](https://github.com/giogziro95/kicad-library/tree/master/eeschema).

I know, @jonathan-iap has already provided the templates and schematic symbols, and although they were nice, I wasn't fully happy with them, so I created my own.

A couple of advantages of the ones I've made are:

- More convenient pin placements;
- Better pin numbering;
- Aesthetically more pleasing;
- Overall, more KLC-compliant.

Here's the screenshot of Arduino Nano symbol:

[image]

Now, what should I do? I'd love to contribute.

Thanks. 🙂

jkriege2, 2017-09-07:
Hi!

please have a look at the library modules.lib in the current kicad-lib. This already contains some symbols for Arduino Nano (and some others). If you want, you can push additional Arduino symbol there. The footprints should then go to this repo: https://github.com/KiCad/Modules.pretty

Best,
JAN

jonathan-iap, 2017-09-07:
If you want before creating templates i have been this : [http://meta-blog.eklablog.com/kicad-librairie-arduino-pretty-p930786](http://meta-blog.eklablog.com/kicad-librairie-arduino-pretty-p930786)
Attention it contains errors...

giogziro95, 2017-09-09:
@jkriege2

> please have a look at the library modules.lib in the current kicad-lib. This already contains some symbols for Arduino Nano (and some others).

Great! I didn't know that. My install on Manjaro Linux doesn't include some of those libraries...

> If you want, you can push additional Arduino symbols there.

As I said, I have symbols for Uno and Nano, but the library already includes them, as you pointed out. Do you perhaps mean alternative symbols?

Also, I have symbols for 7405, 74141, a force-sensitive resistor, HX711 module, and some other things. The full list is [here](https://github.com/giogziro95/kicad-library#kicad-library). Shall I push all?

Thanks. 🙂

jonathan-iap Thank you too. 🙂

jkriege2, 2017-09-09:
Hi!

Arduino: if there is a good reason, you can always propose a changed symbol ... having two variants for one device does not really make sense.

About the other devices: Please go ahead pushing them, just as a sidenode: Before pushing the 74xx-variants, please have a look at the effort to rework those libs, currently undertaken there: https://github.com/KiCad/kicad-library/pull/1578
When pushing your symbols, please ensure they end up in the correct library to your best knowledge (if there is a better place, we will discuss that during the PR-discussion). Also: Please push them in several PRs that all only contain a few symbols that belong together thematically or by lib, this makes review easier.

Best and looking forward to your submissions,
JAN

jkriege2, 2017-09-09:
Can we close this issue?

giogziro95, 2017-09-09:
Got it, thanks.

Yeah, let's close this.

closed 2017-09-09 by giogziro95