KiCad/kicad-library #1622: KLC 3

SchrodingersGat, 2017-09-16:
Hi, interested KLC parties. Here is what is brewing for the new KLC

1. KLC will be integrated into the official KiCad website
1. Each rule will have its own page, and allow for much better description with picture examples, etc
1. KLC will be split into rule categories:
  * G - General rules
  * S - Symbol rules
  * F - Footprint rules
  * M - Model rules

Here is a (mockup) of what this will look like. I have implemented a templating system such that a new rule can be added very easily, simply by adding a new file to the /klc/ directory.

This automatically adds it (in the correct spot) to the list of rules, and each rule is displayed in a consistent (templated) fashion.

[image]

[image]

The new KLC will incorporate the various other high level discussions that have been going on, e.g.

* Organization of libraries by function
* Naming convention for symbols
* Naming convention for footprints

Example - the footprint naming rule page can have very in-depth examples for each footprint type

[image]

The kicad-library-utils scripts will need to be rearranged a little bit, but that should not be too exhaustive.

Thoughts? This should make:

1. The libraries and KLC look a lot more official
1. Make contributing easier to understand
1. Allow us to provide much more informative examples.

@poeschlr @pointhi @jkriege2 @stambaughw et al

jkriege2, 2017-09-16:
Generally a very good idea (splitting/sorting and having them on the homepage). I have one question: Do you have write-access to the homepage? Or: Who can update them there (don't know how the KiCAD homepage is managed)?

Best,
JAN

SchrodingersGat, 2017-09-16:
Good question Jan - PR can be made @ https://github.com/kicad/kicad-website

SchrodingersGat, 2017-09-16:
An example of how an extended rule description could look:

[image]

SchrodingersGat, 2017-09-16:
If ya'll want to contribute, branch is here: https://github.com/SchrodingersGat/kicad-website/tree/libraries

Ratfink, 2017-09-16:
Just throwing an idea out there as an interested spectator: I'd really like to see the next major revision of KLC following language like [RFC 2119](https://tools.ietf.org/html/rfc2119).  I see a few things now that are unclear as a result of not explicitly using formal language, e.g.

* **7.6**: Is that *really* just a SHOULD NOT, or is it supposed to be a MUST NOT?
* **8.2.i**: If the datasheet specifies a pick-and-place origin, I think it would make sense to say the footprint anchor MUST be there, not just that it SHOULD.
* **9.2**: Another rule that should probably say MUST instead of SHOULD.  Is there ever a good reason to not put the anchor of a PTH footprint on pin 1?

Of course, the general rule of exceptions being allowed at the discretion of the KiCad library team would stand.  This change would just make it clearer which rules are more likely to have exceptions approved.

As an added bonus, if KLC used language like this, deciding on warning vs. error in the kicad-library-utils checking scripts would be easy.  Not doing something KLC says you SHOULD would give a warning, not doing something you MUST would give an error.

ikletti, 2017-09-16:
I'm combining a statement of your initial post:
> a new rule can be added very easily

and a quote from [Simon's remark on the mailing list](https://lists.launchpad.net/kicad-developers/msg30751.html):

> outdated elements still have some value, so I'd like to
> keep them in the library but mark them in listings
> so users can decide whether they want to use these as-is,
> improve them or replace them with a new version.

This made me think if it would be possible (maybe even necessary) to include a refined versioning system for the KLC _plus_ every symbol/footprint/model having a field that states the KLC version that was applied during creation/modification.

The reason behind this is that you can always pull a specific release of the libraries and use it for your design, but once you've placed a symbol or footprint in your design, there's no way to tell what library version it came from.

With such version information in place you could even state something like "full compliance", "exemption" or "user modification" for people that require even more control over the state of the symbols/footprints.

SchrodingersGat, 2017-09-16:
@Ratfink yes the language on some rules could use tweaking. I'll have a close look at the particular items you suggested.

SchrodingersGat, 2017-09-16:
@ikletti I really like the idea of a version field within each symbol and footprint. This of course will necessitate a change in the file formats. 

Also how do you propose the version field gets updated? Manually set by individual users as they create or edit an item? That could be easily checked and reported by KLC scripts soaybe that's sufficient.

poeschlr, 2017-09-17:
@Ratfink wrote:
> 9.2: Another rule that should probably say MUST instead of SHOULD. Is there ever a good reason to not put the anchor of a PTH footprint on pin 1?

Is it guaranteed that THT components can not be placed by pick and place machines? If not using must here might contradict the must of rule 8.2.

SchrodingersGat, 2017-09-17:
PR for the updated [helper scripts](https://github.com/KiCad/kicad-library-utils/pull/154)

SchrodingersGat, 2017-09-17:
More website progress shots here - https://imgur.com/a/2V6E3

ikletti, 2017-09-17:
> I really like the idea of a version field within each symbol and footprint.
> This of course will necessitate a change in the file formats.

Which would be an ideal situation to also include the KiCad Part Number (KPN) as a default field ;-)

@SchrodingersGat 
> Also how do you propose the version field gets updated?
> Manually set by individual users as they create or edit an item?
> That could be easily checked and reported by KLC scripts soaybe that's sufficient.

How the parts are versioned/tagged by the users will be their business.

But if a user issues a pull request for their symbols/footprints to be included in the official libraries, the KLC helper scripts will be run against the supplied symbols/footprints and the scripts can handle the versioning/tagging. Since excemptions from the KLC are discussed along with the pull request, the marking as full compliant/excempt could be handled at the same time.

bobc, 2017-09-17:
> I really like the idea of a version field within each symbol and footprint. This of course will necessitate a change in the file formats.

Does it though? New fields can already be added to symbols, and footprints have a keywords field which could include an indicator like "KLC2".

SchrodingersGat, 2017-09-17:
I suppose so, the issue is with aliased parts but if we mark the symbol itself as KLC compliant, then the aliases can be "assumed good"?

pointhi, 2017-09-17:
>   >  9.2: Another rule that should probably say MUST instead of SHOULD. Is there ever a good reason to not put the anchor of a PTH footprint on pin 1?
> 
> Is it guaranteed that THT components can not be placed by pick and place machines? If not using must here might contradict the must of rule 8.2.

Sometimes, parts can have multiple pads with pin 1. This is especially present with [Shielding Cabinets](https://github.com/KiCad/Shielding_Cabinets.pretty)

Ratfink, 2017-09-17:
> Sometimes, parts can have multiple pads with pin 1. This is especially present with [Shielding Cabinets](https://github.com/KiCad/Shielding_Cabinets.pretty)

Okay, that one example of my uncertainty due to informal language has been answered.  The point still stands that there is some uncertainty due to issues like these.  The expanded rule descriptions will go a long way towards answering these sorts of questions, but the main concern from my earlier post is that if the KLC is intending to use some formal language for requirement levels such as RFC 2119, it's important to make that clear to the reader.

(Also, the exception for parts with multiple pins numbered 1 should be explicitly listed, no?  Then the library checking scripts could stop giving errors for rule 9.2 in e.g. Würth_36503305_30x30mm.)

matthijskooijman, 2017-09-17:
@SchrodingersGat, are you converting the current rules as-is into the new format, or are you also changing the rules at the same time? It might be useful to separate these two changes into two stages, which immediately allows taking advantage of the new structure to see the changes in the rules.

matthijskooijman, 2017-09-17:
Also, it might be useful to have a single-page version as well, for easy reading. The current TOC-style layout where you need to go to new page for details is good for reference, but not so good when you just want to read it through. Not sure how easy that would be, though?

SchrodingersGat, 2017-09-18:
@MatthewStephenson a single-page version shouldn't be too hard. Each rule is an individual file and the contents are displayed according to a template. Writing a page that shows all the rules is not difficult.

stambaughw, 2017-09-18:
>     I really like the idea of a version field within each symbol and
>     footprint. This of course will necessitate a change in the file formats.
> 
> Does it though? New fields can already be added to symbols, and
> footprints have a keywords field which could include an indicator like
> "KLC2".

There is support for versioning for the footprint library file format
(although it's never been completed or tested) but this was designed to
be handled by the plugin not in the file itself.  If there was a git or
any other vcs plugin, you could use a LIB_ID like
FP_LIB_NAME:FP_NAME/REVISION to access a specific version of the
footprint the library.  I would prefer that version information not be
added to the files themselves (it's too fragile and easy to abuse) but
tracked by a vcs type plugin.  This same concept will be available for
symbol libraries once the new file formats are in place.

SchrodingersGat, 2017-09-23:
Latest KLC layout is available here: https://github.com/schrodingersgat/kicad-website/tree/libraries (on the `libraries` branch).

I am nearing a point that I am happy with. I would like to get this finished soon. It is more of a reorganization than a rewrite of the rules.

Of particular note is the footprint naming convetion (detailed here - https://github.com/KiCad/kicad-library/issues/1447)

I would like @jkriege2 and @poeschlr to review this page in particular (it is incomplete but basically a copy-paste of that page).

@jkriege2 @evanshultz @poeschlr can you please pull down that branch and review the KLC changes I have made?

poeschlr, 2017-10-06:
For some reason i can not get hugo to render the site.

So i looked at the files without nice rendering.

A few remarks:
The extensive footprint lib reorg can only work if the developers help us by giving us one repo for all footprintprint libs. We can not do this with the current one repo per lib mess. (Would result in way too many repos.)

as of yet i am not sure if the developers got the message.

In the naming guide i miss the description about ep count, ep size, ep mask size, ....
If this is in the table that describes the prefixes (I could not find its definition) then it might be ok. But for a first time reader it might be better if every field is explained in the sections. (Otherwise they need to switch back and forth between different web sites.)

I will try to get hugo to run but it might take some time. (Strangely it renders the quickstart example page. The only thing i found out is that maybe the theme stuff is setup wrong. But then it would not render for you as well except if you did not push something.)

poeschlr, 2017-10-06:
I thought a bit about it.

What if we make the footprint naming convention a bit more top down.

Maybe have a structure like this:
First have a very basic and simple overfew of how the structure of a footprint name would look like (similar to how it is currently described in the KLC)

After that have your table that gives the prefix definition.
After that explain the rule description (including the appreciation used)
Then list the rule description for each type of footprint. This rule description is a link that links to a detailed description for each type of footprint.

In short the first page would look similar to the pdf you linked that we used as a guide. The detailed description on a separate page allows us to give first time users a very detailed guide.

If i get hugo to run today i will make a short mockup about this.

SchrodingersGat, 2017-10-06:
@poeschlr I agree with this top-down idea. Basic concepts are great for most footprints, while some require further attention.

The footprint naming schema is the last major point to cover before I am ready to submit this and start the ball rolling on KLC3.0. Your help here would be appreciated!

SchrodingersGat, 2017-10-06:
For reference I'm running Hugo on Ubuntu 16.04 and did nothing special beyond the instructions at https://github.com/kicad/kicad-website

SchrodingersGat, 2017-10-06:
[image]

Here's a screenshot of that table, for reference. It may be missing a few thing but I took a lot of pointers from your post on footprint naming.

poeschlr, 2017-10-06:
I had installed a version of hugo that is too new. I installed v0.29. Now i have 0.16 installed and it works.

poeschlr, 2017-10-06:
Footprint lib nameing:
i don't think adding deprecated to the lib name is a good idea. If we do that we break fp-lib-tables of people still using the lib. The only reason why i would say we even keep deprecated libs is such that they are not broken.

# Rule F2.1
 `Library naming should not be duplicated in footprint name`
This is contradicted by the next rule. The lib is called Housing_QFN (or Package_QFN, ...) but the footprint has QFN in it.

Orientation: i would suggest we do not mix Horizontal/Vertical and Angled/Straight. We should use only one of them. (Maybe document the other one and say it is not allowed for new contributions.)

# Rule F2.2
The table is not rendered for me. (Is something not pushed from your side or is there something wrong with my server again?)
[image]

Currently i am too tired to work on this further. Would you prefer i give my information here or that i create a pull request on your repo?

poeschlr, 2017-10-06:
Another thing: it is quite hard to identify links in the rendered web side. It took me quite some time to realize that the word "here" is the link that points to the version information.

poeschlr, 2017-10-06:
If the body size can be omitted and mm as a unit is omitted can we write footprint filters filtering for component size? (I did not yet look into this in detail. More a reminder for me to look into this tomorrow.)

SchrodingersGat, 2017-10-07:
Rene,

> Footprint lib nameing:
> i don't think adding deprecated to the lib name is a good idea. If we do that we break fp-lib-tables of people still using the lib. The only reason why i would say we even keep deprecated libs is such that they are not broken.

That was only an example I put there as a filler, and I agree it's bad. I've removed it now.
> 
> Rule F2.1
> 
> Library naming should not be duplicated in footprint name
> This is contradicted by the next rule. The lib is called Housing_QFN (or Package_QFN, ...) but the footprint has QFN in it.

You're right. My intent here was so that users did not have a library e.g. `Conn_JST:` and then name components with the form `Conn_JST:Conn_JST....`. However the footprint name still has to make sense without the footprint library name prefix.

I have deleted this item and I think that this particular requirement will be mostly solved by the context of each library.

> 
> Orientation: i would suggest we do not mix Horizontal/Vertical and Angled/Straight. We should use only one of them. (Maybe document the other one and say it is not allowed for new contributions.)

Agreed. I have changed `Straight` to `Vertical`.

> 
> Rule F2.2
> 
> The table is not rendered for me. (Is something not pushed from your side or is there something wrong with my server again?)
> screenshot from 2017-10-06 20-42-47

Do you have a file at:

```
./layouts/shortcodes/fp_prefix.html
```
?

SchrodingersGat, 2017-10-07:
> Another thing: it is quite hard to identify links in the rendered web side. It took me quite some time to realize that the word "here" is the link that points to the version information.

Good spot. Looks like the <a> has to be wrapped in a <p> for the link to have an underline.

SchrodingersGat, 2017-10-07:
> Currently i am too tired to work on this further. Would you prefer i give my information here or that i create a pull request on your repo?

If you want to tweak the KLC etc then it's probably easier to make a PR on the branch on my repo. That way I don't have to replicate your changes, I can just merge them :+1:

poeschlr, 2017-10-07:
About the table problem.
It seems there was an issue that hugo could not use shortcodes in html pages.
https://discourse.gohugo.io/t/shortcodes-only-for-markdown/934

Do we know what version the kicad server is running? (Does it run one where this has been fixed?)

SchrodingersGat, 2017-10-07:
I'll have to find out. Maybe there's a different way to do it. Do the KLC images show up in the rule pages? They use shortcodes too.

poeschlr, 2017-10-07:
In ubuntu (0.15) everything looks fine.

Fedora is a different story. (I am not sure that hugo is the problem. It could be some problem with asciidoctor, or some other lib.)
On my 0.16 install the images do not appear.

If i update to the newest version even more stuff does not work. (Lots of error messages in the terminal, pages are not reachable, ...) So i presume this version is not used by the server because otherwise somebody would have noticed that there is a problem.

sandhan26, 2017-10-07:
I would like to add that, if a django or flask application for checking KLC can be integrated to the Kicad Official website on the lines of websites like gerblook.org, where we use to review our PCB designs, it will be great i think.

poeschlr, 2017-10-09:
It took me a bit to understand hugo but i think i have a first candidate to look at.
I have split the naming convention stuff into sub pages and added the connector description plus the missing images.
If someone is interested here the link to my fork/branch
https://github.com/poeschlr/kicad-website/tree/fp_naming_convention

SchrodingersGat, 2017-10-10:
Rene I think that's a pretty good way of going about things. Hugo is pretty weird and it took me a while to get stuff looking good. If we can finalise the footprint naming schema I think we're getting pretty close

matthijskooijman, 2017-11-08:
I noticed the KLC is now on the website and the wiki version here is deprecated. What will be the place to report and discuss issues about the KLC now? By creating an issue on the kicad-website repository, or still in this repository?

SchrodingersGat, 2017-11-21:
@matthijskooijman probably the best place is the specific repository for the change

e.g. if it is related to footprints, then https://github.com/kicad/kicad-footprints

closed 2017-11-21 by SchrodingersGat