KiCad/kicad-library #1627: KLC: Connector schematic symbol conventions

matthijskooijman, 2017-09-17:
I was looking at some connector symbols today and noticed that there is quite some difference in styles. Looking at the KLC, I couldn't really figure out what the recommendation is for them.

In particular, I noticed this symbol (atmel:AVR-JTAG-10):

[image]

It violates some KLC constraints (origin is not in the middle, pin origin is on a 50mil grid, not 100). I'm not sure what KLC says about the visual style of connectors.

This particular symbol also has pin labels on the outside instead of on the inside (as is customary for filled blocks for ICs), probably because the visual style resembles the common (but not only) "shrouded header" footprint used for this connection. Because the labels are centered on the pins, any wires that you would connect go right through the labels, which doesn't really look good.

As for the visual style, it seems that there's quite some variation. The generic `CONN_*` symbols use a basic block with pins:

[image]

But more specific connectors use more graphical representations, with or without fill. Here's a fairly random selection from the `conn` library:

[image]

So, the question is: what policies do we (want to) have for connector symbols? I assume they should adhere to the regular policies (e.g. pin grid alignment) wherever possible (though a particular visual style might conflict with that), but are there more specific ideas for the visual style?

As for the AVR-JTAG connector, which I originally wanted to improve, I guess it would be good to keep the "shrouded header" visual style, but realign the pins and move the labels slightly out of the way to be more useful? Any other specific suggestions for this part?

bobc, 2017-09-17:
I don't like the overly graphic connector symbols, IMO the schematic should show logical connections, I don't need a picture of the connector in the schematic. Aesthetics are highly subjective, so these symbols tend to change every few months when someone decides they don't like the look of the current symbols. 

The AVR-JTAG symbol is particularly horrible. A simple box with labels on inside would be much better.

Unless there are labelled pins like USB, I tend to use the generic CONN_xxyy symbols.

poeschlr, 2017-09-17:
You should update your libs before you make an issue. The dilxx connectors do no longer exist and the screw terminals now use the same pin spacing as other connectors.

And the conn_02x05 does not exist in this form any more. It now is called Conn_02x05_Odd_Even, it has background fill and it's pins are on 100 mil grid.

For the AVR symbol i would go the route suggested by @bobc and use a simple rectangle with the pin names inside and the pins outside the rectangle. The current placement of the "pin names" is particularly bad because you can not easily connect wires to the pins without them going right through the text.

SchrodingersGat, 2017-09-17:
I will add a new KLC rule with the AVR img as a reference of what *not* to do! Wire connection points should all be on the *outside* of the symbol.

jkriege2, 2017-09-18:
For the AVR-connector ... maybe you can dopt a style like for the USB-connectors (not as many logos here .... of course ;-)
That's reasonably close to connectors (small rects inside), but is basically a box and the logo says what you have here ...
[image]

matthijskooijman, 2017-09-18:
> You should update your libs before you make an issue. The dilxx connectors do no longer exist and the screw terminals now use the same pin spacing as other connectors.

W00ps, totally right. I thought I had recently updated, but seems I actually didn't. I've updated the screenshot in my first post with the current versions. The AVR-JTAG symbol didn't change.

> And the conn_02x05 does not exist in this form any more. It now is called Conn_02x05_Odd_Even, it has background fill and it's pins are on 100 mil grid.

I also updated that one above. This does show a new contradiction: To get the pins on the 100mil grid, the origin of the component is 50mil off-center horizontally. I also noticed this when fiddling with the AVR symbol, that satisfying both the centered and the 100mil-grid constraints, the symbol can get unreasonably big. Is this an exception to the centered-constraint that should be made explicit in the KCL?

> I don't like the overly graphic connector symbols, IMO the schematic should show logical connections, I don't need a picture of the connector in the schematic.

I guess in some cases it could be useful to convey the exact connection. For example, I think the pin numbers in a jack plug aren't well-defined, and I'm not sure if the pin that disconnects from another pin has a name that makes it clear how it works exactly, while the DC jack image conveys that information quite cleanly. Another advantage is that, e.g. when looking at the schematic for the DIN or DB9 connector, you can easily see which pins connect to what, without having to look up the pin mapping of the connector (which can be useful for testing or production, for connectors that do not have the pin numbers labeled on the actual connector).

Still, a clean box with just pins sticking out has advantages as well. Would it make sense to have both? So a graphical and plain version?

For connectors whose pins have no particular name, just a number, you can just use the `CONN_*` symbols instead of having a separate plain version of the connector, of course. For others, such as the AVR-JTAG or DB-9-used-as-serial, it might be useful to have a dedicated symbol that shows names in addition to numbers.

> For the AVR-connector ... maybe you can dopt a style like for the USB-connectors (not as many logos here .... of course ;-)

Yeah, I think a plain box with labels on the inside is probably best. That still leaves two choices:
 - Order the pins by number, mimicing the physical layout (e.g. like Conn_02x05_Odd_Even but wider to allow for names)
 - Order the pins logically, e.g. GND at the bottom, VCC at the top, etc.

I guess both of these might have merits, so would it also make sense to allow both versions to exist? Should we have a naming convention to distinguish between these variants? e.g. "Graphical", "Ordered by number", "Ordered by function" (or are the first two really the same?)?

jkriege2, 2017-09-18:
I would make only one version, where the pins are ordered by function (see USB) ... especially here, as the symbol is more a logical symbol and less one that stands for one specific connector type (e.g. you can use 2x5 pin-heads in different pin pitches, depending on the space that is available ...)

poeschlr, 2017-09-18:
Center is a bit of a strech. I think we defined it (somewhere in another pull request or issue) as meaning as close to the center as possible while having the pins on 100mil grid. In my case the even numbered connectors are moved down (and for dual row to the right) by 50 mil to satisfy the grid requirenment.

matthijskooijman, 2017-12-06:
It seems most of the things raised in this issue are now clarified in the KLC. Three things might still need clarification:

 1. A general recommendation of how to make connector symbols look (rectangle with filled background, or a graphical representation with or without filled background). It seems the preference here is for a plain rectangle, but the current KLC contains an image of correct pin placement that contains a lot of these graphical connectors, which can be seen as a recommendation for them. Even if there is no strong preference for either, it might be good to mention both options and say both are acceptable?
 2. For rectangular connector symbols, it is customary to use small rectangles on the inside of the pins (as suggested by @jkriege2 [here](https://github.com/KiCad/kicad-library/pull/1736#issuecomment-337332377)), but the KLC does not mention this. From the current symbols, only the generic connectors, USB connectors have these rectangles. What's the rationale for using these rectangles? I just noticed that the USB symbol (see above) does not have this box on the "Shield" pin, is this because it does not correspond to a regular pin, but a mechanical mount? Or is it just an oversight?
 3. This is not really related to connectors specifically, but the KLC does not seem to define the "Pin Name Position Offset".

For the pin name offset, it seems that most symbols use 40mil:

```
$ cat *.lib |grep ^DEF | cut -f 5 -d' '|sort -n |uniq -c
    580 0
     64 1
      7 5
    390 10
      1 15
    401 20
     27 25
    370 30
   3073 40
     41 50

```
I suspect that a lot of symbols that do not use 40, do not actually have pin names set, but I have not investigated closely. To figure out what symbols use different offsets:

    cat *.lib |grep ^DEF | cut -f 2,5 -d' '|grep -v 40$|less

For the small rectangles, as [I mentioned here](https://github.com/KiCad/kicad-library/pull/1736#issuecomment-337529247) there are various sizes in use. Generic connectors (e.g. Conn_02x03_Odd_Even) use 10x50 rectangles, which work because there are no pin names. The USB1 connectors use 10x30, which fits well with the 40 mil pin name offset used by pretty much all connectors. The USB3 connector uses 20x40, which only works because all pin names end in "-". If you replace that be a normal letter, the rectangles are too near the pin name.

It seems that recommending the use of 10x30 rectangles, or 10x50 when there are no pin names is consistent with current practice.

evanshultz, 2017-12-06:
40mil generally looks too big. 20mil looks better, and 0-10mil looks fine on other symbols. Symbols with shapes other than a rectangle often benefit from other pin offsets.

See https://github.com/KiCad/kicad-library/pull/1560. This has been discussed at least a couple other times before, with me making KLC propositions to add it, but I don't find the conversations right now. I do recall that 20mil should be nominal distance with smaller being allowed (but possibly with a warning).

matthijskooijman, 2017-12-06:
#1560 uses 20mil, but for non-connector symbols (so without the rectangle). Did you intend this 20mil also for connectors, or just for parts without the rectangles? If the former, I think there's only limited room for the rectangle, 10x20 is probably too big already.

poeschlr, 2017-12-06:
The grid is only important for pins. Graphical elements can be placed wherever you want. (You can use the smallest grid supported in eeschema.)

The reason for the grid requirement used for pins is that in eeschema you can not snap to pins, but only to the grid. Having symbols designed to the largest grid (100mil) means they can be connected in every grid.

matthijskooijman, 2017-12-07:
> Edit: disclaimer after again reading your stuff i am not sure any more that i understood the reason for your discussion. If you think what i wrote below is nonsense, it probably is.

Heh, I was about to ask what you were responding to exactly when I noticed this edit :-)

If I can clarify anything, let me know.

poeschlr, 2018-01-20:
@matthijskooijman we transferred over to a new lib located in https://github.com/KiCad/kicad-symbols in preparation for the v5 release. If this is still relevant please reopen this issue over at the new repo. (provide a short summary of this discussion and point back to this issue.)

closed 2018-01-20 by poeschlr