KiCad/kicad-library #1645: Licensing for new libraries

SchrodingersGat, 2017-09-21:
Hello everyone. The new libraries (for v5 of KiCad release) will be split from this repo and moved to:

* kicad-symbols
* kicad-packages3d
* kicad-templates

They will also be relicensed under CC-BY-SA with an exception clearly stating that they are free to use in projects (commercial or otherwise) without the project required to honour the BY or SA portions of the license.

For more info on the licensing see the discussion [here](https://forum.kicad.info/t/kicad-library-licensing/7856). Note that the final wording of the license agreement has not been decided but that is very indicative of what it will be. 

The files in this repo are currently licensed under GPL. We need contributor permission to relicense under CC, even though most contributors would not object or have even thought about licensing requirements when they made contributions.

I have tagged below the major contributors to the kicad-library repository. 

**If you do not consent to your contributions being relicensed under CC** then please voice your objections here. From a practical standpoint we are not going to chase down each individual contributor. 

Cheers!

@jkriege2 
@sloth
@evanshultz 
@keruseykaryu 
@hackscribble 
@CarlPoirier 
@ricardocrudo 
@matyro 
@piotr-wiszowaty 
@nicksyn
@El3ctr0nic 
@easyw 
@jeromebeguin 
@swalense 
@ashtonchase 
@FabianInostroza 
@kajusK 
@michal777 
@DASFrank 
@mpkopec 
@jp-charras 
@poeschlr 
@andres53016 
@cedricboudinet 
@dejfson 
@ppelleti 
@ObKo 
@openelectron 
@suzizecat 
@samdolt 
@hyOzd 
@svheymann 
@Fat-Zer 
@razykov 
@pointhi 
@ChristianHirsch 
@myfreescalewebpage 
@Marsfan 
@Ratfink 
@maqifrnswa 
@lorem-ipsum 
@Schlumpf 
@t3hcatpaw 
@liftoff-sr 
@Heiko-L 
@wiebus 
@dfro 
@diggit 
@gauravjuvekar

evanshultz, 2017-09-21:
I agree.

Also, I believe the 10th handle should be @nickysn

SchrodingersGat, 2017-09-21:
Thanks @evanshultz :)

poeschlr, 2017-09-21:
Just to make it official again: I gladly have all my contributions under the new license.
Also are we moving packages3d over to kicad-packages3d or is the later a new repo and we loose the history?

SchrodingersGat, 2017-09-21:
GitHub supports repo renaming - https://github.com/kicad/kicad-packages3d

matyro, 2017-09-22:
Hi,I (Matyro) am fine with changing the license. personal interest: What is the reason to change it?CheersDominik B.

Marsfan, 2017-09-22:
@SchrodingersGat Go ahead, you can change the license. But I do have a question and a request.

The question: Is it one repo for all libraries, or one repo per library, like the footprints? 
The Request: Let me know about any needed changes to the Travis script. I would be happy to re-work it when the time rolls around.

SchrodingersGat, 2017-09-22:
One repo for all libraries. Working on merging all the footprints into a single repo, too.

Ratfink, 2017-09-22:
Tee-hee, I'm considered a major contributor! 😁 
I'm fine with the license change.

Marsfan, 2017-09-22:
> One repo for all libraries. Working on merging all the footprints into a single repo, too.

Yay, That makes it so much eaiser

andres53016, 2017-09-22:
I'm agree with the license change

nickysn, 2017-09-22:
I agree with the license change.

Schlumpf, 2017-09-22:
I agree with the license change.

jkriege2, 2017-09-22:
I agree with the license change.

Heiko-L, 2017-09-22:
I agree with the license change.

svheymann, 2017-09-22:
I agree with the license change.

lorem-ipsum, 2017-09-22:
I agree with the license change

dejfson, 2017-09-22:
I agree with new licensing (@dejfson).

cheers
.d.

----------------------------------------------
Dr. David Belohrad, Div. BE-BI
C.E.R.N. Site de Prevessin, F-01631 CERN CEDEX
http://www.cern.ch
David.Belohrad@cern.ch
Tel +41.22.76.76318  Fax +41.22.76.69056
GSM +41.75.411.3455
----------------------------------------------

gauravjuvekar, 2017-09-22:
(No idea how I became a major contributor :)

I agree to the license change.

samdolt, 2017-09-22:
I agree with the license change.

dfro, 2017-09-22:
I agree with the license change.

piotr-wiszowaty, 2017-09-22:
I agree to this change of the license.

openelectron, 2017-09-22:
I agree with the license change.

I also think the proposed changes are a great idea.

ppelleti, 2017-09-22:
I agree with the license change.

El3ctr0nic, 2017-09-22:
I agree with the license change.

diggit, 2017-09-22:
Hi, can you please write some short summary what is benefit of this change? I am not good at English licensing language. I have nothing against it, just curious...

ObKo, 2017-09-22:
I agree with the license change. CC licenses looks more suitable for such things.

hyOzd, 2017-09-22:
I agree to license change.

hackscribble, 2017-09-22:
I agree with the licence change.

easyw, 2017-09-22:
I agree with the license change.

t3hcatpaw, 2017-09-22:
I agree with the license change.

To date, have never actually checked which license exactly the libraries were subjected to. However, I also believe this is an important topic. Thank you for bringing this up and for the invitation, I'm honored :smile:

ChristianHirsch, 2017-09-22:
I agree with the license change.

ppelleti, 2017-09-23:
> Hi, can you please write some short summary what is benefit of this change? I am not good at English licensing language. I have nothing against it, just curious...

@diggit, I can't speak for the KiCad team, but my understanding is that:

* Creative Commons is better suited to non-software things (such as the KiCad libraries) than GPL/LGPL, which are very software-centric.

* Commercial users of KiCad wanted stronger assurances that using the KiCad libraries in their projects wouldn't require their whole design to be released.

There was further discussion of this in [these](https://forum.kicad.info/t/default-libraries-gpl-licencing-vs-proprietary-designs/140) [two](https://forum.kicad.info/t/kicad-library-licensing/7856) threads on the forum.

mpkopec, 2017-09-23:
Agreed :)

myfreescalewebpage, 2017-09-23:
Also agree with the license change, and totally approve the merge of repo for the footprints!

pointhi, 2017-09-23:
I agree with the license change.

kajusK, 2017-09-24:
I agree with the license change

diggit, 2017-09-25:
Tanks @ppelleti  !
I agree with license change.

El3ctr0nic, 2017-09-25:
Hi all.
It's not right topic on this but maybe U forgive me :)
I have question about 3D files.  We using .WRL files now. KiCad accept .STEP files too.
Will new repo become only .STEP files??

Regards,
Marcin

Shackmeister, 2017-09-25:
No, Step is good for mechanical integration, WRL is good for raytracing and sexy renderings. Both things will be kept

matyro, 2017-09-25:
I agree with the license change.

wiebus, 2017-09-26:
Hello

I agree with CC-BY-SA 4.0 as suggested.

But as far as i remember i published my original library contributions as public
domain or cc0, so you would not need my agree.

Perhaps one day i will be able again to contribute more.

With best regards: Bernd Wiebus alias dl1eic

wiebus, 2017-09-26:
Addendum:

I then also would suggest this CC-BY-SA 4.0 for a Buildingblock library.
For the buildingblock library look at https://github.com/KiCad/kicad-library/tree/master/library/Buildingblocks.
There are also some explanations in the pdf documets.

Maybe you would think of buildingblocks as part of a symbol library. i think, this could also be ok.
What i miss is something like a "standard document" for contributors, because buildingblocks are a little more complex as symbols. Together wit a buildingblock as a schematic you will need the cache library containing the symbols used in the buildingblock 

For the Buildingblock library i also agree with CC-BY-SA 4.0  licensing.

With best regards: Bernd Wiebus alias dl1eic

suzizecat, 2017-09-28:
Didn't even knew the previous license, but i agree with this one :wink:

matthijskooijman, 2017-11-08:
I agree with the license change.

liftoff-sr, 2017-11-22:
When I was researching this, I found a number legal opinions and actual case law that said that specifications like schematic symbols were not copyright-able.   Sadly I did not record those links.  However, if they are out there, they can probably be found again.  Then, if one adopts those legal views, i.e. that schematic symbols and maybe even footprints are not copyrightable, then any kind of license is inappropriate.

I think before getting too excited about which license, I would have spent more time on obtaining a solid legal opinion about this issue.  Not everything can be copyrighted.

poeschlr, 2018-01-20:
Closing this as we are now archiving this repo. The new symbol lib is found under https://github.com/KiCad/kicad-symbols

closed 2018-01-20 by poeschlr