KiCad/kicad-library #1663: NC pins in symbols

SchrodingersGat, 2017-09-25:
I think we need to readdress the NC pins rule. Currently we require that all NC pins are added to the schematic, but this makes for some really messy symbols:

https://user-images.githubusercontent.com/3242402/30396149-f8ee8b06-98c8-11e7-93a4-452812006358.png

If we drop this requirement then footprint filters may not work for some parts, (mostly generic parts as atomic ones already have their footprint associated).

When .sweet format arrives then that brings many-to-one pin association so that be good. Until then, I think we should drop the requirement that ALL NC pins must be added, and take the hit on footprint filtering (I don't think many users will find that an issue).

It makes the symbols a LOT cleaner, and it also reduces contribution complexity.

jkriege2, 2017-09-26:
Hi!

BTW: It's worse for TL431...

does the new format also bring many-to-none pin association? Basically we don't want all the NC-pins connected to each other. 

For now I would propose 2 options, from which the contributor can choose:
1. add all NC and drop the pin-count from the FPFilter (prefered if there are few and there is enough space in the symbol for NC-pins)
2. add non or only few of the NC pins and ADD the pin-count to the FPFilter (at least where that makes sense, e.g. for DIP/SOIC/...-packages) (prefered if the symbol gets significantly less messy and if the NC-pin count is comparable to or larger than the actual pin-count, or for simple devices like reference diodes)

SchrodingersGat, 2017-09-26:
I think that is sensible.

bobc, 2017-09-26:
I also came to the conclusion that it would be better to omit NC pins, particularly as they can actually get connected to a net despite the "No Connect" property.

However, I think the footprint filter must then contain the pin count. There are parts with multiple NC pins, in general how can the user know whether to select SOIC-14 or SOIC-16 without reference to datasheet? 

I've never liked leaving the pin count a wild card, we try to nail down every other parameter to avoid surprises for the user, but in this case we leave it open. If a part is available in different pin counts, then we need a separate symbol anyway? Otherwise, we can add multiple filters for each package/pin count.

SchrodingersGat, 2017-09-26:
> There are parts with multiple NC pins, in general how can the user know whether to select SOIC-14 or SOIC-16 without reference to datasheet?

To be honest, who's picking a footprint without reference to a datasheet? ;)

I think the advantages of dropping mandatory inclusion of NC pins far outweigh the downsides...

jkriege2, 2017-09-26:
@bobc @SchrodingersGat ... I don't really understand that discussion ... my proposal above should fullfill all requirements and the contributor may choose as he wishes ... so where's the problem?

Best,
JAN

SchrodingersGat, 2017-09-26:
There's no problem Jan, the two options you presented will work.

jkriege2, 2017-10-13:
Since this is done as my proposal now ... should we close this issue?

JAN

SchrodingersGat, 2017-10-13:
Yep thanks Jan

closed 2017-10-13 by SchrodingersGat

matthijskooijman, 2017-12-06:
I was still a bit confused a bit after reading the KLC. In particular, it now says "NC pins may be omitted", but also provides rules for hidden NC pins. Would it make sense to make the options more explicit? e.g. replace S4.5.1 by:

> Pins that are never to be connected under any circumstances, may be omitted or can be added as hidden NC pins (when this does not clutter the symbol too much).

And perhaps add a sentence to S4.6.1:

> These pins should be entirely placed within the symbol, to prevent accidental connections to them.